foxthefox/ioBroker.ecoflow-mqtt #27: Delta 2 - Warnings due to values being too high

techflick, 2023-12-17:
**Describe the bug**  
Log is spammed with warnings. Some values seem to be several orders of magnitude too high. And what's even more irritating is that they try to set a read only value ("State value to **set** status")?

**To Reproduce**  
in the logs, the following is shown:

State value to set for "ecoflow-mqtt.0.R331FEB4ZF4E0054.mppt.inVol" has value "1749.20000000000002" greater than max "150"

State value to set for "ecoflow-mqtt.0.R331FEB4ZF4E0054.mppt.carOutAmp" has value "28.810000000000002" greater than max "13"

State value to set for "ecoflow-mqtt.0.R331FEB4ZF4E0054.mppt.carOutVol" has value "1278.9" greater than max "60"

**Expected behavior**  

"inVol" 1749.20000000000002 should be 17.4920000000000002

"inAmp" does not report over value with 5.57 but should be 0.557 at input power of 10Wats.

"carOutAmp" 28.810000000000002 should be 2.8810000000000002

"carOutVol" 1278.9 should be 12.789

It seams Voltage is always factor 100 to high and Amps are always factor 10 to high. Also carOutAmp seams to be an average over time as it is only reported sparsely, perhaps this is how the inertia for power protection is realized?

How does "carOutVol" have a Max value of 60Volt? According to Specs it is 12.6Volts, so maybe in realty it is ~13V and the factor for the max out maybe is ~0.22? 

But even then it does not add up to more the 28Wats (2.88 * 12.79 = 36,84) shown by the Ecoflow app for "carOut" shown. Maybe because it is an instantaneous value that averages out to 28W? 

**Versions:**  
 - Adapter version: 0.0.11
 - JS-Controller version: 5.0.17
 - Node version: 18.19.0
 - Operating system: Raspbian GNU/Linux 10 (buster)

foxthefox, 2023-12-17:
as you may know, the API of EF is not official and therefore the factors must be found by comparing with the assumed values. I do not own a Delta 2 and the factors are used from Delta Max as best guess.
Therefore I appreciate your feedback to improve the adapter.

I will update the factors for Delta 2.
It could be the cases that they are also valid for Delta 2 Max, but until now, nothing was reported.

techflick, 2023-12-17:
Thank you very much, I indeed was afraid that it would come off a little harsh, and I apologize deeply for that. But I asked myself should I really inflate a factual error report, inhabiting some comprehension questions, with compliments? Maybe I was just too stupid to read the instructions and I would have had to adjust the factors myself in the first place?

I also hoped that the Delta 2 Max's factors would apply, but there where already some reports over differences in the "ecoflow-connector-Script zur dynamischen Leistungsanpassung" and "Adapter für Ecoflow Einbindung" threads. 

So again thank you so much for your quick response. I appreciate yours and Waly_de's work.

foxthefox, 2023-12-17:
Indeed the wording could have been more polite, but I am also happy if I get responses for the devices I do not own.
For clear errors the template is too much and it does not to be filled out.
But at other cases is is more than necessary when there is only a snippet from the log and "I have an error here".

There is also a thread in iobroker.forum related to this adapter
https://forum.iobroker.net/topic/69819/neuer-adapter-ecoflow-mqtt/

Please report more for Delta 2 in short form if you find some wrong factors.

foxthefox, 2023-12-17:
I think we should keep it open, I am working on the 0.0.12 but factor adjustments need some rework in data handling

foxthefox, 2023-12-20:
Please check the 0.0.13 from git and give feedback. 
The factors should be improved.

and if you like the adapter, a star would be nice :-)

foxthefox, 2023-12-21:
> Please check the 0.0.13 from git and give feedback. The factors should be improved.

indeed it needs 0.0.13

closed 2024-03-25 by foxthefox