foxthefox/ioBroker.ecoflow-mqtt #55: SmartHome Panel Support

VMI1, 2024-02-29:
hello

is it possible to support the ecoflow SmartHome Panel too?

I use the smart home panel with two Delta Pros. Then control takes place via the SamrtHome Panel.

foxthefox, 2024-02-29:
I am already working on it, but I have only very limited information to the received telegrams. Which ones come cyclically and which ones are rarely cyclically but still not requested. Also the telegrams when starting the App or when viewing the SHP is not known (at the power station we have there "latestQuotas" request).

So if you can support me with tests and received telegrams, it would be great.

VMI1, 2024-02-29:
I would be happy to help you test it. I also live in Germany so we can write in german?

Maybe the ecflow document wioll help you ....

[EcoFlow Developer._SHPpdf.pdf](https://github.com/foxthefox/ioBroker.ecoflow-mqtt/files/14444385/EcoFlow.Developer._SHPpdf.pdf)

foxthefox, 2024-02-29:
> I would be happy to help you test it. I also live in Germany so we can write in german?
> 
> Maybe the ecflow document will help you ....
> 
> [EcoFlow Developer._SHPpdf.pdf](https://github.com/foxthefox/ioBroker.ecoflow-mqtt/files/14444385/EcoFlow.Developer._SHPpdf.pdf)

Sehr gut, das räumt manches Missverständnis aus.
Den Inhalt des Dokumentes/ die Webseite kenne ich. Leider etwas lückenhaft und recht schnell zusammengewürfelt.
Der Adapter nutzt die native Schnittstelle die auch die App benutzt, die IoT Schnittstelle scheint wesentlich weniger Daten bereitzustellen. Mit der Ankündigung scheint etwas Bewegung drin zu sein, aber das wird die Zeit dann zeigen.

Es gibt übrigens einen Thread zum Adapter im iob-forum.
https://forum.iobroker.net/topic/69819/neuer-adapter-ecoflow-mqtt/
Da geht eventuell ein direkterer Austausch.

Für SHP hab ich schon einiges getan, allerdings ist es noch zu früh für eine nächste Version.
Dennoch wäre mit dem derzeitigen Adapter ein Versuch möglich:
- in der Konfiguration von Powerstation einfach das SHP konfigurieren (SHP serial# und z.B. riverpro als Gerät), das legt unnötige Datenpunkte an, veranlasst aber die Kommunikation über das Topic
- den Adapter in debug versetzten
- Häckchen bei Powerstation debug für msgSetGet und msgUpdate setzen,
- mit msgUpdate werden die zyklischen Updates sichtbar (natürlich auch die der powerstations)
- das log zeigt dann die ankommenden Telegramme (JSON), höchstwahrscheinlich immer in einer gleichen Struktur oder auch abwechselnde Typen, da bräuchte ich einen Typvertreter  
- mit msgSetGet werden die Telegramme sichtbar, die über die App ausgelöst werden
- dazu bräuchte ich die ersten Telegramme die nach dem Öffnen der App bzw. mit der Navigation auf SHP entstehen, das können recht kurze und ggf. auch Wiederholungen sein
- weiter Navigation im SHP der App könnte weitere Telegramme verursachen, auch das wäre interessant

Kommandos prüfen oder recherchieren kommt dann, wenn der erste Wurf zum SHP im Adapter drin ist.

VMI1, 2024-02-29:
hört sich sehr gut an. 

Beteilige mich gerne am testen und melde mich iobroker Formum bei Dir.

foxthefox, 2024-03-25:
ab Version 0.0.22 wird es unterstützt