foxthefox/ioBroker.ecoflow-mqtt #81: mppt.dcChgCurrent don't work

planetofrock, 2024-05-09:
Hey.
I noticed that changing (tested with River 2 MAX and River 2 PRO) the amps in the mppt.dcChgCurrent data point doesn't work. If you enter 4, 6 or 8 amps in ioBroker, the Ecoflow does not accept it and the value automatically goes to 4A in the app, 0A is displayed in ioBroker.

Greetz
Pawel

foxthefox, 2024-05-12:
Which values are possible from the App for the River2 Max and which for the River2 Pro? It might be the case that I have to adjust to the possible values.

planetofrock, 2024-05-12:
You can set the values 0, 4 and 8 in the app. If you enter this into ioBroker, you always get back the value 0.

foxthefox, 2024-05-19:
I think I have found the problem.
You can try the version 0.0.30 from git or npm.
Please check and report.

The value 0 is not part of the selections right now(usually this is not a valid value), as well the 6A is still available.
Can you send a screenshot from the App?

planetofrock, 2024-05-25:
Now it works fine. Thank you. 
[image]
[image]

foxthefox, 2024-06-04:
after it is working, issue can be closed

closed 2024-06-04 by foxthefox