foxthefox/ioBroker.ecoflow-mqtt #82: PowerOcean support

TheMonkey81, 2024-05-13:
Is it possible to support EcoFlow PowerOcean?

foxthefox, 2024-05-13:
Yes it is possible. To integrate it some investigation is needed.
You would need to capture incomming telegrams from the communication with the mqtt server.
Hence I do not now this device it might be useful to follow this gist

https://gist.github.com/foxthefox/11ac35433c228085a8f5e25d5651632a

Otherwise it would also be a try when configuring a powerstation or a powerstream with the identification of the the powerocean. Then the adapter needs to be in debug mode and the options for logging:
- msgSetGet
- msgCmd
- msgUpdate (but pollutes the log)
must be ticked.
And if it is not readable (not JSON) then it is powerstream style and "showHex" is additionally needed

TheMonkey81, 2024-07-10:
Sorry for the late answer.
I didn't know how I can do it.
But now there is an API for PowerOcean: https://developer-eu.ecoflow.com/us/document/powerocean

foxthefox, 2024-07-10:
The adapter uses in the current stage the same API as the App. Good to see that the developer API is now also supporting also Power Ocean, but the data there is very limited.
I got data from another user, which I currently use for implementation.

foxthefox, 2024-08-02:
The first datapoints are implemented, you can install 0.0.34 from git or npm.
Please check and report.
The other telegrams I will try to map to the next set of datapoints.

TheMonkey81, 2024-08-08:
Thanks for the implementation, I observed the values a couple of days. It works pretty well.

Sometimes I got realy high values in the log:
bpPwr:
`ecoflow-mqtt.0 (95941) State value to set for "ecoflow-mqtt.0.xxxxxxxxxx.statusReportBattery1.bpPwr" has value "2.249124006852535e+37" greater than max "7000"`

bpAmp:
`ecoflow-mqtt.0 (95941) State value to set for "ecoflow-mqtt.0.xxxxxxxxxx.statusReportBattery1.bpAmp" has value "36971522686665620000" greater than max "60"`
`ecoflow-mqtt.0 (95941) State value to set for "ecoflow-mqtt.0.xxxxxxxxxx.statusReportBattery1.bpAmp" has value "8608103424" greater than max "60"`

bpVol:
`ecoflow-mqtt.0 (95941) State value to set for "ecoflow-mqtt.0.xxxxxxxxxx.statusReportBattery1.bpVol" has value "2.543368362450747e+30" greater than max "60"`
`ecoflow-mqtt.0 (95941) State value to set for "ecoflow-mqtt.0.xxxxxxxxxx.statusReportBattery1.bpVol" has value "2013274752" greater than max "60"`

bpBusVol:
`ecoflow-mqtt.0 (95941) State value to set for "ecoflow-mqtt.0.xxxxxxxxxx.statusReportBattery1.bpBusVol" has value "864695870099030000" greater than max "900"`

bpSoc::
`ecoflow-mqtt.0 (95941) State value to set for "ecoflow-mqtt.0.xxxxxxxxxx.statusReportBattery1.bpSoc" has value "10670" greater than max "100"`

bpSoh:
`ecoflow-mqtt.0 (95941) State value to set for "ecoflow-mqtt.0.xxxxxxxxxx.statusReportBattery1.bpSoh" has value "10714" greater than max "100"`

And there are values that are a little bit higher then max:
bpFullCap:
`ecoflow-mqtt.0 (95941) State value to set for "ecoflow-mqtt.0.xxxxxxxxxx.statusReportBattery1.bpFullCap" has value "102496" greater than max "100000"`

bpRealSoh:
`ecoflow-mqtt.0 (95941) State value to set for "ecoflow-mqtt.0.xxxxxxxxxx.statusReportBattery1.bpRealSoh" has value "102.49687957763672" greater than max "100"`

bpRemainWatth:
`ecoflow-mqtt.0 (95941) State value to set for "ecoflow-mqtt.0.xxxxxxxxxx.statusReportBattery1.bpRemainWatth" has value "5649.13818359375" greater than max "5000"`

bpMinCellVol:
`ecoflow-mqtt.0 (95941) State value to set for "ecoflow-mqtt.0.xxxxxxxxxx.statusReportBattery1.bpMinCellVol" has value "11.964" greater than max "5"`

bpMaxCellVol:
`ecoflow-mqtt.0 (95941) State value to set for "ecoflow-mqtt.0.xxxxxxxxxx.statusReportBattery1.bpMaxCellVol" has value "11.97" greater than max "5"`

foxthefox, 2024-08-08:
thanks for the feedback, if you like the adapter, you can star the repo :-)

for those values completely nonsense I have no idea, what happens there.
as you indicate it is may be rarely or in special occasions, it should be observed.

the slightly higher values:
bpFullCap,bpRealSoh,bpRemainWatth,bpMinCellVol,bpMaxCellVol 
will be solved in the next version 0.0.38
you will need to:
- stop the adapter after upgrade
- delete those values or the whole datapoint object tree of the adapter
- start the adapter

I will inform when you can try the next version

foxthefox, 2024-08-10:
in the meantime it is 0.0.40 which you can install from npm or git
new datapoints for power ocean are done.
More will come, but that needs time to find the next ones.

Please check and report if it works

TheMonkey81, 2024-08-11:
The new datapoints are working. I compared some values with the app, they are correct.
But I noticed the values of "JTS1_ENERGY_STREAM_REPORT" are only updating when the ecoflow app or the ecoflow website is open (and a couple of minutes after it is closed).

"statusReportBattery1" and "JTS1_EMS_HEARTBEAT" are updating after the app / website is closed.

foxthefox, 2024-08-11:
Thanks for the feedback.
Battery and EMS as heartbeat is obviously the cyclic update.
The report might need a special way of triggering, whereby the APP somehow does this and gets it cyclically. 
Similiar behaviour I have seen with the power stream, but there was also one day were the energy data was incomming (even APP closed). Ineterstingly it was after the several hour long provider issue.

I think it would be time to look to the cmds which are sent by the APP.
Therefore you should install 0.0.41 first and then:
- switch the adapter into debug mode (adapter admin page on top "info"->"debug"
- tick the debug in the device row
- tick additionally msgSetGet
- save and close restarts the adapter
- check the log if it is presenting logs to the power ocean

If that is working then at a specific time
- open the APP
- shuffle around the several views
- close the APP
- post the log from that time onwards

Another try is with ticking additionally the msgCmd and perform some commands or settings in the APP.

Gonzo65, 2025-01-10:
Hi,
I got the same problem with not actual data when not using the app and I just read the issue here.
@foxthefox Because there was no response from TheMonkey81 I think the issue is still open and I tried to do your last suggestions.
I hope the following logging will help you to solve the issue or come closer to a solution:

[ecoflow-mqtt_20240110.log](https://github.com/user-attachments/files/18378313/ecoflow-mqtt_20240110.log)

I started the debug mode at about 14:52 with the above mentioned settings.
Then I opened the App, switched around the views and later I changed the backup reserve to 30% and then back to 25%.

After that I terminated the smartphone app.

Please let me know if I can help furthermore.

foxthefox, 2025-01-11:
Thanks, there is still work to do. Nevertheless the power ocean is implemented as a first version.
I will check and verify with the given [hint](https://github.com/rabits/ef-ble-reverse/tree/main/protobuf) from other issue.
There seems to be usefull information to decode the telegrams.

foxthefox, 2025-01-11:
I spent some time on the power ocean and your log.
I could decode all messages, which is huge progress. Making them available as datapoints will take a good amount of time. I will start with seemingly important ones.

The command I could see is 30% for "sysBatDsgDownLimit".

In general I think it would be good, if you could change some settings from the EF-App and provide me the log and some hints what was changed (at which time).

Gonzo65, 2025-01-12:
My PV-System got working at the beginning of December last year. Until now I am only allowed to use my own power. So the settings are very limited. I only found the "battery down limit" to be allowed to change. I do not know, what settings I can change to help you for further investigations. Any suggestions? I want to help.
By the way, is there any solution to get the values from PowerOcean continouesly? In the moment some value stop updating after a while when I close the EF-App.

foxthefox, 2025-01-12:
I think this command was part of the log.
We can skip that command gathering for now.

I have no idea, why the data transmission stops. Some people report such a behavior and others have no problem.
According others there is work at EF ongoing to exclude such adapters/integrations which are using the mechanisms of the App. May be it is related to that. 
I did not find any information what the App opening is doing differently in communication as in the adapter.
Nevertheless the adapter should gather values after 5min or so. Its not contionous, but something.

foxthefox, 2025-01-15:
@TheMonkey81 @Gonzo65 
I created version 1.2.0 where almost all datapoints are made available.
Please update the adapter and check and report.
If there are anomalies with wrong multiplication, please indicate. It can be fixed quickly.

Gonzo65, 2025-01-16:
Hello.
Much more parameters!!!
It will take some time to check them all.

Just now I only get the real power information (solar, house, grid) out of the JTS1_ENERGY_STREAM_REPORT. I cannot find a connection within the other values corresponding to the producing, comsuming or storing energy. But the stream report values are only actual when the app is running.
Does anybody has an idea how to get actual values even when the app is not running? Or do I something wrong?

foxthefox, 2025-01-29:
The issue with updates is known and its a result of EF interfering the datatransfer somehow. They try to identify "non-App" communication like this adapter and block it. But it does not affect all users. Something does the App differently than we do it here. Anyhow, after 10min a update request should bring new values.

Diehilde, 2025-02-05:
Als erste vielen Dank für die Arbeit. Der Adapter läuft und liefert eine Menge Daten.

Lösung zu der Aktualisierung der Daten: ich habe ein ausgemustertes Handy genommen. Die Ecoflow app installiert. 
Per Experteneinstellung die Abschaltung verhindert. Leider muss das Handy dauerhaft ans Netz. 
Hoffe es hilft euch weiter.

Frage zu PowerGlow (den Heizstab) hat jemand schon geschafft den anzubinden?

foxthefox, 2025-02-06:
Wenn du den Power Glow hast, dann bräuchte ich ein Mitschnitt der Telegramme. Dazu den Adapter in debug Modus versetzen und beim PowerOcean ein Haken für debug setzen und auf der ersten Seite msgUpdate und msgSetGet einen Haken setzen. Dann in der App auf Details zum Heizstab gehen, falls vorhanden. Und dann das log posten (zip ist ideal)

Diehilde, 2025-02-06:
Hi,

 

als erstes vielen Dank für deine Arbeit! Ich hoffe das ich dir helfen könnte. Falls ich es falsch gemacht habe bitte um Entschuldigung, denn ich bin in IT nicht gerade ein Normativ. 

Die Datei ist klein daher hab ich die nicht gezipt.

 

Zum Vorgang: ich habe in der App den Heizstab von „Manuel“ „automatik“ also „Solarbetrieb“ und „aus“ umgeschaltet. Die Solltemperatunr umgestellt und den Heizstab kurz arbeiten lassen damit er Strom zieht.

 

Falls du weitere Wünsche hast gerne melden. Nochmal vielen Dank.

 

Dietrich.

foxthefox, 2025-02-07:
Das Vorgehen passt, da sollte etwas in den Telegrammen dabei sein. Aber leider kann ich das log nicht sehen. Evtl vergessen es hineinzukopieren?

foxthefox, 2025-02-07:
Grundsätzlich passt es mit dem log schon. Gibt diverse unbekannte Telegramme.

Ich glaube das Häckchen beim gelben Kringel in der Zeile vom PowerOcean könnte ggf. nicht gesetzt sein (hier Beispiel Delta 2, HA sollte bei dir ohne Häckchen sein)
[image]

Ich beschäftige mich mal mit den Dingen in den Telegrammen, der Heizstab hat 6000W? Hattest du ihn beim logging mit reduzierter Leistung betrieben? 

Wenn die Sonne demnächst mal scheinen sollte, könntest du ja nochmal ein log erzeugen.

PS. den debug Modus im Wetter Adapter könntest du wieder nach Info setzen, der macht das log nur voll.

foxthefox, 2025-02-07:
@Diehilde 
Im iobroker Forum gibt es ein Thread zum Adapter (https://forum.iobroker.net/topic/69819/neuer-adapter-ecoflow-mqtt)
Dort ist evtl. die Kommunikation einfacher.

Diehilde, 2025-02-09:
Hi,

danke für den Tipp in den Adapter. Den habe ich tatsächlich übersehen.

Der Heizstab hat 6KW und arbeiten Stufenlos im Bereich 0-6000W. Er sollte mit Überschuss arbeiten. Um die Auswertung zu beschleunigen habe ich in Manuell zugeschaltet. Die Energy wurde aus den Modulen und Speicher gezogen. Der Tag heute hat den Potenzial auf mehr Sonne. Sollte der Heizstab nochmal von alleine anlaufen werde ich dir noch ein Log zukommen lassen.

Danke für den Tipp mit dem Wetterapapter!

Dir noch ein schönes Wochenende.

 

Bildschirmfoto.2025-01-25.um.21.39.15.png (view on web) <https://github.com/user-attachments/assets/39ae5c38-209a-4997-9b76-f981ba2ed1ce> 

2025-02-09 10:55:04.823 - info: host.ioBroker instance system.adapter.dwd.0 having pid 31172 terminated with code 11 (ADAPTER_REQUESTED_TERMINATION)
2025-02-09 10:55:05.046 - debug: ecoflow-mqtt.0 (1168) [PROTOBUF decode] HJ31ZDH5ZG4L0061 [update] msg#0 unknown packetType cmdId = 34 unknown deviceType cmdFunc = 96
2025-02-09 10:55:05.046 - debug: ecoflow-mqtt.0 (1168) [PROTOBUF decode] HJ31ZDH5ZG4L0061 [update] msg#0 pdata: 0a130dae12794315ae1279432894bfa0bd0638e7070a130db8a4784315b8a4784328d8bea0bd0638e7070a130d99a679431599a67943289cbea0bd0638e7070a130d38677843153867784328e0bda0bd0638e7070a130d125b794315125b794328a4bda0bd0638e7070a130d42a593431542a5934328e8bca0bd0638e7070a130d1811a743151811a74328acbca0bd0638e7070a130d3ad3a843153ad3a84328f0bba0bd0638e7070a130db2c7af4315b2c7af4328b4bba0bd0638e7070a130de025b54315e025b54328f8baa0bd0638e70710aa33
2025-02-09 10:55:05.119 - info: web.0 (1063) ==> Connected system.user.admin from 192.168.188.25
2025-02-09 10:55:05.385 - debug: ecoflow-mqtt.0 (1168) [PROTOBUF decode] HJ31ZDH5ZG4L0061 [set] msg#0 unknown packetType cmdId = 97 unknown deviceType cmdFunc = 96
2025-02-09 10:55:05.385 - debug: ecoflow-mqtt.0 (1168) [PROTOBUF decode] HJ31ZDH5ZG4L0061 [set] msg#0 pdata: 0801
2025-02-09 10:55:05.441 - debug: ecoflow-mqtt.0 (1168) [PROTOBUF decode] HJ31ZDH5ZG4L0061 [update] msg#1 unknown packetType cmdId = 39 unknown deviceType cmdFunc = 96
2025-02-09 10:55:05.441 - debug: ecoflow-mqtt.0 (1168) [PROTOBUF decode] HJ31ZDH5ZG4L0061 [update] msg#1 pdata:
2025-02-09 10:55:06.027 - debug: ecoflow-mqtt.0 (1168) [PROTOBUF decode] HJ31ZDH5ZG4L0061 [update] msg#0 unknown packetType cmdId = 34 unknown deviceType cmdFunc = 96
2025-02-09 10:55:06.027 - debug: ecoflow-mqtt.0 (1168) [PROTOBUF decode] HJ31ZDH5ZG4L0061 [update] msg#0 pdata: 0a130d8491e443158491e44328ecc3a0bd0638e7070a130d1f7eaa43151f7eaa4328b0c3a0bd0638e7070a130d0c409443150c40944328f4c2a0bd0638e7070a130db4318c4315b4318c4328b8c2a0bd0638e7070a130dbe4d7a4315be4d7a4328fcc1a0bd0638e7070a130de24b774315e24b774328c0c1a0bd0638e7070a130d3e9f7943153e9f79432884c1a0bd0638e7070a130d9e057843159e05784328c8c0a0bd0638e7070a130d24d578431524d57843288cc0a0bd0638e7070a130d68f777431568f7774328d0bfa0bd0638e70710ab33
2025-02-09 10:55:06.268 - debug: ecoflow-mqtt.0 (1168) [PROTOBUF decode] HJ31ZDH5ZG4L0061 [update] msg#1 unknown packetType cmdId = 39 unknown deviceType cmdFunc = 96
2025-02-09 10:55:06.268 - debug: ecoflow-mqtt.0 (1168) [PROTOBUF decode] HJ31ZDH5ZG4L0061 [update] msg#1 pdata:
2025-02-09 10:55:07.013 - debug: ecoflow-mqtt.0 (1168) [PROTOBUF decode] HJ31ZDH5ZG4L0061 [update] msg#0 unknown packetType cmdId = 34 unknown deviceType cmdFunc = 96
2025-02-09 10:55:07.013 - debug: ecoflow-mqtt.0 (1168) [PROTOBUF decode] HJ31ZDH5ZG4L0061 [update] msg#0 pdata: 0a130dd86c784315d86c784328c4c8a0bd0638e7070a130d0163774315016377432889c8a0bd0638e7070a130d71be77431571be774328ccc7a0bd0638e7070a130d62cc77431562cc77432890c7a0bd0638e7070a130d449a784315449a784328d4c6a0bd0638e7070a130d05df86431505df86432898c6a0bd0638e7070a130dcb22894315cb22894328dcc5a0bd0638e7070a130dbefbd34315befbd34328a0c5a0bd0638e7070a130d55a1de431555a1de4328e4c4a0bd0638e7070a130d1976de43151976de4328a8c4a0bd0638e70710ac33
2025-02-09 10:55:07.567 - debug: ecoflow-mqtt.0 (1168) [PROTOBUF decode] HJ31ZDH5ZG4L0061 [update] msg#0 unknown packetType cmdId = 5 unknown deviceType cmdFunc = 241
2025-02-09 10:55:07.568 - debug: ecoflow-mqtt.0 (1168) [PROTOBUF decode] HJ31ZDH5ZG4L0061 [update] msg#0 pdata: 25c0184e42e502d48504c4
2025-02-09 10:55:07.615 - info: smartmeter.1 (876) Received 13 values, 9 updated
2025-02-09 10:55:07.770 - debug: ecoflow-mqtt.0 (1168) [PROTOBUF decode] HJ31ZDH5ZG4L0061 [update] msg#0 unknown packetType cmdId = 50 unknown deviceType cmdFunc = 212
2025-02-09 10:55:07.770 - debug: ecoflow-mqtt.0 (1168) [PROTOBUF decode] HJ31ZDH5ZG4L0061 [update] msg#0 pdata: 0a10484633325a4548345a47424630303838
2025-02-09 10:55:07.999 - debug: ecoflow-mqtt.0 (1168) [PROTOBUF decode] HJ31ZDH5ZG4L0061 [update] msg#0 unknown packetType cmdId = 34 unknown deviceType cmdFunc = 96
2025-02-09 10:55:07.999 - debug: ecoflow-mqtt.0 (1168) [PROTOBUF decode] HJ31ZDH5ZG4L0061 [update] msg#0 pdata: 0a130d260888431526088843289ccda0bd0638e7070a130d13717943151371794328e0cca0bd0638e7070a130d6ae57843156ae5784328a4cca0bd0638e7070a130d16f87a431516f87a4328e8cba0bd0638e7070a130dcf69784315cf69784328accba0bd0638e7070a130d84e677431584e6774328f0caa0bd0638e7070a130d49427943154942794328b4caa0bd0638e7070a130da5e2794315a5e2794328f8c9a0bd0638e7070a130d95357943159535794328bcc9a0bd0638e7070a130d0c337743150c3377432880c9a0bd0638e70710ad33
2025-02-09 10:55:08.299 - debug: ecoflow-mqtt.0 (1168) [PROTOBUF decode] HJ31ZDH5ZG4L0061 [update] msg#0 unknown packetType cmdId = 33 unknown deviceType cmdFunc = 212
[... cut ...]
2025-02-09 10:55:56.478 - debug: ecoflow-mqtt.0 (1168) [PROTOBUF decode] HJ31ZDH5ZG4L0061 [update] msg#1 unknown packetType cmdId = 39 unknown deviceType cmdFunc = 96
2025-02-09 10:55:56.478 - debug: ecoflow-mqtt.0 (1168) [PROTOBUF decode] HJ31ZDH5ZG4L0061 [update] msg#1 pdata:
2025-02-09 10:55:57.588 - debug: ecoflow-mqtt.0 (1168) [PROTOBUF decode] HJ31ZDH5ZG4L0061 [update] msg#0 unknown packetType cmdId = 5 unknown deviceType cmdFunc = 241
2025-02-09 10:55:57.588 - debug: ecoflow-mqtt.0 (1168) [PROTOBUF decode] HJ31ZDH5ZG4L0061 [update] msg#0 pdata: 25a2c3

foxthefox, 2025-07-11:
I have created a preview of the next version.
Where it is split between Powerocean DC and DC FIT.
Could you please check it.

it must be installed via expert mode -> custom, described here:
https://github.com/foxthefox/ioBroker.ecoflow-mqtt/blob/1.4.0/doc/en/IOB_HA/adapter.md#new-version-for-testing-on-github-not-main-branch

Gonzo65, 2025-08-19:
Hallo.
Ich habe die Version 1.4.0 installiert. Solange die EcoFlow-App (V6.6.5.265) läuft (reicht auch im Hintergrund), werden auch die Daten der PV-Leistung im ioBroker (ecoflow-mqtt.0.HJxyz.JTS1_ENERGY_STREAM_REPORT.mpptPwr) aktualisiert. Sobalt ich die App beende, bekomme ich auch im ioBroker keine aktuellen Daten mehr von der PV-Leistung. Ist aber schon ein großer Fortschritt!

An welcher Stelle muss ich den PowerGlow implementieren, wenn der schon unterstützt wird?

Diehilde, 2025-09-13:
Vielen Dank. 
Bei mir ist er online.

[image]

foxthefox, 2025-09-13:
OK, dann schließe ich mal das issue.
Powerocean ist ja nun unterstützt.