foxthefox/ioBroker.ecoflow-mqtt #90: AC enable don't work with MAX 2

planetofrock, 2024-05-25:
Hey,
I found a new bug.
For me the AC output cannot be switched on the Delta 2 MAX and River 2 MAX. If you set the variable "inv.cfgAcEnabled" to "true" it will be set back to "false" after a few seconds.

Strangely enough, it works without any problems with the River 2 PRO.

foxthefox, 2024-05-26:
This is strange, because the newer devices should have same structure and content. We have to investigate the telegrams.

Please do the following:
- enable/activate the debug checkboxes for msgCmd, msgSet/Get
- set the adapter into debug mode, usually this restarts the adapter
- do the command action from the app
- do the opposite command from the app
- Do the command action from iob
- Do the opposite command action from iob
- Go to the log and paste here everything what is after adapter start until that point

The last 5 steps should be repeated for the other device.
You can erase the last digits of your device serial number.

planetofrock, 2024-05-27:
River 2 MAX:
Enable AC with APP:
<html><body>

ecoflow-mqtt.0 | 2024-05-27 12:57:09.402 | info | state ecoflow-mqtt.0.R611ZEB5XECQ0437.inv.cfgAcEnabled changed: true (ack = true)
-- | -- | -- | --
ecoflow-mqtt.0 | 2024-05-27 12:57:07.058 | info | state ecoflow-mqtt.0.R611ZEB5XECQ0437.mppt.cfgAcEnabled changed: true (ack = true)

</body>
</html>

Disable AC APP:
<html><body>

ecoflow-mqtt.0 | 2024-05-27 13:00:34.731 | info | state ecoflow-mqtt.0.R611ZEB5XECQ0437.mppt.cfgAcEnabled changed: false (ack = true)
-- | -- | -- | --
ecoflow-mqtt.0 | 2024-05-27 13:00:33.699 | info | state ecoflow-mqtt.0.R611ZEB5XECQ0437.inv.cfgAcEnabled changed: false (ack = true)

</body>
</html>

cfgAcEnabled "true" with ioBroker (no change on River 2 MAX):
<html><body>

ecoflow-mqtt.0 | 2024-05-27 13:01:29.399 | info | state ecoflow-mqtt.0.R611ZEB5XECQ0437.inv.cfgAcEnabled changed: true (ack = false)
-- | -- | -- | --

</body>
</html>

cfgAcEnabled "false" with ioBroker (no change on River2 MAX):
<html><body>

ecoflow-mqtt.0 | 2024-05-27 13:02:34.861 | info | state ecoflow-mqtt.0.R611ZEB5XECQ0437.inv.cfgAcEnabled changed: false (ack = false)
-- | -- | -- | --

</body>
</html>

foxthefox, 2024-05-27:
thanks for the test, but unfortunately it must be repeated after the adapter is in debug mode. You can achieve this by going into adapter setting and in the header change from info to debug via the "pencil" right beside the "info" text.

foxthefox, 2024-07-01:
you can also post the log which includes the timeframe of the test. I can dig into it, to see the interactions.

planetofrock, 2024-07-02:
[iobroker.2024-07-02.log](https://github.com/user-attachments/files/16066601/iobroker.2024-07-02.log)
[Log.txt](https://github.com/user-attachments/files/16066602/Log.txt)

Does this help? Test (AC disabled) on 12:10:08 on the R351ZEB4HF460435 Ecoflow

foxthefox, 2024-07-03:
The log.txt does contain the command which was initiated by iobroker, which looks as expected.
The ACenable command from the APP is missing in the log, that would be needed to compare the syntax.
So a new log.txt would be nice, where the command from the App is captured.

BTW there are many other issues part of that log, from other adapters and scripts.

Some of the other issues are related to the ecoflow adapter, because the telegrams contain unknown datapoints.
Therefore it would be good to tick additionally the msgUpdate, then the log contains the received telegram.

frankyboy1973, 2024-07-27:
Hi, here the same with the River2 Max

1. Off with App
2. On with App
3. Off IOB

ecoflow-mqtt.0
2024-07-27 14:34:12.349	debug	R611ZEB5XF1D1931 ->set w/o params{"id":652020900,"version":"1.0","operateType":"getAllTaskCfg","moduleType":1,"time":4563872,"code":"0","data":{"general":{"taskSavedNum":0,"conflictItem":[128],"taskMaxNum":6,"productType":71},"timeTask":{"taskCfgs":[{"taskIndex":0,"week":4,"timeScale":[0,0,0,0,0,0,0,0,0,255,255,255,63,0,0,0,0,0],"year":2024,"isCfg":0,"type":0,"isEnable":0,"sec":0,"taskPrior":0,"min":0,"timeParam":1036419,"month":4,"hour":0,"timeMode":0,"day":3},{"taskIndex":255,"week":255,"timeScale":[255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255],"year":65535,"isCfg":255,"type":255,"isEnable":0,"sec":255,"taskPrior":255,"min":255,"timeParam":4294967295,"month":255,"hour":255,"timeMode":255,"day":255},{"taskIndex":255,"week":255,"timeScale":[255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255],"year":65535,"isCfg":255,"type":255,"isEnable":0,"sec":255,"taskPrior":255,"min":255,"timeParam":4294967295,"month":255,"hour":255,"timeMode":255,"day":255},{"taskIndex":255,"week":255,"timeScale":[255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255],"year":65535,"isCfg":255,"type":255,"isEnable":0,"sec":255,"taskPrior":255,"min":255,"timeParam":4294967295,"month":255,"hour":255,"timeMode":255,"day":255},{"taskIndex":255,"week":255,"timeScale":[255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255],"year":65535,"isCfg":255,"type":255,"isEnable":0,"sec":255,"taskPrior":255,"min":255,"timeParam":4294967295,"month":255,"hour":255,"timeMode":255,"day":255},{"taskIndex":255,"week":255,"timeScale":[255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255],"year":65535,"isCfg":255,"type":255,"isEnable":0,"sec":255,"taskPrior":255,"min":255,"timeParam":4294967295,"month":255,"hour":255,"timeMode":255,"day":255}],"totalTaskNum":0}}}

ecoflow-mqtt.0
2024-07-27 14:34:12.039	debug	R611ZEB5XF1D1931 river2max -> set undefined {"params":{},"from":"iOS","lang":"de-de","id":"652020900","moduleSn":"R611ZEB5XF1D1931","moduleType":1,"operateType":"getAllTaskCfg","version":"1.0"}

ecoflow-mqtt.0
2024-07-27 14:34:01.189	debug	[store station] R611ZEB5XF1D1931 not processed inv state: FastChgWatts value 0

ecoflow-mqtt.0
2024-07-27 14:34:01.186	debug	[store station] R611ZEB5XF1D1931 not processed inv state: SlowChgWatts value 0

ecoflow-mqtt.0
2024-07-27 14:33:58.266	debug	[store station] R611ZEB5XF1D1931 not processed inv state: FastChgWatts value 0

ecoflow-mqtt.0
2024-07-27 14:33:58.263	debug	[store station] R611ZEB5XF1D1931 not processed inv state: SlowChgWatts value 0

ecoflow-mqtt.0
2024-07-27 14:33:58.257	info	state ecoflow-mqtt.0.R611ZEB5XF1D1931.inv.cfgAcEnabled changed: true (ack = true)

ecoflow-mqtt.0
2024-07-27 14:33:52.332	debug	R611ZEB5XF1D1931 ->set w/o params{"id":632020836,"version":"1.0","operateType":"getAllTaskCfg","moduleType":1,"time":4543902,"code":"0","data":{"general":{"taskSavedNum":0,"conflictItem":[128],"taskMaxNum":6,"productType":71},"timeTask":{"taskCfgs":[{"taskIndex":0,"week":4,"timeScale":[0,0,0,0,0,0,0,0,0,255,255,255,63,0,0,0,0,0],"year":2024,"isCfg":0,"type":0,"isEnable":0,"sec":0,"taskPrior":0,"min":0,"timeParam":1036419,"month":4,"hour":0,"timeMode":0,"day":3},{"taskIndex":255,"week":255,"timeScale":[255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255],"year":65535,"isCfg":255,"type":255,"isEnable":0,"sec":255,"taskPrior":255,"min":255,"timeParam":4294967295,"month":255,"hour":255,"timeMode":255,"day":255},{"taskIndex":255,"week":255,"timeScale":[255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255],"year":65535,"isCfg":255,"type":255,"isEnable":0,"sec":255,"taskPrior":255,"min":255,"timeParam":4294967295,"month":255,"hour":255,"timeMode":255,"day":255},{"taskIndex":255,"week":255,"timeScale":[255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255],"year":65535,"isCfg":255,"type":255,"isEnable":0,"sec":255,"taskPrior":255,"min":255,"timeParam":4294967295,"month":255,"hour":255,"timeMode":255,"day":255},{"taskIndex":255,"week":255,"timeScale":[255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255],"year":65535,"isCfg":255,"type":255,"isEnable":0,"sec":255,"taskPrior":255,"min":255,"timeParam":4294967295,"month":255,"hour":255,"timeMode":255,"day":255},{"taskIndex":255,"week":255,"timeScale":[255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255],"year":65535,"isCfg":255,"type":255,"isEnable":0,"sec":255,"taskPrior":255,"min":255,"timeParam":4294967295,"month":255,"hour":255,"timeMode":255,"day":255}],"totalTaskNum":0}}}

ecoflow-mqtt.0
2024-07-27 14:33:52.041	debug	R611ZEB5XF1D1931 river2max -> set undefined {"params":{},"from":"iOS","lang":"de-de","id":"632020836","moduleSn":"R611ZEB5XF1D1931","moduleType":1,"operateType":"getAllTaskCfg","version":"1.0"}

ecoflow-mqtt.0
2024-07-27 14:33:46.174	debug	R611ZEB5XF1D1931 river2max -> set undefined {"from":"Android","lang":"en-us","id":"664702794","moduleType":3,"operateType":"acOutCfg","version":"1.0","params":{"enabled":0},"moduleSn":"R611ZEB5XF1D1931"}

ecoflow-mqtt.0
2024-07-27 14:33:46.160	debug	/app/1772580463883280385/R611ZEB5XF1D1931/thing/property/set Message succesfully published.

ecoflow-mqtt.0
[... cut ...] meScale":[255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255],"year":65535,"isCfg":255,"type":255,"isEnable":0,"sec":255,"taskPrior":255,"min":255,"timeParam":4294967295,"month":255,"hour":255,"timeMode":255,"day":255}],"totalTaskNum":0}}}

ecoflow-mqtt.0
2024-07-27 14:33:32.041	debug	R611ZEB5XF1D1931 river2max -> set undefined {"params":{},"from":"iOS","lang":"de-de","id":"612020850","moduleSn":"R611ZEB5XF1D1931","moduleType":1,"operateType":"getAllTaskCfg","version":"1.0"}

foxthefox, 2024-07-27:
Now it is clear what needs to be changed. Thanks for the log.

foxthefox, 2024-08-02:
it should be fixed now with version 0.0.35.
you can install it from git or npm.
please check and report.

foxthefox, 2024-08-27:
Did you have a chance to check it?

frankyboy1973, 2024-08-31:
Hi, sorry, i was on holiday the last 3 weeks. I trie to test it in the next days

frankyboy1973, 2024-09-05:
Hi, it dont work, i can not switch the Ac Power Output (ecoflow-mqtt.0.xxxxxxxxxxxxx.inv.cfgAcEnabled) with the Adapter. When its false and i want tu turn it on with true, the datapoint shows true in red color and nothing happens on the Delta 2 Max, a few seconds later the Datapoint changes to false, the same when it on (true) and i want to turn it to off (false), the Datopoint shows red false and a Moment later it turns to true.
Sorry for my bad english. I hope you understand what i mean.

foxthefox, 2024-09-05:
In total it looks a bit strange.
Maybe it has something to do with unexpected messages, which then skip the whole message.
I have created a version 1.0.3 which may help.

If it still shows that problem, then please set some additional "ticks" in the adapter config (see screenshots)

[image]
[image]

as an example the debug enable also at the device 
[image]

frankyboy1973, 2024-09-06:
Hi, it dont work

New Installation over GitHub today Version 1.0.3
i have set the additional "ticks" from your sreenshots
Here is a new Log
Off with App
On with App
Off (false) with IOB

2024-09-06 09:24:44.206 - debug: ecoflow-mqtt.0 (52802) [JSON] R611ZEB5XF1D1931 [update] -> {"addr":0,"cmdFunc":0,"cmdId":0,"id":1811372256939343986,"version":"1.0","timestamp":1725607484,"moduleType":"3","params":{"inv.dcInVol":0,"inv.cfgAcWorkMode":0,"inv.SlowChgWatts":0,"inv.dcInAmp":0,"inv.cfgAcOutFreq":2,"inv.outputWatts":0,"inv.errCode":0,"inv.dcInTemp":0,"inv.invOutFreq":0,"inv.chargerType":0,"inv.reserved":[0,0,0,0,0,0,0,0],"inv.acInAmp":0,"inv.fanState":0,"inv.cfgAcXboost":0,"inv.cfgAcEnabled":0,"inv.outTemp":22,"inv.invType":0,"inv.cfgAcOutVol":0,"inv.acDipSwitch":0,"inv.acInVol":0,"inv.invOutVol":0,"inv.FastChgWatts":0,"inv.inputWatts":0,"inv.standbyMins":0,"inv.chgPauseFlag":0,"inv.acInFreq":0,"inv.dischargeType":0,"inv.invOutAmp":0,"inv.sysVer":67174714}}
--
2024-09-06 09:24:44.241 - debug: ecoflow-mqtt.0 (52802) [store station] R611ZEB5XF1D1931 not processed inv state: standbyMins value 0
2024-09-06 09:24:45.053 - debug: ecoflow-mqtt.0 (52802) [JSON] R611ZEB5XF1D1931 [set] -> key:undefined {"params":{"uploadEn":1,"launchSetType":0,"bsmLaunchDate":1725607485,"resetFlag":0,"sn":"0000000000000000"},"from":"iOS","lang":"de-de","id":"485003007","moduleSn":"R611ZEB5XF1D1931","moduleType":2,"operateType":"getBmsInfo","version":"1.0"}
2024-09-06 09:24:46.868 - debug: ecoflow-mqtt.0 (52802) [JSON] R611ZEB5XF1D1931 [update] -> {"addr":0,"cmdFunc":0,"cmdId":0,"id":1811372275684082176,"version":"1.0","timestamp":1725607486,"moduleType":"1","params":{"pd.usb1Watts":0,"pd.qcUsb1Watts":0,"pd.wattsInSum":0,"pd.qcUsb2Watts":0,"pd.ext3p8Port":0,"pd.lcdOffSec":300,"pd.standbyMin":30,"pd.extRj45Port":0,"pd.beepMode":0,"pd.remainTime":3805,"pd.typec2Watts":0,"pd.carWatts":0,"pd.ext4p8Port":0,"pd.brightLevel":100,"pd.chgDsgState":0,"pd.typec1Watts":0,"pd.usb2Watts":0,"pd.dcOutState":0,"pd.soc":87,"pd.carState":0,"pd.wattsOutSum":0}}
2024-09-06 09:24:46.877 - debug: ecoflow-mqtt.0 (52802) [store station] R611ZEB5XF1D1931 not processed pd state: ext3p8Port value 0
2024-09-06 09:24:46.883 - debug: ecoflow-mqtt.0 (52802) [store station] R611ZEB5XF1D1931 not processed pd state: extRj45Port value 0
2024-09-06 09:24:46.888 - debug: ecoflow-mqtt.0 (52802) station update .pd.remainTime 5999 -> 3805
2024-09-06 09:24:46.903 - debug: ecoflow-mqtt.0 (52802) [store station] R611ZEB5XF1D1931 not processed pd state: ext4p8Port value 0
2024-09-06 09:24:47.248 - debug: ecoflow-mqtt.0 (52802) [JSON] R611ZEB5XF1D1931 [update] -> {"addr":0,"cmdFunc":0,"cmdId":0,"id":1811372283809497228,"version":"1.0","timestamp":1725607487,"moduleType":"3","params":{"inv.fanState":0,"inv.cfgAcEnabled":1,"inv.cfgAcXboost":0,"inv.cfgAcWorkMode":0,"inv.invType":0,"inv.SlowChgWatts":0,"inv.acDipSwitch":0,"inv.FastChgWatts":0,"inv.invOutVol":229696,"inv.inputWatts":0,"inv.outputWatts":0,"inv.errCode":0,"inv.standbyMins":0,"inv.chgPauseFlag":0,"inv.invOutFreq":50,"inv.dischargeType":0,"inv.chargerType":0,"inv.sysVer":67174714}}
2024-09-06 09:24:47.255 - debug: ecoflow-mqtt.0 (52802) station update .inv.cfgAcEnabled false -> true
2024-09-06 09:24:47.262 - info: ecoflow-mqtt.0 (52802) state ecoflow-mqtt.0.R611ZEB5XF1D1931.inv.cfgAcEnabled changed: true (ack = true)
2024-09-06 09:24:47.316 - debug: ecoflow-mqtt.0 (52802) station update .inv.invOutVol 0 -> 229.696
2024-09-06 09:24:47.325 - debug: ecoflow-mqtt.0 (52802) [store station] R611ZEB5XF1D1931 not processed inv state: standbyMins value 0
2024-09-06 09:24:47.330 - debug: ecoflow-mqtt.0 (52802) station update .inv.invOutFreq 0 -> 50
2024-09-06 09:24:50.159 - debug: ecoflow-mqtt.0 (52802) [JSON] R611ZEB5XF1D1931 [update] -> {"addr":0,"cmdFunc":0,"cmdId":0,"id":1811372309776040010,"version":"1.0","timestamp":1725607490,"moduleType":"3","params":{"inv.fanState":0,"inv.cfgAcEnabled":1,"inv.cfgAcXboost":0,"inv.cfgAcWorkMode":0,"inv.invType":0,"inv.SlowChgWatts":0,"inv.acDipSwitch":0,"inv.FastChgWatts":0,"inv.invOutVol":230229,"inv.inputWatts":0,"inv.outputWatts":0,"inv.errCode":0,"inv.standbyMins":0,"inv.chgPauseFlag":0,"inv.dischargeType":0,"inv.chargerType":0,"inv.sysVer":67174714}}
2024-09-06 09:24:50.174 - debug: ecoflow-mqtt.0 (52802) station update .inv.invOutVol 229.696 -> 230.229
2024-09-06 09:24:50.184 - debug: ecoflow-mqtt.0 (52802) [store station] R611ZEB5XF1D1931 not processed inv state: standbyMins value 0
2024-09-06 09:24:50.423 - debug: ecoflow-mqtt.0 (52802) [JSON] R611ZEB5XF1D1931 [update] -> {"addr":0,"cmdFunc":0,"cmdId":0,"id":1811372309579301113,"version":"1.0","timestamp":1725607490,"moduleType":"2","params":{"bms_bmsStatus.sysVer":50397534,"bms_bmsStatus.openBmsIdx":1,"bms_bmsStatus.num":0,"bms_bmsStatus.minCellVol":3290,"bms_bmsStatus.f32ShowSoc":87.04,"bms_bmsStatus.bmsFault":0,"bms_bmsStatus.soc":87,"bms_bmsStatus.outputWatts":14,"bms_bmsStatus.errCode":0,"bms_bmsStatus.maxCellVol":3290,"bms_bmsStatus.inputWatts":0,"bms_bmsStatus.type":1,"bms_bmsStatus.remainCap":15654,"bms_bmsStatus.remainTime":0,"bms_bmsStatus.tagChgAmp":20000}}
2024-09-06 09:24:50.431 - debug: ecoflow-mqtt.0 (52802) station update .bmsMaster.minCellVol 3.293 -> 3.29
2024-09-06 09:24:50.441 - debug: ecoflow-mqtt.0 (52802) station update .bmsMaster.outputWatts 0 -> 1.4000000000000001
2024-09-06 09:24:50.449 - debug: ecoflow-mqtt.0 (52802) station update .bmsMaster.maxCellVol 3.293 -> 3.29
[... cut ...]
2024-09-06 09:25:22.188 - debug: ecoflow-mqtt.0 (52802) [JSON] R611ZEB5XF1D1931 [update] -> {"addr":0,"cmdFunc":0,"cmdId":0,"id":1811372584653946959,"version":"1.0","timestamp":1725607522,"moduleType":"2","params":{"bms_bmsStatus.errCode":0,"bms_bmsStatus.outputWatts":0,"bms_bmsStatus.inputWatts":0,"bms_bmsStatus.type":1,"bms_bmsStatus.sysVer":50397534,"bms_bmsStatus.openBmsIdx":1,"bms_bmsStatus.num":0,"bms_bmsStatus.remainTime":0,"bms_bmsStatus.f32ShowSoc":87.03,"bms_bmsStatus.tagChgAmp":20000,"bms_bmsStatus.bmsFault":0,"bms_bmsStatus.soc":87}}
2024-09-06 09:25:22.195 - debug: ecoflow-mqtt.0 (52802) station update .bmsMaster.inputWatts 0.1 -> 0