foxthefox/ioBroker.ecoflow-mqtt #97: Installation not possible

Git-Skeeve, 2024-06-06:
**Describe the bug**  
A clear and concise description of what the bug is.

**To Reproduce**  
Steps to reproduce the behavior:
1. Go to HACS->Userdefined Repo
2. Repo:= 'https://github.com/foxthefox/ioBroker.ecoflow-mqtt'
3. Categorie as Integration
4. See error

**Expected behavior**  
Installation of your adapter

**Screenshots & Logfiles**  
<Integration foxthefox/ioBroker.ecoflow-mqtt> Repository structure for main is not compliant

**Versions:**  
 - newest HA 2024.6.0 on Green HA

foxthefox, 2024-06-06:
It is not an HACS adapter.

The adapter is for ioBroker and has bidirectional data exchange with HA. That exchange is based on the discovery feature of MQTT service. With that feature no configuration or yaml is needed. 

**Unfortunately it requires to install the iobroker system which is based on nodejs. But that’s not a big deal and does not require a lot of resources.** 

The adapter as a gateway also reduces the HA load, hence only changed data is transferred to HA. A quick introduction I will add when having a keyboard in front of me.

Edit:
Website: https://www.iobroker.net/#en/intro
linux cmd(e.g. on raspberry pi) : curl -sLf https://iobroker.net/install.sh | bash -
Adapter Repo: https://github.com/foxthefox/ioBroker.ecoflow-mqtt

The adapter is not yet in the offical list of adapters, but via expert mode the adapter can be installed from npm/github.

mcm1957, 2024-10-14:
@foxthefox 
gueass this could be closed as it is off topic.

foxthefox, 2024-10-14:
It is correct, it refers to wrong understanding of usage within HA

closed 2024-10-14 by foxthefox