foxthefox/ioBroker.ecoflow-mqtt #99: Need help with a lot of errors in log: warn: ecoflow-mqtt.1 (1031) State "ecoflow-mqtt.1.M109ZEB********.bpundefined.moduleSn" has no existing object, this might lead to an error in future versions

Jordyleef, 2024-06-24:
So i'm trying to get some data from my ecoflow powerkit. And after much trial and error i thought i got it working. But still no data coming in so checked the log and saw this?! I dont know why? Maybe someone can help me with this?

`2024-06-24 10:39:53.757 - info: ecoflow-mqtt.1 (1031) starting. Version 0.0.33 (non-npm: foxthefox/ioBroker.ecoflow-mqtt#e04cc52932a0f10e2f638c99201217e85008563b) in /opt/iobroker/node_modules/iobroker.ecoflow-mqtt, node: v18.20.3, js-controller: 5.0.19
2024-06-24 10:39:53.786 - info: ecoflow-mqtt.1 (1031) adapter entered ready
2024-06-24 10:39:53.787 - info: ecoflow-mqtt.1 (1031) your configration:
2024-06-24 10:39:53.788 - info: ecoflow-mqtt.1 (1031) powerstream -> []
2024-06-24 10:39:53.789 - info: ecoflow-mqtt.1 (1031) powerstation -> []
2024-06-24 10:39:53.790 - info: ecoflow-mqtt.1 (1031) smartplug -> []
2024-06-24 10:39:53.790 - info: ecoflow-mqtt.1 (1031) wave -> []
2024-06-24 10:39:53.791 - info: ecoflow-mqtt.1 (1031) glacier -> []
2024-06-24 10:39:53.791 - info: ecoflow-mqtt.1 (1031) generator -> []
2024-06-24 10:39:53.792 - info: ecoflow-mqtt.1 (1031) panel -> []
2024-06-24 10:39:53.793 - info: ecoflow-mqtt.1 (1031) shelly -> []
2024-06-24 10:39:53.794 - info: ecoflow-mqtt.1 (1031) powerkit -> [{"devName":"Camper","devId":"M109ZEB********","devType":"powerkitbp5000","haEnable":false,"pstationsSlave1":false,"pstationsSlave2":false}]
2024-06-24 10:39:53.809 - info: ecoflow-mqtt.1 (1031) start device state creation ->powerkit for Id M109ZEB********
2024-06-24 10:39:54.520 - info: ecoflow-mqtt.1 (1031) pdevices states created for M109ZEB******** / powerkit / Camper
2024-06-24 10:39:54.544 - info: ecoflow-mqtt.1 (1031) [EF] going to connect to mqtt broker
2024-06-24 10:39:54.988 - info: ecoflow-mqtt.1 (1031) EF connected
2024-06-24 10:39:55.113 - info: ecoflow-mqtt.1 (1031) state ecoflow-mqtt.1.M109ZEB********.action.latestQuotas changed: true (ack = false)
2024-06-24 10:39:55.114 - info: ecoflow-mqtt.1 (1031) (ack=false) ->cmd : channel action state latestQuotas
2024-06-24 10:39:55.484 - info: ecoflow-mqtt.1 (1031) state ecoflow-mqtt.1.M109ZEB********.info.status changed: online (ack = true)
2024-06-24 10:39:55.487 - info: ecoflow-mqtt.1 (1031) [EF] M109ZEB******** is online with cloud
2024-06-24 10:39:56.513 - warn: ecoflow-mqtt.1 (1031) State "ecoflow-mqtt.1.M109ZEB********.bpundefined.moduleSn" has no existing object, this might lead to an error in future versions
2024-06-24 10:39:56.524 - warn: ecoflow-mqtt.1 (1031) State "ecoflow-mqtt.1.M109ZEB********.bpundefined.moduleSn" has no existing object, this might lead to an error in future versions
2024-06-24 10:39:56.535 - warn: ecoflow-mqtt.1 (1031) State "ecoflow-mqtt.1.M109ZEB********.bpundefined.moduleSn" has no existing object, this might lead to an error in future versions
2024-06-24 10:39:56.547 - warn: ecoflow-mqtt.1 (1031) State "ecoflow-mqtt.1.M109ZEB********.bpundefined.moduleSn" has no existing object, this might lead to an error in future versions
2024-06-24 10:39:56.558 - warn: ecoflow-mqtt.1 (1031) State "ecoflow-mqtt.1.M109ZEB********.bpundefined.moduleSn" has no existing object, this might lead to an error in future versions
2024-06-24 10:39:56.571 - warn: ecoflow-mqtt.1 (1031) State "ecoflow-mqtt.1.M109ZEB********.bpundefined.moduleSn" has no existing object, this might lead to an error in future versions
2024-06-24 10:39:56.582 - warn: ecoflow-mqtt.1 (1031) State "ecoflow-mqtt.1.M109ZEB********.bpundefined.moduleSn" has no existing object, this might lead to an error in future versions
2024-06-24 10:39:56.594 - warn: ecoflow-mqtt.1 (1031) State "ecoflow-mqtt.1.M109ZEB********.bpundefined.moduleSn" has no existing object, this might lead to an error in future versions
2024-06-24 10:39:56.604 - warn: ecoflow-mqtt.1 (1031) State "ecoflow-mqtt.1.M109ZEB********.bpundefined.moduleSn" has no existing object, this might lead to an error in future versions
2024-06-24 10:39:56.617 - warn: ecoflow-mqtt.1 (1031) State "ecoflow-mqtt.1.M109ZEB********.bpundefined.moduleSn" has no existing object, this might lead to an error in future versions
2024-06-24 10:39:56.628 - warn: ecoflow-mqtt.1 (1031) State "ecoflow-mqtt.1.M109ZEB********.bpundefined.moduleSn" has no existing object, this might lead to an error in future versions
2024-06-24 10:39:56.640 - warn: ecoflow-mqtt.1 (1031) State "ecoflow-mqtt.1.M109ZEB********.bpundefined.moduleSn" has no existing object, this might lead to an error in future versions
2024-06-24 10:39:56.650 - warn: ecoflow-mqtt.1 (1031) State "ecoflow-mqtt.1.M109ZEB********.bpundefined.moduleSn" has no existing object, this might lead to an error in future versions
2024-06-24 10:39:56.661 - warn: ecoflow-mqtt.1 (1031) State "ecoflow-mqtt.1.M109ZEB********.bpundefined.moduleSn" has no existing object, this might lead to an error in future versions
2024-06-24 10:39:56.672 - warn: ecoflow-mqtt.1 (1031) State "ecoflow-mqtt.1.M109ZEB********.bpundefined.moduleSn" has no existing object, this might lead to an error in future versions
2024-06-24 10:39:56.682 - warn: ecoflow-mqtt.1 (1031) State "ecoflow-mqtt.1.M109ZEB********.bpundefined.moduleSn" has no existing object, this might lead to an error in future versions
2024-06-24 10:39:56.694 - warn: ecoflow-mqtt.1 (1031) State "ecoflow-mqtt.1.M109ZEB********.bpundefined.moduleSn" has no existing object, this might lead to an error in future versions
2024-06-24 10:39:56.706 - warn: ecoflow-mqtt.1 (1031) State "ecoflow-mqtt.1.M109ZEB********.bpundefined.moduleSn" has no existing object, this might lead to an error in future versions
2024-06-24 10:39:56.721 - warn: ecoflow-mqtt.1 (1031) State "ecoflow-mqtt.1.M109ZEB********.bpundefined.moduleSn" has no existing object, this might lead to an error in future versions
2024-06-24 10:39:56.734 - warn: ecoflow-mqtt.1 (1031) State "ecoflow-mqtt.1.M109ZEB********.bpundefined.moduleSn" has no existing object, this might lead to an error in future versions
2024-06-24 10:39:56.746 - warn: ecoflow-mqtt.1 (1031) State "ecoflow-mqtt.1.M109ZEB********.bpundefined.moduleSn" has no existing object, this might lead to an error in future versions
2024-06-24 10:39:56.764 - warn: ecoflow-mqtt.1 (1031) State "ecoflow-mqtt.1.M109ZEB********.bpundefined.moduleSn" has no existing object, this might lead to an error in future versions
2024-06-24 10:39:56.775 - warn: ecoflow-mqtt.1 (1031) State "ecoflow-mqtt.1.M109ZEB********.bpundefined.moduleSn" has no existing object, this might lead to an error in future versions
2024-06-24 10:39:56.790 - warn: ecoflow-mqtt.1 (1031) State "ecoflow-mqtt.1.M109ZEB********.bpundefined.moduleSn" has no existing object, this might lead to an error in future versions
2024-06-24 10:39:56.800 - warn: ecoflow-mqtt.1 (1031) State "ecoflow-mqtt.1.M109ZEB********.bpundefined.moduleSn" has no existing object, this might lead to an error in future versions
2024-06-24 10:39:56.813 - warn: ecoflow-mqtt.1 (1031) State "ecoflow-mqtt.1.M109ZEB********.bpundefined.moduleSn" has no existing object, this might lead to an error in future versions
2024-06-24 10:39:56.865 - warn: ecoflow-mqtt.1 (1031) State "ecoflow-mqtt.1.M109ZEB********.bpundefined.moduleSn" has no existing object, this might lead to an error in future versions
2024-06-24 10:39:56.878 - warn: ecoflow-mqtt.1 (1031) State "ecoflow-mqtt.1.M109ZEB********.bpundefined.moduleSn" has no existing object, this might lead to an error in future versions
2024-06-24 10:39:56.891 - warn: ecoflow-mqtt.1 (1031) State "ecoflow-mqtt.1.M109ZEB********.bpundefined.moduleSn" has no existing object, this might lead to an error in future versions
2024-06-24 10:39:56.901 - warn: ecoflow-mqtt.1 (1031) State "ecoflow-mqtt.1.M109ZEB********.bpundefined.moduleSn" has no existing object, this might lead to an error in future versions
2024-06-24 10:39:56.911 - warn: ecoflow-mqtt.1 (1031) State "ecoflow-mqtt.1.M109ZEB********.bpundefined.moduleSn" has no existing object, this might lead to an error in future versions
2024-06-24 10:39:56.922 - warn: ecoflow-mqtt.1 (1031) State "ecoflow-mqtt.1.M109ZEB********.bpundefined.moduleSn" has no existing object, this might lead to an error in future versions
2024-06-24 10:39:56.935 - warn: ecoflow-mqtt.1 (1031) State "ecoflow-mqtt.1.M109ZEB********.bpundefined.moduleSn" has no existing object, this might lead to an error in future versions
[... cut ...]
2024-06-24 10:40:05.280 - info: ecoflow-mqtt.1 (1031) cleaned everything up...
2024-06-24 10:40:05.281 - info: ecoflow-mqtt.1 (1031) terminating
2024-06-24 10:40:05.283 - info: ecoflow-mqtt.1 (1031) Terminated (ADAPTER_REQUESTED_TERMINATION): Without reason
2024-06-24 10:40:05.355 - info: ecoflow-mqtt.1 (1031) ecoflow connection closed
2024-06-24 10:40:05.785 - info: ecoflow-mqtt.1 (1031) terminating
2024-06-24 10:40:05.842 - info: host.IoBroker instance system.adapter.ecoflow-mqtt.1 terminated with code 11 (ADAPTER_REQUESTED_TERMINATION)`

foxthefox, 2024-06-24:
The power kit implementation is relatively new, so there might not everything discovered and each feedback is appreciated.
For development I had only one testing telegram. So it would be great if you could activate the debug mode and activate the set/getMsg and updateMsg. Opening the App will also produce same telegrams additionally. If you could post the log, then I can check the missing properties causing the issue

foxthefox, 2024-08-04:
I may have found the issue with the undefined batterypack number.
You can try the new version 0.0.37 from git or npm.

If there is a message in the log that the battery +serial# has no kitnum, then I need the incomming JSON file for comparison.

foxthefox, 2024-08-27:
Did you have a chance to test the newer version?
In the meantime version 1.0.1 is available from git and npm.

Jordyleef, 2024-09-01:
I just found out i can enable debug log so did that and in the file is the result for you. Hope this helps.
[Iobroker Log Jordy.txt](https://github.com/user-attachments/files/16829037/Iobroker.Log.Jordy.txt)

foxthefox, 2024-09-01:
The log helps very much. So I can see that I am expecting another structure of the incomming message.
Having almost no live data is a problem, but hence you have managed to put the adapter into debug mode, you can help.

It would be great if you can go to the first page of the adapter settings and tick the msgSetGet, msgUpdate and msgCmd.
That brings some more informations related to the messages.
Hopefully there will be cyclically messages and also some message when you open the App.
It would be great i you could post the log afterwards.

Jordyleef, 2024-09-01:
If you need anything else just let me know. I'll try to give you whatever you need. You just sometimes need to tell me how to, because I don't know how iobroker works

foxthefox, 2024-09-01:
OK, thanks for the help.
For the above mentioned things you need to go to instances and click on the ecoflow-mqtt adapter "wrench" symbol.
[image]
[image]

You can try to click on the button DEBUG JSON as well, but I guess it does not return too much.

Log can be downloaded here
[image]

foxthefox, 2024-09-02:
only for completeness, another important tick:
[image]

I hope that enables you to work with iobroker and logging

Jordyleef, 2024-09-08:
I enabled everything you send me i hope this log file will help you.
[iobroker ecoflow log2 Jordy.txt](https://github.com/user-attachments/files/16922954/iobroker.ecoflow.log2.Jordy.txt)

foxthefox, 2024-09-08:
Thank you very much, that helps a lot.
You did it right.
I will check what is the difference between my expectation and reality.

foxthefox, 2024-09-09:
OK, I think now it should work.

Please do the following:
- stop the adapter (I think it is not started, because of not working properly)
- delete all objects (go to Objects, right click on the powerkit serial number and choose delete whole tree and confirm)
- install the preliminary version (see the circles in below screenshot) go to Adapter -> enable expert mode (head symbol) -> go to custom installation (octacat symbol) -> choose custom (world symbol) -> type in the url ending with 1.0.4
- start adapter

[image]

Jordyleef, 2024-09-10:
Oke i did what you send me and i still dont think it is working. Here is the new log file
[iobroker ecoflow log3 Jordy.txt](https://github.com/user-attachments/files/16941911/iobroker.ecoflow.log3.Jordy.txt)

foxthefox, 2024-09-10:
Thanks for the log.

I think the version is not installed correctly.

2024-09-10 10:24:49.462 - info: host.IoBroker iobroker npm error The git reference could not be foundnpm error command git --no-replace-objects checkout 1.0.4npm error error: pathspec '1.0.4' did not match any file(s) known to git

I will make a merge to the main repository now(5min).
So it is not needed the tree version.

please go to the same window and use GitHub as source (instead custom) and type ecoflow in the search field and select it for installation.

Jordyleef, 2024-09-10:
I think it is working, or at least doing something. There is a thing in the log that makes me think it is nog fully working:  "not processed kitscc/kitscc state: seq value 43"  See the attached log file
[iobroker ecoflow log4 Jordy.txt](https://github.com/user-attachments/files/16948183/iobroker.ecoflow.log4.Jordy.txt)

foxthefox, 2024-09-10:
Definitely it is working and the adapter receives data.
I think missing to get it connected and having data is accomplished.

Beside of the usual telegrams of the already defined values, there are some more related to the energy. 
This kind of telegram is not evaluated right now, thats why there are some messages like
"..not processed bbcin/bbcin state: seq.."
"..not processed ichigh/ichigh state: wattItem.."
this is not to much to worry about. 
"seq" is the sequence number of the telegram and "wattItem" is an array of hourly values.
bbcout": "watt":[21,21,20,16,17,20,21,21,20,19,18,20,16,16,17,26,16,0,0,0,0,0,0,0] is summed up to 325Wh.
The problem with hourly values is the timezone, so the first value might not be the first value in your timezone. 
Thats finetuning by adding such datapoint definition. I will try to do it in the next update.

I had experience with other devices that the energy values are updated in the App after each hour, whereby the data in the adapter might be continuously updated. That could create a  difference.
What might be good to know, which datapoint you are referring to.

if bbcin and bbcout should have a difference, I can not judge. Here again, it might be good to indicate the name of datapoint what is of concern. Then it is also easy to search in the log for it and look what is incomming.
 
In order to reduce the logging, you might want to untick/deselect the msgStateCreation.

foxthefox, 2025-04-07:
I think that issue could be closed.

closed 2025-04-07 by foxthefox