foxthefox/ioBroker.ecoflow-mqtt #130: Smart Home Panel - change circuit mode doesn't work as expected

thyr0ne, 2024-08-17:
Hi foxthefox,

I have 2 Delta Pros and a Smart Home Panel. I can read all the data but setting the circuit mode from grid to battery or deactivate the auto modus doesn't work as expected.

When I turn off the auto modus within the Ecoflow app the toggle for the Contol mode channel 1 turn from off to on (so the opposite).
When I try to change from grid to battery, the circuit turns off - so no power. 
Can you look into the values that are posted please?

This is the log when I try to change something 
`
ecoflow-mqtt.0
2024-08-17 09:15:50.883	info	state ecoflow-mqtt.0.SP10XXXX.loadCmdChCtrlInfos.ctrlSta_0 changed: 2 (ack = true)

ecoflow-mqtt.0
2024-08-17 09:15:50.630	info	(ack=false) ->cmd : channel loadCmdChCtrlInfos state ctrlSta_0

ecoflow-mqtt.0
2024-08-17 09:15:50.630	info	state ecoflow-mqtt.0.SP10XXXX.loadCmdChCtrlInfos.ctrlSta_0 changed: 0 (ack = false)

ecoflow-mqtt.0
2024-08-17 09:15:48.739	info	(ack=false) ->cmd : channel loadCmdChCtrlInfos state ctrlSta_0

ecoflow-mqtt.0
2024-08-17 09:15:48.739	info	state ecoflow-mqtt.0.SP10XXXXXX.loadCmdChCtrlInfos.ctrlSta_0 changed: 2 (ack = false)

ecoflow-mqtt.0
2024-08-17 09:15:42.927	info	state ecoflow-mqtt.0.SP10XXXXXXX.loadCmdChCtrlInfos.ctrlSta_0 changed: 2 (ack = true)

ecoflow-mqtt.0
2024-08-17 09:15:41.883	info	state ecoflow-mqtt.0.SP10XXXXXXX.loadCmdChCtrlInfos.ctrlSta_0 changed: 0 (ack = true)

ecoflow-mqtt.0
2024-08-17 09:15:41.612	info	(ack=false) ->cmd : channel loadCmdChCtrlInfos state ctrlSta_0

ecoflow-mqtt.0
2024-08-17 09:15:41.611	info	state ecoflow-mqtt.0.SP10XXXXXXX.loadCmdChCtrlInfos.ctrlSta_0 changed: 2 (ack = false)

ecoflow-mqtt.0
2024-08-17 09:15:32.022	info	state ecoflow-mqtt.0.SP10XXXXXXX.loadCmdChCtrlInfos.ctrlSta_0 changed: 0 (ack = true)

ecoflow-mqtt.0
2024-08-17 09:14:23.799	info	state ecoflow-mqtt.0.SP10XXXXXXX.loadCmdChCtrlInfos.ctrlSta_0 changed: 2 (ack = true)

ecoflow-mqtt.0
2024-08-17 09:14:23.439	info	(ack=false) ->cmd : channel loadCmdChCtrlInfos state ctrlSta_0`

thyr0ne, 2024-08-17:
maybe this will help: 

MODES = {
    "Auto": {"sta": 0, "ctrlMode": 0},
    "Grid": {"sta": 0, "ctrlMode": 1},
    "Battery": {"sta": 1, "ctrlMode": 1},
    "Off": {"sta": 2, "ctrlMode": 1}
}

foxthefox, 2024-08-17:
Thank you for the feedback. 
unfortunately the commands did not have much testing yet. So I am happy that you are contributing.

I am aware of that modes definition within another repo, but first I would try to work without introducing a new datapoint which handles 2 others.

Can you please do the following(assuming you have version 0.0.41 installed):
- go to the adapter admin page
- tick the debug checkbox in the device config
- tick the checkbox msgSetGet and msgCmd on the first page
- change from info to debug in the header section
- after restart of the adapter do some actions in the App
- Post the log with the timeframe of these app commands

then I can crosscheck and fix the problem

thyr0ne, 2024-08-17:
Happy to support ;)

Here is the log:
`e ctrlSta_0

ecoflow-mqtt.0
2024-08-17 20:35:05.567	info	state ecoflow-mqtt.0.SP10ZEW5ZEA40017.loadCmdChCtrlInfos.ctrlSta_0 changed: 2 (ack = false)

ecoflow-mqtt.0
2024-08-17 20:34:57.949	info	state ecoflow-mqtt.0.SP10ZEW5ZEA40017.loadCmdChCtrlInfos.ctrlSta_0 changed: 2 (ack = true)

ecoflow-mqtt.0
2024-08-17 20:34:56.964	info	state ecoflow-mqtt.0.SP10ZEW5ZEA40017.loadCmdChCtrlInfos.ctrlMode_0 changed: true (ack = true)

ecoflow-mqtt.0
2024-08-17 20:34:56.492	info	(ack=false) ->cmd : channel loadCmdChCtrlInfos state ctrlMode_0

ecoflow-mqtt.0
2024-08-17 20:34:56.491	info	state ecoflow-mqtt.0.SP10ZEW5ZEA40017.loadCmdChCtrlInfos.ctrlMode_0 changed: false (ack = false)

admin.0
2024-08-17 20:34:32.977	info	<== Disconnect system.user.admin from 2a02:3030:ad0:6d44:e9c1:d0bf:166b:388b admin

ecoflow-mqtt.0
2024-08-17 20:34:07.001	warn	[HA STATE INIT DATA] SP10ZEW5ZEA40017.emergencyStrategy.overloadMode getState returned -> null

ecoflow-mqtt.0
2024-08-17 20:34:06.752	info	state ecoflow-mqtt.0.SP10ZEW5ZEA40017.loadCmdChCtrlInfos.ctrlMode_6 changed: true (ack = true)

ecoflow-mqtt.0
2024-08-17 20:34:06.732	info	state ecoflow-mqtt.0.SP10ZEW5ZEA40017.loadCmdChCtrlInfos.ctrlSta_6 changed: 1 (ack = true)

ecoflow-mqtt.0
2024-08-17 20:34:06.652	info	state ecoflow-mqtt.0.SP10ZEW5ZEA40017.loadCmdChCtrlInfos.ctrlMode_4 changed: true (ack = true)

ecoflow-mqtt.0
2024-08-17 20:34:06.614	info	state ecoflow-mqtt.0.SP10ZEW5ZEA40017.loadCmdChCtrlInfos.ctrlSta_4 changed: 1 (ack = true)

ecoflow-mqtt.0
2024-08-17 20:34:06.573	info	state ecoflow-mqtt.0.SP10ZEW5ZEA40017.loadCmdChCtrlInfos.ctrlMode_3 changed: true (ack = true)

ecoflow-mqtt.0
2024-08-17 20:34:06.549	info	state ecoflow-mqtt.0.SP10ZEW5ZEA40017.loadCmdChCtrlInfos.ctrlSta_3 changed: 1 (ack = true)

ecoflow-mqtt.0
2024-08-17 20:34:06.506	info	state ecoflow-mqtt.0.SP10ZEW5ZEA40017.loadCmdChCtrlInfos.ctrlMode_2 changed: true (ack = true)

ecoflow-mqtt.0
2024-08-17 20:34:06.482	info	state ecoflow-mqtt.0.SP10ZEW5ZEA40017.loadCmdChCtrlInfos.ctrlSta_2 changed: 1 (ack = true)

ecoflow-mqtt.0
2024-08-17 20:34:06.436	info	state ecoflow-mqtt.0.SP10ZEW5ZEA40017.loadCmdChCtrlInfos.ctrlMode_1 changed: true (ack = true)

ecoflow-mqtt.0
2024-08-17 20:34:06.399	info	state ecoflow-mqtt.0.SP10ZEW5ZEA40017.loadCmdChCtrlInfos.ctrlSta_1 changed: 1 (ack = true)

ecoflow-mqtt.0
2024-08-17 20:34:06.350	info	state ecoflow-mqtt.0.SP10ZEW5ZEA40017.loadCmdChCtrlInfos.ctrlMode_0 changed: true (ack = true)

ecoflow-mqtt.0
2024-08-17 20:34:06.312	info	state ecoflow-mqtt.0.SP10ZEW5ZEA40017.loadCmdChCtrlInfos.ctrlSta_0 changed: 1 (ack = true) `

foxthefox, 2024-08-17:
Thanks, but it seems that you missed to put the adapter into debug mode. The log should have entries with „info“ and „debug“, I see only info

thyr0ne, 2024-08-17:
Sorry my bad, 
What I did - I‘ve:
- tried to deactivate the auto mode (doesn’t work for me at all)
- changes from off (at the this time it was battery) to grid
- changed from grid to battery 
- changed from battery to off

log:
2024-08-17 23:13:52.118 - info: admin.0 (710) <== Disconnect system.user.admin from 2a02:3030:ad0:6d44:f1da:7e98:b27a:b6b6 admin
2024-08-17 23:13:56.430 - info: admin.0 (710) ==> Connected system.user.admin from 2a02:3030:ad0:6d44:f1da:7e98:b27a:b6b6
2024-08-17 23:15:13.842 - info: host.ioBrokerPi "system.adapter.ecoflow-mqtt.0" enabled
2024-08-17 23:15:14.202 - info: host.ioBrokerPi instance system.adapter.ecoflow-mqtt.0 in version "0.0.41" (non-npm: foxthefox/ioBroker.ecoflow-mqtt#8XXXXXXxXxXXXxXxXxXxXxXx) started with pid 1968
2024-08-17 23:15:16.004 - debug: ecoflow-mqtt.0 (1968) Redis Objects: Use Redis connection: 127.0.0.1:9001
2024-08-17 23:15:16.034 - debug: ecoflow-mqtt.0 (1968) Objects client ready ... initialize now
2024-08-17 23:15:16.036 - debug: ecoflow-mqtt.0 (1968) Objects create System PubSub Client
2024-08-17 23:15:16.038 - debug: ecoflow-mqtt.0 (1968) Objects create User PubSub Client
2024-08-17 23:15:16.061 - debug: ecoflow-mqtt.0 (1968) Objects client initialize lua scripts
2024-08-17 23:15:16.070 - debug: ecoflow-mqtt.0 (1968) Objects connected to redis: 127.0.0.1:9001
2024-08-17 23:15:16.092 - debug: ecoflow-mqtt.0 (1968) Redis States: Use Redis connection: 127.0.0.1:9000
2024-08-17 23:15:16.101 - debug: ecoflow-mqtt.0 (1968) States create System PubSub Client
2024-08-17 23:15:16.102 - debug: ecoflow-mqtt.0 (1968) States create User PubSub Client
2024-08-17 23:15:16.157 - debug: ecoflow-mqtt.0 (1968) States connected to redis: 127.0.0.1:9000
2024-08-17 23:15:16.277 - info: ecoflow-mqtt.0 (1968) starting. Version 0.0.41 (non-npm: foxthefox/ioBroker.ecoflow-mqtt#8XXXXXXxXxXXXxXxXxXxXxXx) in /opt/iobroker/node_modules/iobroker.ecoflow-mqtt, node: v20.16.0, js-controller: 6.0.9
2024-08-17 23:15:16.300 - info: ecoflow-mqtt.0 (1968) adapter entered ready
2024-08-17 23:15:16.301 - info: ecoflow-mqtt.0 (1968) your configration:
2024-08-17 23:15:16.302 - info: ecoflow-mqtt.0 (1968) powerstream -> []
2024-08-17 23:15:16.303 - info: ecoflow-mqtt.0 (1968) powerstation -> [{"devName":"DELTA Pro Primary","devId":"DCXXXXXXXXXXXX","devType":"deltapro","debugEnable":false,"haEnable":true,"pstationsSlave1":true,"pstationsSlave2":false},{"devName":"DELTA Pro Zuhause","devId":"DCXXXXXXXXXXXX","devType":"deltapro","debugEnable":false,"haEnable":true,"pstationsSlave1":false,"pstationsSlave2":false}]
2024-08-17 23:15:16.303 - info: ecoflow-mqtt.0 (1968) smartplug -> []
2024-08-17 23:15:16.304 - info: ecoflow-mqtt.0 (1968) wave -> []
2024-08-17 23:15:16.304 - info: ecoflow-mqtt.0 (1968) glacier -> []
2024-08-17 23:15:16.304 - info: ecoflow-mqtt.0 (1968) generator -> []
2024-08-17 23:15:16.305 - info: ecoflow-mqtt.0 (1968) panel -> [{"devName":"Smart-Home-Panel","devId":"SP10XXXXXXXXX","devType":"panel","debugEnable":true,"haEnable":true}]
2024-08-17 23:15:16.305 - info: ecoflow-mqtt.0 (1968) shelly -> []
2024-08-17 23:15:16.305 - info: ecoflow-mqtt.0 (1968) powerkit -> []
2024-08-17 23:15:16.306 - info: ecoflow-mqtt.0 (1968) powerocean -> []
2024-08-17 23:15:16.307 - debug: ecoflow-mqtt.0 (1968) device upd {"bmsMaster":{"number":{"amp":{"max":50}}},"inv":{"number":{"cfgFastChgWatts":{"min":200,"max":2900}},"level":{"cfgSlowChgWatts":{"min":200,"max":2900}}},"mppt":{"level":{"cfgDcChgCurrent":{"min":4,"max":13}},"number":{"inWatts":{"mult":0.1,"max":1650},"outWatts":{"mult":0.1,"max":1600},"inVol":{"max":150},"mpptTemp":{"max":100},"outAmp":{"max":30},"inAmp":{"max":16}},"diagnostic":{"xt60ChgType":{"xt60ChgType":{"0":"not detected","1":"adapter","2":"MPPT"}}}}}
2024-08-17 23:15:16.307 - debug: ecoflow-mqtt.0 (1968) manipulate: bmsMaster/amp old--new 25 -- 50
2024-08-17 23:15:16.308 - debug: ecoflow-mqtt.0 (1968) manipulate: inv/cfgFastChgWatts old--new 200 -- 200
2024-08-17 23:15:16.308 - debug: ecoflow-mqtt.0 (1968) manipulate: inv/cfgFastChgWatts old--new 2400 -- 2900
2024-08-17 23:15:16.308 - debug: ecoflow-mqtt.0 (1968) manipulate: inv/cfgSlowChgWatts old--new 200 -- 200
2024-08-17 23:15:16.309 - debug: ecoflow-mqtt.0 (1968) manipulate: inv/cfgSlowChgWatts old--new 1000 -- 2900
2024-08-17 23:15:16.309 - debug: ecoflow-mqtt.0 (1968) manipulate: mppt/cfgDcChgCurrent old--new 4 -- 4
2024-08-17 23:15:16.310 - debug: ecoflow-mqtt.0 (1968) manipulate: mppt/cfgDcChgCurrent old--new 8 -- 13
2024-08-17 23:15:16.310 - debug: ecoflow-mqtt.0 (1968) manipulate: mppt/inWatts old--new 0.1 -- 0.1
2024-08-17 23:15:16.310 - debug: ecoflow-mqtt.0 (1968) manipulate: mppt/inWatts old--new 500 -- 1650
2024-08-17 23:15:16.311 - debug: ecoflow-mqtt.0 (1968) manipulate: mppt/outWatts old--new 0.1 -- 0.1
2024-08-17 23:15:16.311 - debug: ecoflow-mqtt.0 (1968) manipulate: mppt/outWatts old--new 500 -- 1600
2024-08-17 23:15:16.311 - debug: ecoflow-mqtt.0 (1968) manipulate: mppt/inVol old--new 150 -- 150
2024-08-17 23:15:16.312 - debug: ecoflow-mqtt.0 (1968) manipulate: mppt/mpptTemp old--new 80 -- 100
2024-08-17 23:15:16.312 - debug: ecoflow-mqtt.0 (1968) manipulate: mppt/outAmp old--new 13 -- 30
2024-08-17 23:15:16.312 - debug: ecoflow-mqtt.0 (1968) manipulate: mppt/inAmp old--new 13 -- 16
2024-08-17 23:15:16.312 - debug: ecoflow-mqtt.0 (1968) manipulate: mppt/xt60ChgType old--new [object Object] -- [object Object]
[... cut ...]
2024-08-17 23:17:39.654 - debug: ecoflow-mqtt.0 (1968) [Compare] SP10XXXXXXXXX not processed heartbeat/heartbeat state: infoList value [object Object],[object Object],[object Object],[object Object],[object Object],[object Object],[object Object],[object Object],[object Object],[object Object],[object Object],[object Object]
2024-08-17 23:17:44.502 - debug: ecoflow-mqtt.0 (1968) [Compare] SP10XXXXXXXXX not processed heartbeat/heartbeat state: infoList value [object Object],[object Object],[object Object],[object Object],[object Object],[object Object],[object Object],[object Object],[object Object],[object Object],[object Object],[object Object]

foxthefox, 2024-08-18:
Thanks, for the log.
The log itself is now showing the debug messages, very good.
Looking at the log, it seems that there are only commands captured which were given from the iobroker/HA.

Can you post again a log from the timeframe where you change settings through the Ecoflow App only?

thyr0ne, 2024-08-19:
I've attached the log from my existing EcoFlow SHP integration in HA:

[https://github.com/thyr0ne/hassio-ecoflow-cloud/blob/main/custom_components/ecoflow_cloud/devices/smart_home_panel.py](https://github.com/thyr0ne/hassio-ecoflow-cloud/blob/main/custom_components/ecoflow_cloud/devices/smart_home_panel.py)

LOG: 
2024-08-19 10:55:07.028 DEBUG (MainThread) [custom_components.ecoflow_cloud.devices.smart_home_panel] ModeDictSelectEntity._update_value = {'sta': 0, 'ctrlMode': 1}
2024-08-19 10:55:07.028 DEBUG (MainThread) [custom_components.ecoflow_cloud.devices.smart_home_panel] ModeDictSelectEntity._update_value = {'powCh': 1, 'ctrlSta': 0, 'ctrlMode': 0, 'priority': 1}
2024-08-19 10:55:07.028 DEBUG (MainThread) [custom_components.ecoflow_cloud.devices.smart_home_panel] ModeDictSelectEntity._update_value = {'powCh': 1, 'ctrlSta': 0, 'ctrlMode': 0, 'priority': 2}
2024-08-19 10:55:07.028 DEBUG (MainThread) [custom_components.ecoflow_cloud.devices.smart_home_panel] ModeDictSelectEntity._update_value = {'powCh': 1, 'ctrlSta': 0, 'ctrlMode': 0, 'priority': 3}
2024-08-19 10:55:07.028 DEBUG (MainThread) [custom_components.ecoflow_cloud.devices.smart_home_panel] ModeDictSelectEntity._update_value = {'powCh': 1, 'ctrlSta': 0, 'ctrlMode': 0, 'priority': 4}
2024-08-19 10:55:07.028 DEBUG (MainThread) [custom_components.ecoflow_cloud.devices.smart_home_panel] ModeDictSelectEntity._update_value = {'powCh': 0, 'ctrlSta': 0, 'ctrlMode': 0, 'priority': 5}
2024-08-19 10:55:07.029 DEBUG (MainThread) [custom_components.ecoflow_cloud.devices.smart_home_panel] ModeDictSelectEntity._update_value = {'powCh': 1, 'ctrlSta': 0, 'ctrlMode': 0, 'priority': 6}
2024-08-19 10:55:07.029 DEBUG (MainThread) [custom_components.ecoflow_cloud.devices.smart_home_panel] ModeDictSelectEntity._update_value = {'powCh': 0, 'ctrlSta': 0, 'ctrlMode': 0, 'priority': 7}
2024-08-19 10:55:07.029 DEBUG (MainThread) [custom_components.ecoflow_cloud.devices.smart_home_panel] ModeDictSelectEntity._update_value = {'powCh': 0, 'ctrlSta': 0, 'ctrlMode': 0, 'priority': 8}
2024-08-19 10:55:07.029 DEBUG (MainThread) [custom_components.ecoflow_cloud.devices.smart_home_panel] ModeDictSelectEntity._update_value = {'powCh': 0, 'ctrlSta': 0, 'ctrlMode': 0, 'priority': 9}
2024-08-19 10:55:07.029 DEBUG (MainThread) [custom_components.ecoflow_cloud.switch] Updating switch ecoflow-SP10XXXXXXXXXXXX-energyInfos-0--stateBean-isGridCharge to 0
2024-08-19 10:55:07.029 DEBUG (MainThread) [custom_components.ecoflow_cloud.switch] Updating switch ecoflow-SP10XXXXXXXXXXXX-energyInfos-1--stateBean-isGridCharge to 0
2024-08-19 10:55:07.030 DEBUG (MainThread) [custom_components.ecoflow_cloud.mqtt.ecoflow_mqtt] Sending {"from": "HomeAssistant", "id": "999922850", "version": "1.0", "moduleType": 0, "operateType": "TCP", "params": {"sta": 0, "ctrlMode": 1, "ch": 0, "cmdSet": 11, "id": 16}} :(0, 39)(False)
2024-08-19 10:55:11.715 DEBUG (MainThread) [custom_components.ecoflow_cloud.devices.smart_home_panel] ModeDictSelectEntity._update_value = {'sta': 1, 'ctrlMode': 1}
2024-08-19 10:55:11.715 DEBUG (MainThread) [custom_components.ecoflow_cloud.devices.smart_home_panel] ModeDictSelectEntity._update_value = {'powCh': 1, 'ctrlSta': 0, 'ctrlMode': 0, 'priority': 1}
2024-08-19 10:55:11.715 DEBUG (MainThread) [custom_components.ecoflow_cloud.devices.smart_home_panel] ModeDictSelectEntity._update_value = {'powCh': 1, 'ctrlSta': 0, 'ctrlMode': 0, 'priority': 2}
2024-08-19 10:55:11.715 DEBUG (MainThread) [custom_components.ecoflow_cloud.devices.smart_home_panel] ModeDictSelectEntity._update_value = {'powCh': 1, 'ctrlSta': 0, 'ctrlMode': 0, 'priority': 3}
2024-08-19 10:55:11.715 DEBUG (MainThread) [custom_components.ecoflow_cloud.devices.smart_home_panel] ModeDictSelectEntity._update_value = {'powCh': 1, 'ctrlSta': 0, 'ctrlMode': 0, 'priority': 4}
2024-08-19 10:55:11.715 DEBUG (MainThread) [custom_components.ecoflow_cloud.devices.smart_home_panel] ModeDictSelectEntity._update_value = {'powCh': 0, 'ctrlSta': 0, 'ctrlMode': 0, 'priority': 5}
2024-08-19 10:55:11.715 DEBUG (MainThread) [custom_components.ecoflow_cloud.devices.smart_home_panel] ModeDictSelectEntity._update_value = {'powCh': 1, 'ctrlSta': 0, 'ctrlMode': 0, 'priority': 6}
2024-08-19 10:55:11.716 DEBUG (MainThread) [custom_components.ecoflow_cloud.devices.smart_home_panel] ModeDictSelectEntity._update_value = {'powCh': 0, 'ctrlSta': 0, 'ctrlMode': 0, 'priority': 7}
2024-08-19 10:55:11.716 DEBUG (MainThread) [custom_components.ecoflow_cloud.devices.smart_home_panel] ModeDictSelectEntity._update_value = {'powCh': 0, 'ctrlSta': 0, 'ctrlMode': 0, 'priority': 8}
2024-08-19 10:55:11.716 DEBUG (MainThread) [custom_components.ecoflow_cloud.devices.smart_home_panel] ModeDictSelectEntity._update_value = {'powCh': 0, 'ctrlSta': 0, 'ctrlMode': 0, 'priority': 9}
2024-08-19 10:55:11.716 DEBUG (MainThread) [custom_components.ecoflow_cloud.switch] Updating switch ecoflow-SP10XXXXXXXXXXXX-energyInfos-0--stateBean-isGridCharge to 0
2024-08-19 10:55:11.716 DEBUG (MainThread) [custom_components.ecoflow_cloud.switch] Updating switch ecoflow-SP10XXXXXXXXXXXX-energyInfos-1--stateBean-isGridCharge to 0
2024-08-19 10:55:11.716 DEBUG (MainThread) [custom_components.ecoflow_cloud.mqtt.ecoflow_mqtt] Sending {"from": "HomeAssistant", "id": "999922851", "version": "1.0", "moduleType": 0, "operateType": "TCP", "params": {"sta": 1, "ctrlMode": 1, "ch": 0, "cmdSet": 11, "id": 16}} :(0, 40)(False)
[... cut ...]
2024-08-19 10:55:30.514 DEBUG (Thread-6 (_thread_main)) [custom_components.ecoflow_cloud.devices.smart_home_panel] ModeDictSelectEntity._update_value = {'powCh': 0, 'ctrlSta': 0, 'ctrlMode': 0, 'priority': 9}
2024-08-19 10:55:30.515 DEBUG (Thread-6 (_thread_main)) [custom_components.ecoflow_cloud.switch] Updating switch ecoflow-SP10XXXXXXXXXXXX-energyInfos-0--stateBean-isGridCharge to 0
2024-08-19 10:55:30.515 DEBUG (Thread-6 (_thread_main)) [custom_components.ecoflow_cloud.switch] Updating switch ecoflow-SP10XXXXXXXXXXXX-energyInfos-1--stateBean-isGridCharge to 0

foxthefox, 2024-08-20:
Thank you, I assume that the commands given by the HA integration are working and the log does contain the cmd from HA.
I am not familiar with the HA integration and their logging, but I found lines  "[custom_components.ecoflow_cloud.mqtt.ecoflow_mqtt] Sending {"from": "HomeAssistant"

Comparing this with the IOB adapter, it seems that "moduleSn" is excessive and "moduleType" is missing.
I will change the code in such a way.

Anyhow having the command captured, which is send by the APP would be the best source for comparison. 
I would appreciate if you have the chance to post a IOB log showing commands from the APP (because the adapter subscribes to all topics of the device, the adapter will also see the commands from the APP)

foxthefox, 2024-08-21:
Please check version 0.0.42 from git or npm and report.

thyr0ne, 2024-08-21:
Thanks for the quick release.

I've tested it and:
* the toggle for manual and auto is still the opposite - manual = auto and the other way around
* when I'm switching from auto to manual in the EcoFlow app  and try to change - still it turns it off
* the other items in the dropdown don't change anything. 
* How can I log the EcoFlow app on an iPhone? Happy to send you something but don't know how

foxthefox, 2024-08-21:
There is no need to log something on the iPhone.
The EF App is communicating with the same MQTT server as the adapter. Both clients are publishing on the same topic and have the same subscriptions. That enables each client to see what the other does.
So when commands are given through the EF App then the adapter will have entries in the log.

It is only needed to do some commands in the EF App and then extract the part of the adapter log and post it.
With a small hint which commands where given through the EF App I can check for the syntax.

So 'Contol mode xyz 0-Auto/1-Manual' is then 'Contol mode xyz 1-Auto/0-Manual' correct?
With drop down you mean 'Grid supply', 'storage power supply', 'close' ?

thyr0ne, 2024-08-22:
Thanks again!
Did not expect that the adapter will log the message from the app too. I did now all the changes with the app and attached it. 

> So 'Contol mode xyz 0-Auto/1-Manual' is then 'Contol mode xyz 1-Auto/0-Manual' correct?
With drop down you mean 'Grid supply', 'storage power supply', 'close' ?
correct it's the opposite and yes that is what I mean ;)

Logs:

change from "auto" to "manual":
	2024-08-22 11:01:45.864	debug	[JSON] SP10XXXX [set_reply] -> set w/o params{"id":305313097,"version":"1.0","operateType":"TCP","code":"0","data":{"sta":0,"cmdSet":11,"ack":0,"id":16}}
ecoflow-mqtt.0
2024-08-22 11:01:45.482	debug	[JSON] SP10XXXX [set] -> key:16 {"from":"iOS","operateType":"TCP","id":"305313097","lang":"de-de","params":{"ch":0,"cmdSet":11,"sta":0,"ctrlMode":1,"id":16},"version":"1.0"}

change from "grid supply" to "storage power supply":
	2024-08-22 11:03:56.029	info	state ecoflow-mqtt.0. SP10XXXX.loadCmdChCtrlInfos.ctrlSta_0 changed: 1 (ack = true)

ecoflow-mqtt.0
2024-08-22 11:03:55.632	debug	[JSON] SP10XXXX [set_reply] -> set w/o params{"id":434962920,"version":"1.0","operateType":"TCP","code":"0","data":{"sta":0,"cmdSet":11,"ack":0,"id":16}}

ecoflow-mqtt.0
2024-08-22 11:03:55.126	debug	[JSON] SP10XXXX [set] -> key:16 {"from":"iOS","operateType":"TCP","id":"434962920","lang":"de-de","params":{"ch":0,"id":16,"cmdSet":11,"ctrlMode":1,"sta":1},"version":"1.0"}

change from "storage power..." to "off":

ecoflow-mqtt.0
2024-08-22 11:05:17.360	debug	[JSON] SP10XXXX [set_reply] -> set w/o params{"id":516875290,"version":"1.0","operateType":"TCP","code":"0","data":{"sta":0,"cmdSet":11,"ack":0,"id":16}}

ecoflow-mqtt.0
2024-08-22 11:05:17.202	debug	[Compare] SP10XXXX not processed heartbeat/heartbeat state: infoList value [object Object],[object Object],[object Object],[object Object],[object Object],[object Object],[object Object],[object Object],[object Object],[object Object],[object Object],[object Object]

ecoflow-mqtt.0
2024-08-22 11:05:17.053	debug	[JSON] SP10XXXX [set] -> key:16 {"from":"iOS","operateType":"TCP","id":"516875290","lang":"de-de","params":{"ch":0,"ctrlMode":1,"id":16,"sta":2,"cmdSet":11},"version":"1.0"}

foxthefox, 2024-08-27:
Please test the version 1.0.2 from git or npm and report.

thyr0ne, 2024-08-29:
Hi [foxthefox](https://github.com/foxthefox),

I've tested v1.0.2 and we're getting closer. Thanks a lot again.

What works:
- SP10ZEW5ZEA40017_loadCmdChCtrlInfos Contol mode channel #XX 0-Auto/1-Manual/Grid 
- "off" -> Auto
- "on" -> Manual

- when I change the SP10ZEW5ZEA40017_loadCmdChCtrlInfos Power supply type channel #XX Grid supply, 1 storage power supply, 2 close/off
- SP10ZEW5ZEA40017_loadCmdChCtrlInfos Contol mode channel #XX 0-Auto/1-Manual/Grid  will set to "on" -> Manual

What doesn't work:
- when I try to select "Grid supply", "1 storage power supply" or "2 close/off" nothing happens - it stick to "Grid supply"

What do you need to help you?

foxthefox, 2024-09-02:
I tested a little bit and it seems OK, for me.

Toggle channel 7 to storage power (1)
should give "sta":1,"ctrlMode":1,"ch":7,", and it does
```
ecoflow-mqtt.0
2024-09-02 21:52:42.806	debug	[JSON] SHP10ZFB5EF412345 [set] -> key:16 {"from":"Android","lang":"en-us","id":"150581466","moduleType":0,"operateType":"TCP","version":"1.0","params":{"sta":1,"ctrlMode":1,"ch":7,"cmdSet":11,"id":16}}

ecoflow-mqtt.0
2024-09-02 21:52:42.805	debug	Message succesfully published.SHP10ZFB5EF412345 ../..set

ecoflow-mqtt.0
2024-09-02 21:52:42.805	debug	publish {"from":"Android","lang":"en-us","id":"150581466","moduleType":0,"operateType":"TCP","version":"1.0","params":{"sta":1,"ctrlMode":1,"ch":7,"cmdSet":11,"id":16}}

ecoflow-mqtt.0
2024-09-02 21:52:42.805	debug	publish /app/testuser1234testuser1234/SHP10ZFB5EF412345/thing/property/set

ecoflow-mqtt.0
2024-09-02 21:52:42.805	debug	prep cmd SHP10ZFB5EF412345 panel ctrlSta 1 {"from":"Android","lang":"en-us","id":"150581466","moduleType":0,"operateType":"TCP","version":"1.0","params":{"sta":1,"ctrlMode":1,"ch":7,"cmdSet":11,"id":16}}

...

ecoflow-mqtt.0
2024-09-02 21:52:42.805	info	state ecoflow-mqtt.0.SHP10ZFB5EF412345.loadCmdChCtrlInfos.ctrlSta_7 changed: 1 (ack = false)
```

Toggle channel 7 to off (2)
should give "sta":2,"ctrlMode":1,"ch":7,", and it does

```
coflow-mqtt.0
2024-09-02 21:52:42.806	debug	[JSON] SHP10ZFB5EF412345 [set] -> key:16 {"from":"Android","lang":"en-us","id":"150581466","moduleType":0,"operateType":"TCP","version":"1.0","params":{"sta":1,"ctrlMode":1,"ch":7,"cmdSet":11,"id":16}}

ecoflow-mqtt.0
2024-09-02 21:52:42.805	debug	Message succesfully published.SHP10ZFB5EF412345 ../..set

ecoflow-mqtt.0
2024-09-02 21:52:42.805	debug	publish {"from":"Android","lang":"en-us","id":"150581466","moduleType":0,"operateType":"TCP","version":"1.0","params":{"sta":1,"ctrlMode":1,"ch":7,"cmdSet":11,"id":16}}

ecoflow-mqtt.0
2024-09-02 21:52:42.805	debug	publish /app/testuser1234testuser1234/SHP10ZFB5EF412345/thing/property/set

ecoflow-mqtt.0
2024-09-02 21:52:42.805	debug	prep cmd SHP10ZFB5EF412345 panel ctrlSta 1 {"from":"Android","lang":"en-us","id":"150581466","moduleType":0,"operateType":"TCP","version":"1.0","params":{"sta":1,"ctrlMode":1,"ch":7,"cmdSet":11,"id":16}}

...

ecoflow-mqtt.0
2024-09-02 21:52:42.805	info	state ecoflow-mqtt.0.SHP10ZFB5EF412345.loadCmdChCtrlInfos.ctrlSta_7 changed: 1 (ack = false)
```

Can you crosscheck with your debug log, if there is same "[JSON] SHP10xxxx [set]" message line?

Are you using the HA-gateway?

thyr0ne, 2024-09-18:
Hi,
sorry for my late response. You are right - the smart panel can be read and configured and is working within ioBroker. 
Yes I'm using HA and MQTT to control everything. 

Is there a way to check that please?

foxthefox, 2024-09-20:
I think my question is answered.

as asked above, can you crosscheck the debug log concerning  "sta“,"ctrlMode" when you change the mode. Since from my perspective it is doing right, see above

closed 2024-09-20 by thyr0ne