foxthefox/ioBroker.ecoflow-mqtt #137: 'Backup reserve' option missing on D2M

walderston, 2024-08-26:
Using ecoflow-mqtt via HA and noticed the 'Backup reserve' switch / option is not shown. If I use the hassio-ecoflow-cloud via HACS the option is there.

What do you need from me to look at getting this added?

foxthefox, 2024-08-26:
I am not sure what is the data point behind "backup reserve"
Do you mean pd.newAcAutoOnCfg or pd.bpPowerSoc ?
Both are available and can be found in the iobroker object tree and should also be shown in HA.

If you refer to something else, I need more information.

foxthefox, 2024-08-27:
pd.watchIsConfig is available in the adapter as "diagnostic" data.
I did not know that this is a switching datapoint.

Can you please enable the debug mode in the adapter and additionally 
- the debug of the D2M in the configuration line (tick in box)
- the msgSet/Get (tick in box)

After debug mode is working, please perform this command in the APP and post the log file with some pre- and following lines.

foxthefox, 2024-09-02:
Unfortunately the log does not show the command from the APP. it only shows some unidentified values, which might come from a newer FW.
I think before of repeating the test, please check the enable status of the device and and general debug settings.
I have some exemplary screenshots:

[image]

[image]

[image]

If these checkmarks are set, then the log should have more info inside (and will also report the missing datapoints) and should also have the command given by the APP

walderston, 2024-09-02:
ah sorry, didn't realise I had to enable debug in the device config too.

Hopefully I timed that right as its very chatty - I filtered on 0077 being the last digits of the serial number to reduce the amount of chatter. 

[off.log](https://github.com/user-attachments/files/16839389/off.log)
[on.log](https://github.com/user-attachments/files/16839388/on.log)

foxthefox, 2024-09-02:
Thanks for the log.
There are new data structures inside, which fail to be processed. This is handled, but some important data updates are not visible. 
I will prepare a quick fix for that tomorrow.
After upgrade unfortunately the test needs to becrepeated.
I’ll keep you informed when I am ready.

ps.
There are a lot of new datapoints inside the telegram, which need to be defined. That comes later.

foxthefox, 2024-09-03:
I have not finished the next version 1.0.3, but the branch is uploaded.
You can already try to create a new on.log and off.log with that version.

You need to go to adapters-> octacat ->custum and type the url as in the screenshot

[image]

walderston, 2024-09-03:
updated and attached: 
[backup off.log](https://github.com/user-attachments/files/16846168/backup.off.log)
[backup on.log](https://github.com/user-attachments/files/16846169/backup.on.log)

foxthefox, 2024-09-03:
Thanks a lot, now it is clear how it is arranged.
I will update accordingly.

foxthefox, 2024-09-03:
You could have a try, once again with the "tree" version (still 1.0.3)
You have to delete the datapoint pd.watchIsConfig and pd.bpPowerSoc  or the whole pd-tree and restart the adapter.
That creates the new datapoints including their new behaviour.

pd.watchIsConfig is disabling/anabling the backup mode
pd.bpPowerSoc is changing the value and when changing the value it is enabling the mode (not sure if the APP does it in the same way)

walderston, 2024-09-04:
seems to be working as expected. thanks very much!

foxthefox, 2024-09-04:
OK, I will make it official.

Can you check if the issue with SlowChangeWatt is still there, it might be a result of the not expected telegrams.

walderston, 2024-09-05:
yup thats not updating in HA when I update the value in HA. It does update the value within the app itself.

walderston, 2024-09-05:
hopefully this has the right debugging on.

[SlowChangeWatt.txt](https://github.com/user-attachments/files/16898213/SlowChangeWatt.txt)

Basically, change the SlowChgWatts value in HA to 1800, that value it sent to EF however nothing is sent back to HA to set the new value to 1800

So when I refresh the HA page, it contains the old value.

foxthefox, 2024-09-05:
Very good, you discovered the tick for HA debugging.
What I am missing in this log is the [set_reply] message.

May be there is some behaviour which I do not anticipate.
It could be the case that the HA always needs a response to the changed value and therefore it is not memorizing the given command.
In iobroker a msg could contain the Ack=false and true to distinguish betwenn command and feedback, but a given command is stored and not artificially reset (due to missing acknowledge).

I am so far not aware of a bit in HA to give feedback/acknowledgment or the necessity of such.

The adapter is only sending updated values to HA if there is a change to the current value or a complete update after adapter start.

foxthefox, 2024-09-05:
I could also see it at my implementation, after the page refresh the value is back to the previous one.

walderston, 2024-09-05:
going to assume its the same thing as permanentWatts that I mentioned on the last comment of #143 

not tested the other values but I'd imagine its the same behaviour that HA keeps the old value rather than the newly set value so when you refresh in HA, its now out of sync until either ioBroker adapter is restarted or some other unknown event.

foxthefox, 2024-09-05:
I would assume it innige same way. But still it is strange for me that a command is not persistent stored. At least it is processed as such and was going through the different processes. 🤔

foxthefox, 2024-09-05:
Maybe we should continue in the other issue, hence this one is solved.

closed 2024-09-09 by foxthefox