foxthefox/ioBroker.ecoflow-mqtt #140: Inverter power change for Powerstream does not propagate to device

DJFliX, 2024-08-27:
First off: thanks for your work on this project! It has helped me a lot in automating charging/discharging but also monitoring of my Ecoflow Powerstream 800/Delta 2 Max EB combo!

**Describe the bug**  
It looks like I'm not able to change the inverter power from Home Assistant. However changes from the EcoFlow app do seem to propagate back to Home Assistant.

**To Reproduce**  
Steps to reproduce the behavior:
1. Enable debug logging in iobroker for home assistant input and output
2. Modify inverter wattage from Home Assistant from 140 to 100
3. Check battery screen, HA battery output to see if the wattage decreases from a bit over 140 to a bit over 100
4. See that the wattage stays at a bit over 140

**Expected behavior**  
The value of 100 is propagated to the device and after a couple of seconds the battery shows that it is discharging at approx 100 watts

**Screenshots & Logfiles**  
```
HA msg: iob_ef/HW51ZEH4SF5R5062/set/inverter_heartbeat/permanentWatts 100 {"type":"Buffer","data":[49,48,48]}

ecoflow-mqtt.0
2024-08-27 20:31:31.409	debug	[HA] pstream800: processing HW51ZEH4SF5R5062 inverter_heartbeat permanentWatts value : 100

ecoflow-mqtt.0
2024-08-27 20:31:31.410	debug	[HA] sending cmd value 100 to HW51ZEH4SF5R5062.inverter_heartbeat.permanentWatts
```

**Versions:**  
 - Adapter version: 1.0.0
 - JS-Controller version: 7.0.23
 - Node version: 20.17.0
 - Operating system: Raspbian 64-bit

**Additional context**  
I have the feeling that this worked on 0.3.0 and stopped working after upgrading to 1.0.0.

foxthefox, 2024-08-27:
Thank you for the report.

With version 1.0.0 I have used some different state roles with additional context like level.power, that caused not recognizing the change.

With version 1.0.1 from git or npm it should work again.
Please try 1.0.1 and report.

DJFliX, 2024-08-30:
Amazing, thanks for the quick fix! I can confirm that setting the inverter wattage now works as it used to 🙌

closed 2024-08-30 by DJFliX