foxthefox/ioBroker.ecoflow-mqtt #143: D2M - SlowChgWatts not being displayed correctly in HA

walderston, 2024-08-29:
using v1.0.2 with HA. I'm able to set the SlowChgWatts however the value is not stored within HA so when I revisit the page it retains the old previous value. If I make the change via the app, the correct value is displayed in HA

foxthefox, 2024-09-02:
I have tested several thing on my DM and could not reproduce the problem.
Can you do another check and change the value from the iobroker object list and verify the change in the APP, it could be the case that the given command is not having the right information in it (similiar to one issue with r2max).
If this is the case then I would needt to have a debug log (device  and msgSet/Get, msgCMD enabled) and changes via APP performed. Then I can see the original command string. 

PS.
What device is concerned?

walderston, 2024-09-02:
Its a Delta 2 Max

It is able to set the charging speed correctly, however it doesn't seem to update the value within HA so it continues to display the old value. This only happens when the value is set via HA, if I set the value via the app, HA updates correctly. 

Noticed permanentWatts on the powerstream (inverter_hearbeat) also sometimes seems to stick with the old value. 

Shows that permanentWatts value is at 370 despite it being set to 380. The app correctly shows 380. 
[image]
[image]

walderston, 2024-09-06:
@foxthefox happy to continue on this thread as you rightly said, the other issue has been resolved.

walderston, 2024-09-10:
@foxthefox do you think this is a HA issue for not persisting the state or is it expected behaviour and we should get ioBroker to send an update?

foxthefox, 2025-04-06:
I did some changes related to HA data transfer.
Please have a look to the final 1.3.1 at git or npm.
Hopefully everythink is working properly.

walderston, 2025-04-07:
this hasn't made any difference

[image]

via HA, I changed the value (900) however the value still retains its old value.

walderston, 2025-04-07:
also its working on the powerstream with permanentWatts - assume the 2 ish second heartbeat is the trick?

[image]

foxthefox, 2025-04-07:
Thanks for the feedback, I will search further.

foxthefox, 2025-04-08:
now I think that I nailed it with final version 1.3.2 (on git and npm)

walderston, 2025-04-08:
Thanks, I'll give it a test shortly and report back

foxthefox, 2025-04-13:
@walderston 
Did you have a chance for checking?

walderston, 2025-04-15:
Apologies I'll do it this evening

walderston, 2025-04-15:
@foxthefox nope hasn't worked.

In HA, I changed the value for number.r35XXX0077_inv_maximum_charging_power_for_ac_slow_charging_w
I can see it being set in MQTT and the EF app updates however when I refresh HA page, its back to its old value

[image]

foxthefox, 2025-04-15:
Thanks for the testing.
Then for now I have no idea what could be the reason. Through other issues it turned out that some confirmation telegram was needed for the sensors of mqtt. But here it is obviously not enough.

walderston, 2025-04-16:
If I was to have a guess at the issue, the D2M (and assume other Delta's doesn't have a 'heartbeat' which is setting the values. 

On the powerstream, there is a heartbeat:
> also its working on the powerstream with permanentWatts - assume the 2 ish second heartbeat is the trick?
> 
> [image]

tbh, its not a massive issue, I've got a workaround by using a helper, setting the value of the helper which then sets the value of SlowChgWatts and I just read the value from the helper instead

foxthefox, 2025-04-17:
The functionality in HA is sometimes more hidden than clearly described. For the switch connecting to mqtt entity there was a toggling, which was caused by an internal algorithm to wait x ms for an confirmation telegram (the external feedback from the command). Then similar problem was discovered at numbers, where adjusted value was not populated within the HA system, there seems also a confirmation needed.
For both cases the iobroker.adapter immediately sends back the received setting in order to give a "pre-confirmation" since the long way HA->IOB->EF->DEVICE->EF->IOB->HA is most likely to long.
Usually the devices respond with the next update telegram with the changed value inside.

I checked this behavior of number population and before of my "pre-confirmation" the gauge was not following the slider, after it is following perfectly.
[image]