foxthefox/ioBroker.ecoflow-mqtt #152: ECOFlow Powerocean DCFIT | Use Adapter to run DC-Fit without Smart Meter

stevoboy, 2024-09-19:
Dear all

I am thinking of buying a DC Fit to add to my existing Solar System.
Is it possible to read values from the DC Fit and also write Start Charge, Stop Charge, Start Discharge, Stop Discharge all with target values. 
Do you think it is possible to run the DC Fit without the Smart Meter when control is executed by this MQTT Adapter? I would like to avoid the expensive Eletrician Installation of the Smart Meter and I have a SMA Sunny Home Manager already that has all the values needed.

Thanks a lot for your help

foxthefox, 2024-09-20:
I can hardly make suggestions, but if you think it fits to you after thorough investigation of the features, then it might be an option. Please consider the following
- the power ocean is not part of the official API, so there is not much information around here
- The power ocean is using protobuf decoding, where many things are already correctly guessed and implemented, but it’s not complete yet
- the commands from the app can be reproduced by the adapter, after some investigation
- The adapter uses the unofficial API(same as the App), which might become unavailable 
- But if we see data and can interpret data, then it can be implemented in the adapter

stevoboy, 2024-09-21:
Thanks a lot.
I spoke with an installer and he mentioned I need to have an electrician work with his app to install it. That makes it really unattractive cost wise.
I check now another battery like sax that is maybe cheaper to install and maybe ac based. That would allow storing net energy as well.

closed 2024-10-01 by foxthefox