foxthefox/ioBroker.ecoflow-mqtt #153: Cannot add this as a custom repository - Home assistant says that it is not a compliant directory structure

Stuggy, 2024-09-27:
I go into HACS and then try to add a custom repository: https://github.com/foxthefox/ioBroker.ecoflow-mqtt

I then says the repository structure for main is not compliant.

Am I doing something wrong?

foxthefox, 2024-09-27:
The adapter is not meant to be directly used within HA.
You need an instance of iobroker running where you can install this adapter. Then you can use the gateway functionality to have the EF integrated into HA.

Stuggy, 2024-09-27:
ok I thought it might be something like that.  I have an Ubuntu server.  I need to see if there are different versions of iobroker and take it from there.

I may ask some more questions.

thanks

foxthefox, 2024-09-28:
If you have Ubuntu running, the installation should be straight forward.
Yes, please ask.

Stuggy, 2024-09-28:
Ok I got iobroker installed as a container - pretty easy.
I worked out how to add your repository and it installed v1.04.
The readme link in iobroker is broken for your repository.
I'm struggling to find the configuration screen for you ecoflow app in iobroker (first time using iobroker)

Stuggy, 2024-09-28:
I had to click on add instance - now making more progress.  Hopefully this might be useful for someone later.

Stuggy, 2024-09-29:
Almost the last step I think.  I can't successfully add the mqtt broker to home assistant even though I am providing the iobroker IP and port as well as the user and pw set in the home assistant tab e.g. hamqttuser and pw.

The only other thing is that I am doing all this in portainer but I think the ports look good.

Stuggy, 2024-09-30:
I still can't connect between containers or maybe it is the login info.  Marking this as closed as a I try and work it out.

foxthefox, 2024-10-01:
Unfortunately I am not at home and can not post some screenshots. Basically you should use port 1883 and use type in the login credentials of HA in the adapter.

foxthefox, 2024-10-01:
Have also a look into readme
https://github.com/foxthefox/ioBroker.ecoflow-mqtt?tab=readme-ov-file#ha-connectorgateway

You may need to activate autodiscovery.

Stuggy, 2024-10-01:
Thanks I'll take a look,  I tried the hassio cloud integration yesterday and realised that it only updated when I had the app open on my phone and I guess that triggered data.

So no solution yet - I'll read your link.