foxthefox/ioBroker.ecoflow-mqtt #169: Is it possible to support EcoFlow 800W Alternator Charger?

evolvo187, 2024-11-12:
Is it possible to support EcoFlow 800W Alternator Charger?

foxthefox, 2024-11-17:
In principle yes, but it requires some reverse engineering on the telegrams which are transmitted.
A good basis would be most likely the powerstream. But I would guess that there is another telegram structure.

If you have the iobroker adapter running, then I think that you might add a powerstream with the serial# of the charger.
That would not decode something but it should bring up some messages in the log.
The adapter needs to be in debug mode and on first tab you have to tick the "msgUpdate" + "msgSetGet" + "msgCmd".
On the second tab please only make a tick at debug enable for the charger(powerstream).

Then the log should bring up some messages (coded as protobuf). If you could post these messages the I could have a look to it.

foxthefox, 2024-11-27:
Ja, hilft. sind auf jeden Fall die Daten drin, die ich brauch.

foxthefox, 2024-12-05:
Ich habe mal begonnen die enthaltenen Werte einem Namen zu geben. Leider ist in den zyklischen Daten nicht allzuviel drin.
Eine erste Interpretation habe ich kommentiert, evtl. gibt es ja etwas was palusibel erscheint.

  alternatorHeartbeat: {
    unknown1: 32, 
    unknown102: 9, // status?
    unknown105: 0, // switch mode?
    unknown130: 1, // switch mode ?
    unknwon138: 130,
    batVolt139: 12.412946701049805, //car bat voltage?
    unknown262: 32, 
    unknow268: 7573, //temp? 75°C
    unknwon269: 5048, //temp? 50°C
    unknown425: 0, //status/switch?
    unknwon427: 65535, // error?
    unknwon428: 65535, //error?
    unknwon581: 1, //status/switch?
    unknown597: 1, //status/switch?
    power598: 800, // power to load power station?
    unknown602: -45, //current?
    power603: 800 //power to load car?
  }

Es wäre noch ganz gut zusätzlich bei msgSet/Get ein Häckchen zu setzen und dann von der App ein paar Änderungen vorzunehmen.
Jeder Befehl aus der App ist wegen MQTT auch im Adapterlog sichtbar. 
Somit kann man die Befehle ermitteln und im nachfolgenden Telegramm die Änderungen an den Werten sehen.
Dann wäre wieder das log (Ausschnitte) zu posten und wann welche Befehle gegeben wurden. Im log steht dazu "[set]", "[set_reply]", "[get]" und das danach folgende "[update]"

evolvo187, 2024-12-10:
hier das Protokoll.

Hoffe du kannst es ansehen und hilft wieder weiter.

[Batterieladegerät.txt](https://github.com/user-attachments/files/18080625/Batterieladegerat.txt)

evolvo187, 2024-12-10:
Hier noch Bilder von der App.

[image]
[image]

Maximale Ausgangsleistung 0 - 800 Watt
Startspannung 11 - 30 V

foxthefox, 2024-12-10:
Danke dir. Ich schau es mir am Wochenende an.

foxthefox, 2024-12-15:
Das Log hat geholfen.

Die Befehle sind so codiert:

message alternatorSet {
	optional int32 operationMode = 116; // 1=Charge; 2=reverse charge; 3=battery maintenance;
	optional int32 startStop = 122; // 0=off; 1= on
	optional float powerLimit = 123; // 0 - 800 Watt
	optional int32 startVoltage = 137; // 11 - 30 V *10
}

Die Kennungen sind nicht die gleiche wie die aus dem Update, aber diese ändern sich auch.

Somit ergibt sich die in nächster Näherung das Update Telegram:

message alternatorHeartbeat {
	optional int32 status1 =1;
	optional int32 temp102 = 102;
	optional float actPower105 = 105;
	optional int32 unknown130 = 130;
	optional int32 startVoltage138 = 138; // *0.1
	optional float batVolt139 = 139;
	optional float soc262 = 262;
	optional int32 unknown268 = 268; // Zeit?
	optional int32 unknown269 = 269; // Zeit?
	optional float current425 = 425;
	optional int32 unknown427 = 427;
	optional int32 unknown428 = 428;
	optional int32 operationMode581 = 581;
	optional int32 startStop597 = 597;
	optional float powerLimit598 = 598;
	optional float unknown602 = 602;
	optional float ratedPower603 = 603;
	optional int32 unknown608 = 608;
	optional int32 unknown609 = 609;
}
Ich denke ich baue dies testweise als Objekte in den Adapter und dann müsstest du die Plausibilität mal anschauen.

evolvo187, 2024-12-15:
Das freut mich zu lesen.

Die alternatorSet Daten sind verständlich.
Die meisten alternatorHeartbeat auch.

Die Plausibilität schau ich mir nach dem Einbau natürlich gerne an.

foxthefox, 2025-01-02:
Ich habe die Version 1.1.0 auf git und npm erstellt. Bitte über Expertenmodus und octacat icon installieren.
In der Config gibt es einen neuen Abschnitt und dort das Batterieladegerät definieren.
Die Namen sind erstmal vorläufig, aber da es nun unter den Objekten aufgeführt wird, lässt sich dann gut ableiten, was sie darstellen.
Kommandos sind ja schon geklärt, aber habe ich noch nicht drin.

evolvo187, 2025-01-03:
Ich habe getestet.
4 Sachen sind mir aufgefallen.

**### 1.** 
Bei alternatorHeartbeat.operationMode ist ein Fehler.
bei  "2": "reverse charge",   wird in der APP auf Batteriewartung umgestellt.
und bei  "3": "battery maintenance" wird in der App auf Reverse Charge Modus umgestellt.

**### 2.**
alternatorHeartbeat.unknown269  ist **Batterie bis Voll**

alternatorHeartbeat.unknown608  ist **Kabellänge** (siehe Bild)     Beschreibbar?
[image]

**### 3.**
Mann kann in der App das Batterieladegerät ausschalten.   (siehe Bild)
[image]

Ich vermute das das mit 
alternatorHeartbeat.unknown130
und  
alternatorHeartbeat.status1
zu tun hat. 

Denn wenn ich ausschalte sind das die Letzen Einträge im Log.  (siehe log)

ecoflow-mqtt.1 | 2025-01-03 12:29:33.325 | debug | [Compare] update .alternatorHeartbeat.carBatVolt 12.222460746765137 -> 12.222482681274414
-- | -- | -- | --
ecoflow-mqtt.1 | 2025-01-03 12:29:33.323 | debug | [Compare] update .alternatorHeartbeat.unknown130 1 -> 0
ecoflow-mqtt.1 | 2025-01-03 12:29:33.320 | debug | [Compare] update .alternatorHeartbeat.status1 32 -> 0
ecoflow-mqtt.1 | 2025-01-03 12:29:33.317 | debug | [PROTOBUF decode] F371ZEB49G6Q0879 [update] msg#0 => {"status1":0,"temp":5,"actPower":0,"unknown130":0,"startVoltage":130,"carBatVolt":12.222482681274414,"batSoc":10,"unknown268":2466,"unknown269":1644,"current425":0,"unknown427":14,"unknown428":2,"operationMode":1,"startStop":1,"permanentWatts":800,"wifiRssi":-41,"ratedPower":800,"unknown608":7,"unknown609":0}
ecoflow-mqtt.1 | 2025-01-03 12:29:33.297 | debug | [PROTOBUF decode] F371ZEB49G6Q0879 msg#0 => pdata(hex) 0800b00605cd0600000000900800d0088201dd084a8f4341b51000002041e010a213e810ec0ccd1a00000000d81a0ee01a02a82401a82501b52500004844d525000024c2dd250000484485260000e0408d2600000000
ecoflow-mqtt.1 | 2025-01-03 12:29:33.297 | debug | [PROTOBUF decode] F371ZEB49G6Q0879 [update] msg#0 => {"src":20,"dest":32,"dSrc":1,"encType":1,"checkType":3,"cmdFunc":254,"cmdId":21,"dataLen":86,"seq":6510789,"productId":8193,"version":3,"payloadVer":1}
ecoflow-mqtt.1 | 2025-01-03 12:29:33.297 | debug | Message encrypted
ecoflow-mqtt.1 | 2025-01-03 12:29:33.297 | debug | [PROTOBUF decode] F371ZEB49G6Q0879 [update] stream has 1 message(s)
ecoflow-mqtt.1 | 2025-01-03 12:29:33.280 | debug | [PROTOBUF decode] F371ZEB49G6Q0879 [update] raw (hex): 0a770a56cdc575c3c008c3c5c5c5c555cdc515cd47c418cd8f4a868470d5c5c5e58425d567d62dd529c908dfc5c5c5c51ddfcb25dfc76de1c46de0c470e0c5c58d8110e0c5c5e10718e0c5c58d8140e3c5c5258548e3c5c5c5c51014182020013001380340fe014815505670c5b18d03788140800103880101
ecoflow-mqtt.1 | 2025-01-03 12:29:32.976 | debug | [PROTOBUF decode] F371ZEB49G6Q0879 [set] msg#0 => id set_dp3 error at: Error: decodeMsg -> RangeError: index out of range: 1 + 8 > 2

**### 4.**
alternatorHeartbeat.actPower  max Wert erhöhen auf ca. 830 (Warnungen im Log)

foxthefox, 2025-01-11:
Ich habe die Version 1.1.3 mit Verbesserungen zum Alternator erstellt.
Bitte testen und berichten.

Musste erstmal einiges umbauen, damit es kein Mapping auf andere Geräte mit gleichen Telegrammheader gibt (derFehler ".... id set_dp3 error at: Error: decodeMsg -> RangeError ..." sollte bei Befehlen aus der APP nicht mehr auftreten.

Die anderen Dinge habe ich so gut es ging übernommen (einige Datenpunkte sind damit veraltet, können gelöscht werden)

evolvo187, 2025-01-24:
Hallo, entschuldige die späte Rückmeldung.

Leider kann man jetzt nichts mehr steuern. 

Ich habe Version 1.1.3  und  1.2.1 getestet.

ecoflow-mqtt.1 | 2025-01-24 15:46:38.959 | info | state ecoflow-mqtt.1.F371ZEB49G6Q0879.alternatorHeartbeat.startStop changed: true (ack = true)
ecoflow-mqtt.1 | 2025-01-24 15:46:38.409 | info | (ack=false) ->cmd : channel alternatorHeartbeat state startStop
ecoflow-mqtt.1 | 2025-01-24 15:46:38.408 | info | state ecoflow-mqtt.1.F371ZEB49G6Q0879.alternatorHeartbeat.startStop changed: false (ack = false)
ecoflow-mqtt.1 | 2025-01-24 15:46:01.816 | info | state ecoflow-mqtt.1.F371ZEB49G6Q0879.alternatorHeartbeat.permanentWatts changed: 100 (ack = true)
ecoflow-mqtt.1 | 2025-01-24 15:46:00.887 | info | (ack=false) ->cmd : channel alternatorHeartbeat state permanentWatts
ecoflow-mqtt.1 | 2025-01-24 15:46:00.887 | info | state ecoflow-mqtt.1.F371ZEB49G6Q0879.alternatorHeartbeat.permanentWatts changed: 200 (ack = false)

foxthefox, 2025-01-24:
Die Befehle sind doch noch gar nicht implementiert. Siehe oben.
Ich habe es schon als Schalter und Wertänderung definert, aber es gibt noch keinen Code der daraus ein Telegramm macht.
Wollte erstmal Stabilität bei den Datenpunkten und dann kommt der Rest.

evolvo187, 2025-01-24:
Datenpunkte schauen gut aus.

außer 
alternatorHeartbeat.chargeToFull269
alternatorHeartbeat.unknown268
bleiben leer.

[image]

Ich glaube die Zeit wurde vorher Angezeigt. (wird von mir aber nicht benötigt)

Sonst keinerlei Fehlermeldungen oder andere Auffälligkeiten. 
TOP

foxthefox, 2025-01-29:
Ok, die null schau ich mal an, ob da wegen Umrechnung was in die Hosen gegangen ist

evolvo187, 2025-01-30:
Die Zeiten werden jetzt richtig angezeigt.

Bitte noch 2 kleine Änderungen 
permanentWatts
ist:  "max": 830,      ändern auf 800
ist:  "step": 100,      ändern auf 50

wifiRssi
ist   "min": -90,       ändern auf -95

foxthefox, 2025-01-31:
Kommt in die finale 1.2.2 rein.
Danach kümmern wir uns um die Kommandos und deren Test. Da ist wieder dein Hilfe gefragt

radeonorama, 2025-02-25:
Hey, I'm in a position with ioBroker installed, the addon in and configured to connect to my ecoflow alternator charger. I have an active instance of HomeAssistant also, so can test that side of the integration. What can I do to help get the set side of things going? Cheers

radeonorama, 2025-02-25:
Just noticed current425 is wrong. It's currently down as Amps which you would expect to be correct however its actually the current wattage the charger is running at.

foxthefox, 2025-03-09:
@radeonorama @evolvo187 

A preview of the version 1.3.0 is available at github for testing.
Commands are now implemented.

Update can be done by Adapter -> Expert mode -> octacat -> github

The obsolete datapoints (due to renaming (actualPower, current425)) can be deleted.

radeonorama, 2025-03-09:
Nice work! I'm away with my alternator charger now so updated and been having a play around. The commands seem to be working out fine The pull request I set has been partially reverted though. 

I'd changed the alternatorPower to be a entitytype: sensor, it's now set as number and really shouldn't be. 

The stationPower mult is now 0.001 and should just be 1.

Seperately, startVoltage should have steps of 0.5 and isn't updating within homeassitant quite right but I have a feeling that's something to do with the current steps. 

I'll keep testing and add more as I find them. 

Do you want me to just make a pull request with the amendments above?

foxthefox, 2025-03-09:
Thanks for testing, reverting was not my intention I will correct it immediatly. No need to make a PR, I correct my faults :-)
I have not checked the HA communication so far, but should be observed.

radeonorama, 2025-03-09:
Well so far this is great! I should have mentioned before that I tweaked the code to enable HA integration as the tick box option in your iobroker config page isn't actually available for the alternator. I'd suggest you add it in for others as it works almost perfectly so definitely worth being enabled

The one issue I'm seeing is with the startVoltage. It doesn't want to play nicely, particularly when setting to 11V (the lowest setting) within both iobroker and ha. In io I seem to have to set it as a strong value rather than a number, but only then it seems to only work for values higher than 12 but not below. That said, for me and I'd imagine most others, that's probably a set and forget option than isn't frequently interacted with. 

Permanent Power (permanentWatts) should be renamed to "Maximum output power" to match the app. This is essentially the user definable charge rate in watts

I don't fully understand what the switch off CMD (switchOFF130) is. The device does have a shutdown button under settings which is used when you are reconfiguring cables and perhaps moving the device to another vehicle etc but it doesn't quite make sense being switched on. I'm not in a position to click it right now but I presume it's actually "Power" as it's in the on position. I'll check tomorrow but I'd imagine that's what it'll be. 

I'd also rename Start Stop to "Start Working" to match to app. 

Very nice release though! I'm loving it. Really nice work, I really appreciate your efforts and time. I actually use it slightly differently with it connected to 600ah of extra LiFePO4 batteries instead of an alternator which is then connected to a delta pro which is also changed via another alternator charger. Yeah, a little confusing! Haha. This makes it incredibly powerful for me to be able to dump power to those extra 600ah if I hit certain states of charge on the delta pro whilst travelling around.

radeonorama, 2025-03-19:
Hey @foxthefox do you think you could give me some pointers on looking over the HQ integration code for the alternator charger? I'm seeing an issue when setting the charge rate. (permanentWatts). It seems to set ok when changed within HA however doesnt confirm it's update in HA. E.g. it'll be set to 800, I'll knock that down to say 400, iobroker and the actual alternator charger will update to the set value but HA won't and still displays 800. 

Any help would me much appreciated!

foxthefox, 2025-03-20:
Thank you for the feedback.

Thinking about the problem I may found the issue.
The adapter looks for changes and if change is detected it will be transferred to HA. In case of commands the change is happening with the command and and the feedback from device is no longer different and the check is not bringing it to the HA instance.
I have to verify that with my code, but I can do it earliest on the weekend.

Sorry for the inconvenience.

evolvo187, 2025-03-21:
Sorry für die verspätete Antwort,

ich habe heute mit Version 1.3.0 getestet. 

Das Batterieladegerät lässt sich bis auf  X_cableLength608 steuern und es werden die Werte richtig ausgelesen.

bei X_cableLength608 wird richtig gelesen kann aber nicht gesteuert werden.

bei X_status1 gibt es 2 Werte.
"0": "keine Unterspannung"
"32": "Unterspannung am Anschluss der Autobatterie"  (eingestellter Wert Start voltage)

ACHTUNG:
Mit  Version 1.3.0 kann ich meine Power Stream's nicht mehr steuern. 
( aber ich glaube das ist  schon bekannt. ? )

foxthefox, 2025-03-21:
@evolvo187 
Danke für die Rückmeldung.
Ja, cable length geht noch nicht. Da bräuchte ich ein Debug log wo der Befehl von der App drin ist. Dann kann ich es live schalten.
Das mit PowerStream ist bekannt und ich muss schauen was da noch fehlt, nach meinem Umbau. Zur Not auf 1.2.2 downgraden, falls benötigt.

foxthefox, 2025-03-24:
@evolvo187 
neue Version auf git löst das Problem mit der Powerstream

radeonorama, 2025-04-04:
Just updated to 1.3.1 and seeing some strange behaviour. 

The start stop command isn't staying one in home assistant when switched however does actually switch on on the alternator charger
The operation mode doesn't seem to like being set to charge mode. I change it within home assistant and whilst it appears ok within home assistant, it doesn't actually pass etthtpigh to the alternator charger

foxthefox, 2025-04-04:
@radeonorama 
1.3.1 is in working state, so its not final.
Usually I advice involved persons of a specific issue to test the current working stage.
If the version is final, also the npm version is created.

The 1.3.1 relates to some command issues (toggling back/forth). I try to prevent that, but I am also unsure if this is an issue of my adapter.

Sorry, but you need for now to roll back to 1.3.0.

PS. What is a little strange that it seems that the HA-cmd is not passed through the adapter and finally to the alternator.

foxthefox, 2025-04-06:
Please have a look to the final 1.3.1 at git or npm.
Hopefully everythink is working properly.

radeonorama, 2025-04-06:
Ok, great, changing the operation mode is working well now. 

I'm still seeing issues with the Charge rate ( alternatorheartbeat_permanent_power_limit ) not updating after change within home assistant. 

It actually seems worse now than it was before. It will feed through ok from home assistant to iobroker then onto the EF alternator charger but doesn't feed back into home assistant to confirm the change has been made.

foxthefox, 2025-04-06:
I am not so familiar with HA and its internal behavior.
With having an analysis within another issue, I found out that there is for numbers a feedback needed. A change in the HA (i.e. slider) is not populated within HA.
So I made a small change in my code and now numbers should also be aligned in each case.

Since I am not ready with 1.3.2, you can update it via ocatacat

[image]

At least for a powerstation the issue is gone with that change.

radeonorama, 2025-04-06:
Wow, that's so quick and stable now! Nice work!!

As and FYI, I just tried the alternatorheartbeat_start_voltage

and

alternatorheartbeat_x_cablelength608

and they're not working at all from within homeassitant.

For me, I'm not bothered at all as they're the kinda of set and forget initial config items, however figured you might want to fix them.

foxthefox, 2025-04-06:
Thanks for feedback.
The startVoltage should be working, I will check.
For the cableLength I do not have an example telegram from a change via APP, thats why it is not working.

Anyhow, I agree, these values are once set via APP and you are done.
But for startVoltage I will dig into it.

radeonorama, 2025-04-06:
Ah ok, is that something I can help with? If you can give me some guidance on getting the telegram you speak of, I'm sure I can get you what you need?

foxthefox, 2025-04-06:
starteVoltage had an missing e in my code, so that will be solved in next update.

If you want to halp, then there is some picturized help for the steps you should do:
https://github.com/foxthefox/ioBroker.ecoflow-mqtt/blob/main/doc/en/IOB_HA/logging.md

What is to be donein ioBroker:
- put the adapter in debug mode
- tick the boxes for msgSet/Get and msgCmd on the first tab
- tick the box for debugEnable in the alternator row on the second tab
- save the changes (will restart the adapter)
- go to the EF-App and change the cable length between some values
- go to the logs, the log will have some entries, post it here (you can also zip it)
- you then put the adapter in info mode again

radeonorama, 2025-04-06:
[image]

Great, thanks for the directions. 

Check the attached out. Hopefully it gives you the info you need. Let me know if I messed something up!

foxthefox, 2025-04-07:
Thanks for the help.
I could exactly see these values.

-> optional float cableLength608 = 203; is inserted into the code

It is uploaded to the intermediate version 1.3.2

foxthefox, 2025-04-08:
github tab or the npm tab, the custom is not working anymore since the code is merged to the main repo.

radeonorama, 2025-04-08:
Perfect! I'll report back shortly!

radeonorama, 2025-04-08:
So cable length is a little out. 

It can work on increments of .1 of a M starting at 0.2M up to 10M

If set within the official EF app the value displayed within HA is coming out as below:

[image]

I just have to do pop out but will report back about the start voltage shortly

foxthefox, 2025-04-09:
Basically it works, but we face the problem how float values can look like.
May be I find a beautifying solution.