foxthefox/ioBroker.ecoflow-mqtt #170: Energy values don't update automatically

manny4566, 2024-11-13:
**Describe the bug**  
Following values don't update automatically. Only after open the smart phone app:
I have checked the box "enable check for reconnects (10min)"
But it not help.
I use a Ecoflow Powerstream 800W

[image]

**To Reproduce**  
Check the values. They don't update automatically. Only after you open the app.

**Expected behavior**  
A automatically update from the adapter.
**Versions:**  
 - Adapter version: 1.0.5
 - JS-Controller version: 6.0.11
 - Node version: v20.18.0
 - Operating system: Debian 12

Thank you

foxthefox, 2024-11-18:
There are a lot of users reporting the same problems since EF is trying to cancel (or at least interfere) such connections. The adapter uses the same account as the App does and emulates the traffic. EF somehow detects that and the updates are somehow canceled after certain time.
Nevertheless the adapter does cyclically check the presence of the devices and that should at least give a reading then.

Until now I have not figured out what the App does differently with the subscriptions and the mqtt telegrams.

janopan, 2025-03-09:
Hi,
Thanks for you work and your adapter. I am having similar problems than manny:
All the objects under "energy" are zeroed. Also all the values under "inverter_heartbeat" that are measured in kWh are 0.
Strange behaviour. Is there a workaround or any other ideas that i could try?
Best wishes,
Jan

[image]

foxthefox, 2025-03-09:
I have to admit that there was a problem in versions > 1.1.1 up to 1.2.2 with evaluating the energy values.
With the upcomming 1.3.0 it will be corrected.
You might downgrade to 1.1.1 or upgrade to 1.3.0 (right now only via git)
For upgrading: Adapter -> Expertmode -> octacat icon -> github.
Otherwise Adapter -> Expertmode -> adpter tile + on right side and choosing 1.1.1

janopan, 2025-03-09:
Wow, thanks for that spontaneous answer.
I tried both 1.3.0 and 1.1.1, unfortunately both gave me the zeroed values for "energy"  (I deleted the object "energy" with all child objects, restarted, still no result).
I'll patiently wait for further investigation ^^

janopan, 2025-03-16:
Update: Now it works with 1.3.0. Maybe i just had to wait for daylight

foxthefox, 2025-03-16:
Thanks for feedback. Glad to hear hat it is working.

foxthefox, 2025-04-07:
I think the issue can be closed

closed 2025-04-07 by foxthefox