foxthefox/ioBroker.ecoflow-mqtt #174: Is it possible to also retrieve the circuit power consumption as shown in the app?

jxlarrea, 2024-11-24:
This is the information I'd also like to have, specifically how many watts each circuit is consuming:

[image]

foxthefox, 2024-11-25:
It could be a directly transmitted value or a calculated one.
Let me check the datapoints, before calculation.

The total power is the load of all circuits? Or is ther some solar/battery substracted and it is only the value from the grid?

jxlarrea, 2024-11-25:
The total power is probably calculated between` iob_ef/HD31XXXXXXXXX_Energy1Info/outputPower` and `iob_ef/HD31XXXXXXXXX_Energy1Info/inputPower` but those are already reported.  But what I am looking for is **the consumption of each individual circuit** as listed in the app.

I believe this should be reported via `hall1_watt.curr` and `hall1.watt` but those values are never pushed into MQTT. I do see those defined in the [https://github.com/foxthefox/ioBroker.ecoflow-mqtt/blob/main/lib/ef_shp2_data.js](https://github.com/foxthefox/ioBroker.ecoflow-mqtt/blob/main/lib/ef_shp2_data.js) file however.

jxlarrea, 2024-11-25:
Just for addition information. In the `[pd303.proto](https://github.com/rabits/ef-ble-reverse/blob/main/protobuf/pd303.proto)` file, these are the message definitions for both all loads and the circuit loads:

```
message WattTimeInfo {
  optional float grid_watt = 1;
  repeated float ch_watt = 11;
  optional float all_hall_watt = 21;
}
```

```
message LoadTimeInfo {
  repeated float hall1_watt = 1;
  repeated float hall1_curr = 2;
}
```

foxthefox, 2024-11-26:
It could be the case that the data for the individual circuits is only transmitted when the app is open and not with the regular updates. So it might be updated when the App is openend. Or there is a special request command, that meight be discovered first.

The hint to the other repo is closing a gap what I have, awsome.
I will check it and extend my adapter accordingly.

jxlarrea, 2024-11-26:
> It could be the case that the data for the individual circuits is only transmitted when the app is open and not with the regular updates. So it might be updated when the App is openend. Or there is [...]

@foxthefox I can confirm that the the data for the individual circuits is always transmitted, not only when the app is opened. I created a quick python project that connects to Ecoflow's MQTT and the circuit load data is transmitted almost every second, at least partially. Unfortunately, my knowledge of protobufs are extremely limited and have no clue how to get the complete data from Ecoflow's MQTT or even how to identify which message type I am receiving.

foxthefox, 2024-11-26:
in this case the data should be converted and be available with each update.
May it possible that you als make a log with the SHP2 (disable log on the DPU and enable SHP2 in the device tab).

jxlarrea, 2024-11-26:
@foxthefox Here are the logs. Actually you can see the hall1_watt and hall1_curr data right there!

[iobroker.2024-11-26.zip](https://github.com/user-attachments/files/17914079/iobroker.2024-11-26.zip)

```
2024-11-26 07:36:15.585 - debug: ecoflow-mqtt.0 (195404) [PROTOBUF decode] HD31ZAS4HG6N0873 [update] msg#0 => {"masterInfo":{"gridWatt":7041,"sysTimezone":999,"timezoneId":"America/Guayaquil"},"loadInfo":{"hall1Watt":[0,6464,68,0,0,8688,0,10752,68,0,64,1,68,0,0,0,0,0,0,106,68,0,0,8456,0,0,8524,0,0,0,0,0,0,40,66,0,0,64,66,0,0,8500],"hall1Curr":[15,48,8296,134504348,135905371,134511631,129439180,19,1059210,116,61,108,63,1417,8121,0,0,0,0,94,9504,63,111,13043,63,65,45,8091]}}
2024-11-26 07:36:15.586 - debug: ecoflow-mqtt.0 (195404) [Compare] update .ProtoTime.master_gridWatt 6792 -> 7041
```

foxthefox, 2024-11-26:
yes, indeed the load and current data is already converted from the protobuf and it is written to ProtoTime/hall1Curr or hall1Watt as an array. And I stopped  there, because there were to many values (more than the expected 12).
That is because I used int32 instaed of float.
Changing the code, returns now nicely 12 values.

Knowing this I can add new data structure for curr and watt and including the sum of it.

jxlarrea, 2024-11-26:
> yes, indeed the load and current data is already converted from the protobuf and it is written to ProtoTime/hall1Curr or hall1Watt as an array. And I stopped there, because there were to many values [...]

@foxthefox excellent thank you! This information is really useful for automation during blackouts for turning off stuff that consumes too much power during an outage like air conditioning systems.

foxthefox, 2024-11-26:
When I have finished with this issue and the DPU battery, I will inform and ask you for checking

foxthefox, 2025-01-02:
I have created the version 1.1.0, this can be installed via expert mode (head icon)and the octacat icon, via npm or from git.
From the WattTimeInfo it has the additional wattInfoChWatt, wattInfoAllHallWatt beside of the existing wattInfoGridWatt.
Most likely they are not always updated, so I created also a sum in the channel section (loadPower, loadCurrent)
You may compare the values and how often they arrive.

Please check and report.

jxlarrea, 2025-01-02:
> I have created the version 1.1.0, this can be installed via expert mode (head icon)and the octacat icon, via npm or from git. From the WattTimeInfo it has the additional wattInfoChWatt, [...]

Hi @foxthefox! Just tested this and works perfectly. All watts and current values are being pushed in real-time.

Thank you so much!

foxthefox, 2025-01-02:
That sounds good.
Thanks for testing.

I will close this issue then.

closed 2025-01-02 by foxthefox

ekashuba, 2025-02-16:
@foxthefox, I'm sorry to revisit this issue but it doesn't seem to be working for me using 1.2.2   I've just hooked up my SHP2 so I don't know if was broken in versions newer than 1.1.0 or if it's something that I'm doing wrong on my end.

I'm getting a bunch of data from both the DPU and and SHP2 but under the SHP2 I'm just getting NULL's for everything under loadCurrent, loadPower, and ProtoTime. 

@jxlarrea, is it working for you under 1.2.2 or are you still using 1.1.0?

Thanks for any help, and  please consider adding a donation option for all of the work that you've been putting into this!  It's much appreciated.

jxlarrea, 2025-02-16:
@ekashuba I am on 1.1.2. Works perfectly. I have not upgraded to any newer versions.

jxlarrea, 2025-02-16:
@ekashuba just upgraded to 1.2.2. Works perfectly too.

ekashuba, 2025-02-16:
@jxlarrea  Thanks for checking (and doing a gutsy update)!  I guess I can rule out the version as a culprit, then.

ekashuba, 2025-02-17:
Just one last update on this....  I ran out of things to check on my end so on a whim I tried re-installing the 1.2.2 version of the adapter and I started getting values for those missing attributes.  I'm going to chalk up my problem to some sort of issue when I'd originally updated the adapter.