foxthefox/ioBroker.ecoflow-mqtt #181: Connection with new Delta devices

vplustute, 2024-12-18:
I have a Delta 2 max and the adapter works perfectly. Now I've bought a Delta 3 Pro and I can't adjust the MQTT connection at the moment because the plugin doesn't support the new devices yet. Is there any information on whether and when the new devices will be supported?

foxthefox, 2024-12-19:
The Delta Pro3 is planned to be included, but unfortunately the communication is via protobuf encoding, which needs to be decoded. I got already some samples, but it is not near to have a detailed glue what the data represents.
Most likely I will make some preliminary hack to make the data visible and then the educated guesses is easier.
It would be nice if you can support here.

vplustute, 2024-12-23:
I can gladly support you in testing. Unfortunately, I cannot assist with any kind of programming.

foxthefox, 2024-12-30:
Most likely I will find time today and tommorrow to make some additional code for better analysis.

vplustute, 2025-01-02:
Ok, just let me know how I can help you.

foxthefox, 2025-01-02:
I have created the version 1.1.0, this can be installed via expert mode (head icon)and the octacat icon, via npm or from git.
This brings a new section in the second tab where you can configrue the serial number and the delta pro 3 as a unknown device.
After restart it will create log entries with [PROTOBUF unknown] tagging.
Then you may open the App and open some pages of the DP3 and may also change some settings.
You can also change toggle output and attach some loads to the device or have solar incomming.
All that will be captured in the transmitted data.
After a while you can download the log and provide it as zip here.

vplustute, 2025-01-03:
I have installed and configured everything.
I don't know if I have changed enough settings on the Delta Pro 3. I am still happy to help.

[iobroker.2025-01-03.zip](https://github.com/user-attachments/files/18297869/iobroker.2025-01-03.zip)

DW876, 2025-01-06:
I also have a Delta Pro 3, configured iobroker with your 1.1.1 verison. Please let me know what I can do to assist.

foxthefox, 2025-01-07:
@DW876 that look good.
From another sample I found the following commands:

message set_dp3 {
	optional int32 beepCmd = 9;
    optional int32 switchACoutHvCmd = 15;
    optional int32 switchACoutLvCmd = 16;
    optional int32 switch12VdcCmd = 18;
    optional int32 switchXboostCmd = 25;
    optional int32 levelChargeLimitCmd = 33;
    optional int32 levelDisChargeLimitCmd = 34;
    optional int32 level12VchargeCmd = 52;
    optional int32 level48VchargeCmd = 53;
    optional int32 levelACchargeSpeedCmd = 54;
    optional int32 levelACpowerInOutCmd = 56;
	optional int32 highChargeLimit = 59;
    optional int32 switchGFCIRCDCmd = 61;
	optional int32 lowDischargeLimitCmd =102;
}
you may verify the field numbers if you want, and if there are other commands then let me know

foxthefox, 2025-01-12:
@vplustute 

I have prepared the needed protobuf definitions to make the telegram content visible.
Over 300 values are throughout the telegrams and I can already see some which could be interpreted as temp, voltage, etc.
But I need some more help, especially on the operating conditions.

Can you create a log in the same manner as previously plus a small writeup from your performed actions related to the below suggestions?

Additionally I would hope to see change when you do the following (suggestion, but I am not sure if it is feasable):
- note state of charge, 
- is Solar attached and what are the values according the EF-App?
- switch on the DP3, but no AC in and no activation of any output
- apply the AC in plug (hopefully it does not start to charge immediately, otherwise it may help to reduce the maxCharge value to lower than actual state)
- start charging the DP3 via AC in
- stop charging via AC in
- switch on USB
- attach a load to the 1st USB-C port (cell phone charging, or something what draws some watts), note down what the APP reports
- move the load to 2nd USB-C
- move the load to 1st USB-A or disconnect the 2nd USB-C load and use another load at USB-A, note down what the APP reports
- move the load to 2nd USB-A
- switch off USB
- switch on car output
- attach a load if you have one, note down what the APP reports
- switch off car output
- switch on AC output
- attach a load to AC port (like a hair dryer), note down what the APP reports
- move the load to the different ports (I dont think that we have here several measurements)
- switch off the AC output
- note state of charge
this is to see the appearing values and map to voltage and current and state

another set of actions would be related to the EF-App, I do not knwo what is possible, but we should toggle all possible switches or sliders.
- change the standby for LCD, to a new value and back to previous
- change the standby for device, to a new value and back to previous
- change the brightness of LCD, to a new value and back to previous
- change the lower discharge value, to a new value and back to previous
- change the upper charge value, to a new value and back to previous
- change the value for AC charge
- toggle AC output
- toggle USB output
- toggle Car output
this helps to identify the settings inside the data.

After all the log should contain the telegrams and data and with knowing what was done and which values, I can make my educated guesses.

foxthefox, 2025-01-12:
BTW, I could enhance the command mapping to the protobuf structure a little bit more (by looking into the official API).

message set_dp3 {
    optional int32 shutdown = 3;
	optional int32 beepCmd = 9;
    optional int32 acTimeOut = 10;
    optional int32 dcTimeOut = 11;
    optional int32 screenTimeOut = 12;
	optional int32 deviceTimeOut = 13; 
    optional int32 screenBrightness = 13; 
    optional int32 switchACoutHvCmd = 15;
    optional int32 switchACoutLvCmd = 16;
    optional int32 switch12VdcCmd = 18;
    optional int32 switchXboostCmd = 25;
    optional int32 levelChargeLimitCmd = 33;
    optional int32 levelDisChargeLimitCmd = 34;
    optional int32 levelPVcurrentLVCmd = 52;
    optional int32 levelPVcurrentHVCmd = 53;
    optional int32 levelACchargePowerCmd = 54;
    optional int32 levelACpowerInOutCmd = 56;
    optional int32 generatorStartStopCmd = 58;
	optional int32 highChargeGeneratorStopLimit = 59;
    optional int32 lowDischargeGeneratorStartLimit = 59;
    optional int32 switchGFCIRCDCmd = 61;
    optional int32 acEnergySavingOpen = 99;
    optional int32 multiBpChgDsgMode = 100;
	optional int32 lowDischargeLimitCmd =102; //not EF-API
}

vplustute, 2025-01-13:
Ok, that really is a lot.
I'll have to see how I can get it documented and coordinated. I'm a bit short of time at the moment. I have a lot of other things on my plate at the moment.

I have ecoflow developer access. The DeltaPro 3 is already listed there. Can that help you too?

[image]

foxthefox, 2025-01-13:
@vplustute 
Yes, its a lot of activities, which need to be done one after another in order to see more or less singular changes in the incomming data.
FYI, the telegrams with protobuf coding do only contain values at a position in the telegram data. If they would be sent via JSON like Delta/Delta2 it would be much easier, since the object "speaks" with a name. Here it has to be mapped by educated guesses and that is supported by knowing what is the condition or the change.
There is no hurry, I am also limited on time.

The official API documentation is limited and in that special case very detailed for the commands.
The exchanged data with this API is reduced and does not explain the protobuf coding.
In particular cases it is helpful, i.e. naming the measurements. Even the multiplication factor is at many times not useable.

foxthefox, 2025-01-23:
@vplustute , @DW876 
Can you give me some hints?
I am looking for the ranges in the EF-APP, e.g. for timeout values in other models only dedicated values are possible, or is not starting at 0.

Especially I am looking for:
- range SOC
- range Generator start
- range Generator stop
- range Charging Power IN/OUT port
- range Charging AC power
- range LV Mppt current
- range HV Mppt current
- range/values device timeout
- range/values screen timeout
- range/values AC timeout
- range/values DC timeout

When I have them, I can finalize implement the commands portion.

DW876, 2025-01-26:
Hello @foxthefox, apologies as I have been unable to collect more data from iobroker due to time. Here are the values in the app I was able to locate. I hope this is helpful.

- range SOC
   - Discharge Limit 0-30
   - Charge limit 50-100
- range Generator start
   - 10-30
- range Generator stop
   - 50-100
- range Charging Power IN/OUT port
   - 500-4000watt
- range Charging AC power
   - 400-1500watt
- range LV Mppt current
   - not adjustable
- range HV Mppt current
   - not adjustable
- range/values device timeout
   - 30m,1h,2h,4h,6h,12h,24h,never
- range/values screen timeout
   - 10s,30s,1m,5m,30m,never
- range/values AC timeout
   - 30m,1h,2h,4h,6h,12h,24h,never
- range/values DC timeout
   - 30m,1h,2h,4h,6h,12h,24h,never
- Wifi/BT timeout
   - 1h,6h,12h,1day,7day,30day

foxthefox, 2025-01-29:
@DW876 
Thank you very much, I will prepare the datapoints accordingly.

vplustute, 2025-03-04:
Hi,

I might have a little more time at the moment and could help with information if anything else is needed.

phoniclynx, 2025-03-04:
Would love tobe able to see things like the solar inputs, DC in's, USB out, AC in, AC out

foxthefox, 2025-03-04:
I continued on the delta pro 3 and will have soon values and the commands. Unfortunately I catcherd the flu and feeling tired and sick right now. As soon as I am ready, I will post here and you can verify the results.

radeonorama, 2025-03-05:
Sorry to hear you're not well pal! All the best and get well soon!

phoniclynx, 2025-03-07:
I've also been following this and got it to partially work: https://gist.github.com/nalditopr/493af377e9928b88c2639a9e1607f127

foxthefox, 2025-03-09:
With the help [of a nice other project](https://github.com/rabits/ef-ble-reverse/tree/main/protobuf) I was able to figure out many states. There are more telegrams, especially for the battery, but that takes a while more (or I expose for now only the found states)
Thanks for your patience, its time for the testing :-)

I have created version 1.3.0 on github for now and this can be used for update.
Switch to adapter -> expert mode -> octacat icon -> github
see here for details: https://github.com/foxthefox/ioBroker.ecoflow-mqtt/blob/main/doc/en/IOB_HA/adapter.md#new-version-at-npmgithub

in the configuration tab you can now configure a powerstation as Delta Pro 3, you can delete the config within the unknown devices.

some remarks:
- range HV/LV Mppt current is shown as adjustable and in principle the command is implemented, but since it is not done through the App I am in doubt if it should be done through the adapter (coammdning will not issue something)
- ble/wifiTimeOut is also implemented as command, but I need a caputure from the APP before making it live
- energyBackup settings is missing, I need before a capture from the APP command.
- many commands are part of the structure DisplayPropertyUpload, because I found hopefully the state matching the command. Other without a matching state are appearing in setDp3 and have no initial value until operating from the adapter

For testing:
- please focus on plausibility of the received values
- when changing through the APP the values should be visible in the adapter as well and vice versa

It would be also nice if I can get another debug log for further testing (especially with conditions different to 0 (e.g. PV input, AC input, AC output, USB charging ...))

I am curious about your comments and test results.

kenthua, 2025-03-10:
thanks! this is a great start, i'm seeing values for my Delta 3 plus and River 3 plus

foxthefox, 2025-03-10:
@kenthua 
Nice to hear that it is also working for the D3P and R3P. It is expected, that there are common datapoints, but the devices are also different in abilities and features.
Therefore it would be very kind of you, if you could parametrize both devices under unknown devices, put the adapter in debug mode and post log. The log duration should be cover also some actions from the EF-APP (shuffling around the different viewes and devices).

kenthua, 2025-03-12:
it looks like there are some issues deciphering the ack messages coming back from the r3p and d3p?

When I select `none`, I just get no topic subscription, I believe this is WAI? However, when I do select the respective `River 3 Plus` I get back the failed json responses.

[iobroker.txt](https://github.com/user-attachments/files/19200143/iobroker.txt)

foxthefox, 2025-03-12:
@kenthua 
Thanks for testing. I did major refactoring and forgot the  handling if „unknown“ devices. I will correct it today in the evening.

foxthefox, 2025-03-12:
@kenthua 
you can please try again with the updated version on github (Expert mode -> ocatact -> github)
it should now run stable and put the incomming msg of unknown device into the log.

kenthua, 2025-03-13:
This is when the `Delta 3 Plus` is selected, looks like it needs to be handled?

```
2025-03-13 04:17:24.846 - info: ecoflow-mqtt.0 (551) unknown -> [{"devName":"d3p","devId":"<redacted>","devType":"delta3plus","debugEnable":true}]
2025-03-13 04:17:24.846 - debug: ecoflow-mqtt.0 (551) device upd undefined
2025-03-13 04:17:24.846 - warn: ecoflow-mqtt.0 (551) did not get devupd for delta3plus
2025-03-13 04:17:24.846 - info: ecoflow-mqtt.0 (551) start device state creation ->delta3plus for Id <redacted>
2025-03-13 04:17:24.906 - debug: ecoflow-mqtt.0 (551) SET initial <redacted>.info.status to offline
2025-03-13 04:17:24.907 - info: ecoflow-mqtt.0 (551) pdevices states created for <redacted> / delta3plus / d3p
2025-03-13 04:17:24.968 - info: ecoflow-mqtt.0 (551) [EF] going to connect to mqtt broker
...
2025-03-13 04:17:25.213 - error: ecoflow-mqtt.0 (551) Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch().
2025-03-13 04:17:25.213 - error: ecoflow-mqtt.0 (551) unhandled promise rejection: Cannot convert undefined or null to object
2025-03-13 04:17:25.214 - error: ecoflow-mqtt.0 (551) TypeError: Cannot convert undefined or null to object
at Function.keys ()
at EcoflowMqtt.onStateChange (/opt/iobroker/node_modules/iobroker.ecoflow-mqtt/main.js:1748:19)
at processTicksAndRejections (node:internal/process/task_queues:95:5)
at runNextTicks (node:internal/process/task_queues:64:3)
at process.processImmediate (node:internal/timers:454:9)
2025-03-13 04:17:25.214 - error: ecoflow-mqtt.0 (551) Cannot convert undefined or null to object
2025-03-13 04:17:25.215 - debug: ecoflow-mqtt.0 (551) lastQuotInterval interval stopped
```

kenthua, 2025-03-13:
Here is the log for the `River 3 Plus`

[iobroker-r3p.log](https://github.com/user-attachments/files/19222547/iobroker-r3p.log)

foxthefox, 2025-03-13:
> This is when the `Delta 3 Plus` is selected, looks like it needs to be handled?

The fault should not appear, a little bit strange that there was no logging at all, since the error is part of the command section.
Anyhow I will take care and provide an update

foxthefox, 2025-03-13:
once again, the code is updated, hopefully now delta3plus will be handled correctly

the r3p.log shows quite some differences to the deltapro3, but I think that I am very close.

I saw also some commands from the App.
Which commands are available in the App and what are the ranges?

kenthua, 2025-03-13:
> once again, the code is updated, hopefully now delta3plus will be handled correctly
> 
> the r3p.log shows quite some differences to the deltapro3, but I think that I am very close.
> 
> I saw also some commands from the App. Which commands are available in the App and what are the ranges?

Charge & Discharge
Backup reserve
Scheduled task
Input and output settings
- AC  charge speed (80W-650W)
- DC charging mode
- Car input (4A-8A)
More Settings
- Auto timeout
- Time zone
- Wi-FI
- Fault history

kenthua, 2025-03-13:
See data for d3p

[iobroker-d3p.log](https://github.com/user-attachments/files/19238369/iobroker-d3p.log)

vplustute, 2025-03-15:
First of all, thank you for putting so much effort into this.
What I have currently noticed is that the input power from the LPV and HPV input is not displayed.
Am I seeing this correctly or am I making a mistake?

foxthefox, 2025-03-15:
@vplustute 
I think there are at least 2 datapoints with lv and hv in the name, but we have to doublecheck. Sometimes the device do not send all data, when there is no need. It could also the case that data is sent but not recognized, because it was not sent in the examples and I did not prepare datapoints for it. 
If you have some data from log files I can verify if data is sent.

@kenthua 
D3P and R3P is basically implemented in git, but I did not verify and crosscheck very much. I will make next changes on next weekend, when back at home.

vplustute, 2025-03-17:
So that I can help properly, which log file is required, how do I create it and where can I find it?

foxthefox, 2025-03-17:
I think you had already made logging previously. Basically the similar logging within iobroker.

The logging of adapter has to be enabled as well as the device log. On the first tab the msgUpdate and msgSetGet should be enabled.

I have made also some screenshots for explanation.
https://github.com/foxthefox/ioBroker.ecoflow-mqtt/blob/main/doc/en/IOB_HA/logging.md

vplustute, 2025-03-17:
Here is the current log file

[iobroker.2025-03-17.log](https://github.com/user-attachments/files/19286221/iobroker.2025-03-17.log)

foxthefox, 2025-03-17:
@vplustute 
Thanks for the log.
Just checked roughly and it seems that some corrections are needed for the powerstream, since the log is full of type mismatches. I have to dig, why these are not appearing at my setup.

Also I guess that the adapter is not turned into debug mode. Within instance in the row of ecoflow-mqtt debug should be indicated. To turn it on, there is a small pencil icon in the header when opening the adapter config page. From info to debug when clicking on it.

vplustute, 2025-03-18:
Yes, the adapter is in debug mode.

vplustute, 2025-03-19:
Could it be that there have also been changes to the Powerstream? Somehow I have recently had the problem that I can no longer write to the ecoflow/HW5XXXXX_inverter_heartbeat/permanentWatts value. The value is always reset.

foxthefox, 2025-03-19:
I made a comprehensive refactoring of code, may be something slipped away. 
If it is feasible, you could delete the instance of the adapter and install it new from git via octacat icon. That might clean up things.

vplustute, 2025-03-19:
Unfortunately, deleting the instance did not help either. The LOG only shows me the following.

2025-03-19 14:14:49.791	info	state ecoflow-mqtt.0.HW51ZOH4XXXXXXXX.inverter_heartbeat.permanentWatts changed: 0 (ack = true)
2025-03-19 14:14:48.943	info	(ack=false) ->cmd : channel inverter_heartbeat state permanentWatts
2025-03-19 14:14:48.942	info	state ecoflow-mqtt.0.HW51ZOH4SXXXXXXX.inverter_heartbeat.permanentWatts changed: 200 (ack = false)

Have a nice vacation for now.

foxthefox, 2025-03-23:
@vplustute 
I have an intermediate feedback.

With the new version from git the issue for controlling the powerstream should be solved.

I checked one of your logs and it seems that the Delta3Pro is still configured at the "unknown devices" section and with type = delta3plus
Please reconfigure the Delta3Pro within the "powerstations" section and delete it in the "unknown devices" section.
That should bring the values od LV and HV solar.

The messages like inverter_heartbeat.xxxx has to be type "boolean" but received type "number" I could not reproduce.
Can you stop the adapter, delete the object tree of the powerstream and then restart the adapter.
This recreates the states and it should not assign boolean to the states.

vplustute, 2025-03-27:
Hello,

Thank you for including the new Ecoflow products.

Everything is working fine now.
I can adjust the ‘PermanetWatts’ value again via ioBroker and after reintegrating the Delta 3, everything is working great.

I can't detect any errors at the moment.

foxthefox, 2025-03-27:
I think I will make the 1.3.0 now final and next feedback/issues are going into 1.3.1

kenthua, 2025-03-28:
thank you for the quick turnaround @foxthefox 

An interesting observation with the river 3 plus I have an extended battery, `bms*` is the main battery and `cms*` is the combination of the main & extended battery.

Rainer1969, 2025-03-30:
Delta3 Series ???
Hi all,
caution: my very first contribution here, please have mercy ;-)
I tried to read everything and searched intensively, but found no information about support of the new Delta3-Series (not Plus)
. This is the successor of Delta2 products. I've bought a  
https://eu.ecoflow.com/products/delta-3-series-portable-power-station
Obviously, this is not supported right now?
If I am right, I would be happy to assist any willing hand to fix this by testing and giving needed information (log files etc)
I think this is quite important since this is for EcoFlow the mainstream product for the future.
If I am wrong: I'm working on the last beta 1.3.0-beta4, and at adding the device I can not select the model (he says not defined). If I select Delta-2 (best match), the entities are listed, but mostly wrong/empty and I cannot adjust power charging level.
Any help appreciated!

foxthefox, 2025-03-30:
@Rainer1969 
With version 1.3.0 (from npm) the Delta3Plus is supported. A 1.3.0-beta4 is not existing from my side.
Maybe you do not have the latest and finalized 1.3.0, therefore you need to 
- stop the adapter
- in the objects tree, delete all entries of the D3P (right click on the line with your serial and select all objects to delete)
- enable expert mode in adapter view (head symbol)
- click on octacat icon
- use npm and type ecoflow-mqtt, the name will appear and you install the newest version from npm (1.3.0)
- configure your D3P device in the second tab under powerstations and select Delta3Plus (most likely only changing the type)
- start adapter

Delta2 is completely different and will not provide any results.

See some hints here (use npm instead github!):
https://github.com/foxthefox/ioBroker.ecoflow-mqtt/blob/main/doc/en/IOB_HA/adapter.md

Rainer1969, 2025-03-31:
@foxthefox: thank you for the quick answer.
Good to hear that delta3 is supported. I followed your recommendations and re-setup everything, it seems to work now.

foxthefox, 2025-04-06:
There is a new version 1.3.1 out, which solves some HA problems and also provides some more adjustments in HA now.
Happy to hear feedback

foxthefox, 2025-07-11:
I have created a preview of the next version.
Could you please check it.

it must be installed via expert mode -> custom, described here:
https://github.com/foxthefox/ioBroker.ecoflow-mqtt/blob/1.4.0/doc/en/IOB_HA/adapter.md#new-version-for-testing-on-github-not-main-branch

kenthua, 2025-08-20:
so far so good for Delta 3 Plus and River 3 Plus, no issues on 1.4, thanks!

closed 2025-08-20 by foxthefox