foxthefox/ioBroker.ecoflow-mqtt #183: Connection refused

taz-man, 2024-12-31:
**Describe the bug**  
I get the following error when I start the ecoflow instance in iobroker

error | Error inconnection to Ecoflow MQTT-Broker:ErrorWithReasonCode: Connection refused: Not authorized

**To Reproduce**  
Steps to reproduce the behavior:
I installed the ecoflow-mqtt
Put my email and password in for Ecoflow
Clicked CREATE LOGIN DATA
Once saved it gives the error no matter if I list devices or not
I checked the data created for mqtt and it matches what the link to retrieve data got also

**Expected behavior**  
I expected to retrieve information from ecoflow

**Versions:**  
 - Adapter version: 1.0.5
 - Node.js: v20.18.1
 - NPM: 10.8.2
 - Operating system: win32

foxthefox, 2024-12-31:
Not nice, the log says that there is problem with autorization.
This was never reported in the past, I have no idea what would be the problem.
If the MQTT-server refuses the login/subscription I can't do much against it.
Maybe there is something changed at EF, which is unknown and you are the first experiencing it.

To exclude the adapter and its algorithm to get the credentials, you may go to the website

https://energychain.github.io/site_ecoflow_mqtt_credentials/

and it should return credentials which you then can copy to the adapter.
Save and restart the adapter.

Does your EF-App work?

taz-man, 2024-12-31:
Thank you for your help @foxthefox!
I don't know what information to use from the link you provided for all fields in the adapter.
The link you has USER which I put in mqtt user name.
The link has PASSWORD which I put in mqtt pwd
The link has CLIENT ID (sample) which I put in mqtt clientid.
The CREATE LOGIN DATA supplies a numerical value for mqtt userid which I see no value for in the link.

The app works fine for me and I have powrinsigt which streams the data and I tried logging out of everything since the powerinsight has a constant connection and that did not seem to help

taz-man, 2024-12-31:
I called Ecoflow support and they opened a ticket and escalated it to the higher-ups. It feels like they can not even "see" your account when you call in but maybe someone will look to see if there is a problem with my ccount.

danielcoltonwood, 2025-01-06:
I just installed 1.1.1 and the credential section used the hostname mqtt-e.ecoflow.com, however the site https://energychain.github.io/site_ecoflow_mqtt_credentials/ provides the hostname mqtt.ecoflow.com.

I was getting the same error, but it worked after changing the hostname to mqtt.ecoflow.com.

DW876, 2025-01-06:
I just installed 1.1.1 and the credential section used the hostname mqtt-e.ecoflow.com, however the site https://energychain.github.io/site_ecoflow_mqtt_credentials/ provides the hostname mqtt.ecoflow.com.

I was getting the same error, but it worked after changing the hostname to mqtt.ecoflow.com.

taz-man, 2025-01-06:
@DW876 Thanks that worked for me also!
I was able to get the DPU to show up but not the SHP2.
If Ecoflow changes something on my open ticket with them I don't know what will happen.

taz-man, 2025-01-07:
I'm going to close this because I think this issue is fixed.
Thanks for the support.

closed 2025-01-15 by foxthefox