foxthefox/ioBroker.ecoflow-mqtt #191: Adapter installiert - Instanz nicht sichtbar

DerMarioH, 2025-01-05:
Io-Broker: 7.1.5
Node.js: v20.18.1
NPM: 10.8.2

Ich habe die Version 1.05 und danach auch die 1.11 installiert. Der Adapter ist auch sichtbar. Allerdings ist keine Instanz zu finden. So kann keine Einstellung erfolgen. Woran kann es liegen?

foxthefox, 2025-01-05:
Nach der Installation des Adapters muss auch über „+“ eine Instanz erstellt werden. Es geht auch mehrere Instanzen zu erzeugen. Die erste hat im Objektbaum die 0 und zweite die 1 usw. 
gibt es im objektbaum eine ecoflow-mqtt Hierachie?
Gibt es Fehlermeldungen beim erzeugen einer Instanz?

DerMarioH, 2025-01-05:
Hat geklappt. Ich hatte die Instanzen vergessen.
Vielen Dank für die schnelle Hilfe.

closed 2025-01-05 by foxthefox