foxthefox/ioBroker.ecoflow-mqtt #194: AC switch dont work on Delta 2

rogerbl, 2025-01-07:
The AC switch is not possible to control via Mqtt on my Delta 2. It just bounce back to close.
On my River 2(works with River 2 Max setting) the AC switch works via Mqtt, and with my River 2 Max also. So would it be an idea for you to split this into River 2 and River 2 Max drivers?
Keep on with the good work! I got adapter ver. 1.0.5

foxthefox, 2025-01-07:
The change between 1.0.5 and 1.1.0/1.1.1 was mainly related to SHP2 and DPU. For Delta2 was only the change of measurement type from power to energy. Both should not have an effact on the data transmission to the MQTT-Server.

If you want you can also try to downgrade, via the 3dots
[image]

and click ob the + and selecting the 1.0.5 version
[image]

I can easily make a separate implementation of the River2, based on the River2Max.
Here it would be good to have the set of datapoints.
Therfore you can go to the config page of the adapter, enable the msgUpdate, msgSetGet and msgCmd plus additionally the debug enable in the device row. 

When adapter is in debug mode (via header of adapter config page)
[image]
the log will contain the transferred messages in JSON format.
That I would need to implement the driver.

The same setting would be needed to look for the Dalta2 AC switch command. 
Here the Delta2 shouldbe enabled in the device row and the in the APP you should switch the AC switch.
These cmd JSON telegrams I would also need to crosscheck with the existing code.

rogerbl, 2025-01-07:
Thanks, I will check this out.

Sendt fra Outlook for Android<https://aka.ms/AAb9ysg>

Bildschirmfoto.2025-01-07.um.18.01.34.png (view on web)<https://github.com/user-attachments/assets/05640473-db18-4f26-8434-203d72ea76ae>

Bildschirmfoto.2025-01-07.um.18.01.48.png (view on web)<https://github.com/user-attachments/assets/4801eb44-24fa-490e-98e2-538bdc1998f5>

Bildschirmfoto.2025-01-07.um.18.12.32.png (view on web)<https://github.com/user-attachments/assets/05d7c057-ee67-4f29-adc6-6d23a76c3580>

rogerbl, 2025-01-07:
Hi.¨
 I stay at the ver 1.1.1 cause the Delta 2 is nr 1 priority now.
 I toggled the AC on the Delta 2 from the App and this is the debug info from the log in that moment I believe:

ecoflow-mqtt.0
2025-01-07 19:49:44.468 debug [JSON] R331ZEB5SG961433 [update] -> {"addr":0,"cmdFunc":0,"cmdId":0,"id":1903012255553159250,"version":"1.0","timestamp":1736275784,"moduleType":"5","params":{"mppt.carState":0,"mppt.carOutVol":8685,"mppt.acStandbyMins":0,"mppt.cfgAcXboost":1,"mppt.carStandbyMin":0,"mppt.cfgAcEnabled":0,"mppt.chgType":0,"mppt.dcChgCurrent":8000,"mppt.inWatts":0,"mppt.beepState":1,"mppt.cfgChgWatts":1200,"mppt.cfgAcOutFreq":50}}

ecoflow-mqtt.0
2025-01-07 19:49:43.851 debug station update .inv.acInVol 220.089 -> 219.458

ecoflow-mqtt.0
2025-01-07 19:49:42.615 debug [JSON] R331ZEB5SG961433 [update] -> {"addr":0,"cmdFunc":0,"cmdId":0,"id":1903012238238023818,"version":"1.0","timestamp":1736275782,"moduleType":"3","params":{"inv.outTemp":19}}

ecoflow-mqtt.0
2025-01-07 19:49:41.718 info  state ecoflow-mqtt.0.R331ZEB5SG961433.mppt.cfgAcEnabled changed: false (ack = true)

ecoflow-mqtt.0
2025-01-07 19:49:41.716 debug station update .pd.acEnabled on -> off

ecoflow-mqtt.0
2025-01-07 19:49:41.715 debug station update .mppt.cfgAcEnabled true -> false

ecoflow-mqtt.0
2025-01-07 19:49:41.712 debug station update .mppt.carOutVol 10.146 -> 9.405

ecoflow-mqtt.0
2025-01-07 19:49:41.709 debug [JSON] R331ZEB5SG961433 [update] -> {"addr":0,"cmdFunc":0,"cmdId":0,"id":1903012229648089234,"version":"1.0","timestamp":1736275781,"moduleType":"1","params":{"pd.remainTime":-5939,"pd.carWatts":0,"pd.usb1Watts":0,"pd.wattsInSum":0,"pd.typec1Watts":0,"pd.dcOutState":0,"pd.lcdOffSec":300,"pd.soc":80,"pd.acEnabled":0,"pd.wattsOutSum":0,"pd.standbyMin":0}}

ecoflow-mqtt.0
2025-01-07 19:49:41.709 debug [JSON] R331ZEB5SG961433 [update] -> {"addr":0,"cmdFunc":0,"cmdId":0,"id":1903012230422462595,"version":"1.0","timestamp":1736275781,"moduleType":"5","params":{"mppt.carState":0,"mppt.carOutVol":9405,"mppt.acStandbyMins":0,"mppt.cfgAcXboost":1,"mppt.carStandbyMin":0,"mppt.cfgAcEnabled":0,"mppt.chgType":0,"mppt.dcChgCurrent":8000,"mppt.inWatts":0,"mppt.beepState":1,"mppt.cfgChgWatts":1200,"mppt.cfgAcOutFreq":50}}

ecoflow-mqtt.0
2025-01-07 19:49:41.417 debug [JSON] R331ZEB5SG961433 [set_reply] -> set w/o params{"id":403931202,"version":"1.0","operateType":"acOutCfg","moduleType":5,"time":206408706,"code":"0","data":{"ack":0}}

ecoflow-mqtt.0
2025-01-07 19:49:41.176 debug [JSON] R331ZEB5SG961433 [set] -> key:undefined {"from":"Android","id":"403931202","isMatter":0,"moduleType":5,"operateType":"acOutCfg","params":{"out_voltage":-1,"out_freq":255,"xboost":255,"enabled":0},"version":"1.0"}

ecoflow-mqtt.0
2025-01-07 19:49:39.643 debug [JSON] R331ZEB5SG961433 [update] -> {"addr":0,"cmdFunc":0,"cmdId":0,"id":1903012213333557376,"version":"1.0","timestamp":1736275779,"moduleType":"3","params":{"inv.dcInVol":0,"inv.cfgAcWorkMode":0,"inv.SlowChgWatts":0,"inv.dcInAmp":0,"inv.cfgAcOutFreq":1,"inv.outputWatts":0,"inv.errCode":0,"inv.dcInTemp":0,"inv.invOutFreq":0,"inv.chargerType":0,"inv.reserved":[0,0,0,0,0,0,0,0],"inv.acInAmp":0,"inv.fanState":0,"inv.cfgAcXboost":1,"inv.cfgAcEnabled":0,"inv.outTemp":0,"inv.invType":0,"inv.cfgAcOutVol":230000,"inv.acDipSwitch":0,"inv.acInVol":0,"inv.invOutVol":0,"inv.FastChgWatts":0,"inv.inputWatts":0,"inv.standbyMins":720,"inv.chgPauseFlag":0,"inv.acInFreq":0,"inv.dischargeType":0,"inv.invOutAmp":0,"inv.sysVer":50397310}}

ecoflow-mqtt.0
2025-01-07 19:49:39.165 debug station update .pd.acEnabled off -> on

ecoflow-mqtt.0
2025-01-07 19:49:39.161 debug [JSON] R331ZEB5SG961433 [update] -> {"addr":0,"cmdFunc":0,"cmdId":0,"id":1903012204652658913,"version":"1.0","timestamp":1736275778,"moduleType":"1","params":{"pd.remainTime":-5939,"pd.carWatts":0,"pd.usb1Watts":0,"pd.wattsInSum":0,"pd.typec1Watts":0,"pd.dcOutState":0,"pd.lcdOffSec":300,"pd.soc":80,"pd.acEnabled":1,"pd.wattsOutSum":0,"pd.standbyMin":0}}

ecoflow-mqtt.0
2025-01-07 19:49:38.465 info  state ecoflow-mqtt.0.R331ZEB5SG961433.mppt.cfgAcEnabled changed: true (ack = true)

ecoflow-mqtt.0
2025-01-07 19:49:38.463 debug station update .mppt.cfgAcEnabled false -> true

ecoflow-mqtt.0
2025-01-07 19:49:38.460 debug station update .mppt.carOutVol 10.912 -> 10.146

ecoflow-mqtt.0
2025-01-07 19:49:38.457 debug [JSON] R331ZEB5SG961433 [update] -> {"addr":0,"cmdFunc":0,"cmdId":0,"id":1903012204652658759,"version":"1.0","timestamp":1736275778,"moduleType":"5","params":{"mppt.carState":0,"mppt.carOutVol":10146,"mppt.acStandbyMins":0,"mppt.cfgAcXboost":1,"mppt.carStandbyMin":0,"mppt.cfgAcEnabled":1,"mppt.chgType":0,"mppt.dcChgCurrent":8000,"mppt.inWatts":0,"mppt.beepState":1,"mppt.cfgChgWatts":1200,"mppt.cfgAcOutFreq":50}}

ecoflow-mqtt.0
2025-01-07 19:49:38.114 debug [JSON] R331ZEB5SG961433 [set_reply] -> set w/o params{"id":371591201,"version":"1.0","operateType":"acOutCfg","moduleType":5,"time":206405396,"code":"0","data":{"ack":0}}

ecoflow-mqtt.0
2025-01-07 19:49:37.946 debug [JSON] R331ZEB5SG961433 [set] -> key:undefined {"from":"Android","id":"371591201","isMatter":0,"moduleType":5,"operateType":"acOutCfg","params":{"out_voltage":-1,"out_freq":255,"xboost":255,"enabled":1},"version":"1.0"}

foxthefox, 2025-01-07:
Thanks, I will check it.

rogerbl, 2025-01-08:
with the 1.1.2 version things got better with the River 2 and The River 2 Max.
these commands dont work though:
river 2 max dcoutstate
river 2 dcoutstate
delta 2 cfgAcEnabled

But it looks promising! :-)

rogerbl, 2025-01-08:
I suspect both of my River units don't have controls to turn off/on the USB out ports. Cant see any in the App.
So then its only an issue with the Delta 2 AC control to solve.

rogerbl, 2025-01-08:
I looked up the API of River 2 MAX.
Seems like it supports USB power control after all.
To set it use this :
{
    "id": 123,
    "sn": "R351ZFB4HF6L0030",
    "version": "1.0",
    "moduleType": 1,
    "operateType": "dcOutCfg",
    "params": {
        "enabled": 1
    }
}

foxthefox, 2025-01-15:
We have to verify that we are looking at the correct datapoint.
Are you operating on pd.acAutoOutConfig or on inv.cfgAcEnabled?
pd.acAutoOutConfig can not be written, but the inv.cfgAcEnabled is configured as switch.

Can you try to operate on inv.cfgAcEnabled and check whats the behaviour?

rogerbl, 2025-01-15:
I updated to ver 1.2.0
There is no change when I set inv.cfgAcEnabled to true or false. 

When I set acAutoOutConfig to true AC is enabled.
When I set acAutoOutConfig to false nothing happens. It must be shut down manually on the app.

foxthefox, 2025-01-30:
I have checked my setup and the snippet from the API.

It seems that acOutCfg belongs to module 5 = mppt. Within **mppt** I have the **cfgAcEnabled** as a switch, which toggles acOutCfg.

Can you check if it works at your Delta 2?

The pd.acAutoOutConfig is not turning the switch on/off it is more this "Set AC always on (acAutoOutConfig: 0: disabled; 1: enabled;minAcOutSoc: minimum SoC for turning on "AC always on" )"