foxthefox/ioBroker.ecoflow-mqtt #209: D3P Support

nalditopr, 2025-01-31:
I was able to get some very basic support for the D3P using MQTT and reverse engineered protobuf files for encoding and decoding. Here is what I have so far. https://gist.github.com/nalditopr/426f662d1247e898e847429e220baa6e

I would like to contribute to the project as I have 3 units I can test with. @foxthefox

foxthefox, 2025-01-31:
Perfect.
If you are able to install iobroker and the adapter it would be the easiest way, since I’m have made a config for unsupported devices, which makes easier logging of telegrams. For this I made step by step description, the link can be found in readme in the HA section.

Otherwise it the telegrams from the debug nodes of my original flow would be needed. Maybe there is a need to extend the size and change to base64 encoding. But that would be more comfortable with the adapter and the logs there.

foxthefox, 2025-01-31:
https://github.com/foxthefox/ioBroker.ecoflow-mqtt/blob/main/doc/en/IOB_HA/navi.md

The link to some hints for installation and configuration. Btw the adapter and your node red should not use the same androidId.

foxthefox, 2025-03-09:
@nalditopr 
I managed to have the first implementation of DeltaPro3.
If you could provide data for Delta 3 plus, I could do the same.

jointerCZ, 2025-03-26:
I just got Delta 3 Plus, this solution was the only way how I was able to pull data from it to HA, NodeRed did not worked for me (never used it, my skill problem) and other extensions does not support it.
I will be trying BT addon, but need to get BT ESPHome proxy first.

There is naming typo in the 1.3.0, value bmsMinCellVol is named as "Minimum temperature of the main battery".
Also log shows:
....RuntimePropertyUpload.plugInInfoAcOutVol" has value "238.0778350830078" greater than max "60"

Too early to provide more feedback.

foxthefox, 2025-03-27:
@jointerCZ 
Thanks for the feedback, I have changed the wrong naming and the max range of the AC voltage.

Plug in info pv chg amp max should be around 10A max or even less. We have to observe these values.
It could also a missing multiplication of 0.1 to come to 6.1A.

jointerCZ, 2025-03-27:
Solar amp limit is probably right, without XT60i cable, it should be 8A. Only with XT60i it might change to 15A.
Plugging in original PV cable does nothing (but I do not have any panel yet, just a cable).

Sorry, I looked into incorrect line!
plug_in_info_pv_chg_vol_max - 61V, 60V is official limit so no decimal error, sorry for the confusion.
plug_in_info_pv_chg_amp_max - 8A
But
plugInInfoPv2ChgVolMax and plugInInfoPv2DcAmpMax have both 0 (even in IOBroker) where the "1st channel" does have limits.
No big issue, either it is not reported or not read correctly, nobody will probably need to use this ever anyway.

Thanks for the fix!

foxthefox, 2025-03-27:
OK, thanks for feedback.

BTW. the changed datapoint. settings are not getting active after update without the following actions

Stop adapter -> delete conecrend datapoint(s) from object tree -> start adapter

jointerCZ, 2025-03-27:
Not sure how to update, so I removed the adapter and reinstalled it.
But the name is still wrong.
These values are max and min from BMS, monitoring cell balance, not overall minimum of the whole pack.
[image]

Also, the values "OK?" - do you need to confirm those?

foxthefox, 2025-03-28:
@jointerCZ 
The name mismatch min temp vs. min cell voltage is corrected, that is the only thing I did.
Basically the other names are created from the datapoint. It will take some time to make for all datapoints nice names from the abbreviation (around 200). The already beautified names are comming from matches with the official API. 
I focussed first on creating the communication to the devices and the names come later.
With another feedback, now it is clear that there is a difference between cms and bms.

Since the API is not known, especially the state information is not known how to map an incomming number to a meaningful wording. So if you discover something with "?" at the end, then it indicates a first guess of the state. Same applies to "not part of array", where the incomming value is not part of the code and can not be transferred into an "guessed" or validated wording.
If you find such occasions and can verify the state with the condition of the device then I can implement the right wording.