foxthefox/ioBroker.ecoflow-mqtt #235: Powerstream inverter_heartbeat_supply_priority not being set in v1.3.0

walderston, 2025-03-20:
**Describe the bug**  
Updated iobroker adapter to v1.3.0 and noticed that the powerstream isn't functioning correctly. Seems to only impact the 'switch.xxx_inverter_heartbeat_supply_priority' value within HA.

**Expected behavior**  
The toggle of 'switch.xxx_inverter_heartbeat_supply_priority' within HA should toggle the 'power supply mode' within the EF app

**Screenshots & Logfiles**  
<html>
<body>

ecoflow-mqtt.0 | 2025-03-20 00:29:31.589 | debug | missing msgBuf for sending
-- | -- | -- | --
ecoflow-mqtt.0 | 2025-03-20 00:29:31.587 | info | (ack=false) ->cmd : channel inverter_heartbeat state supplyPriority
ecoflow-mqtt.0 | 2025-03-20 00:29:31.587 | info | state ecoflow-mqtt.0.HW51ZKH4SF5Q0072.inverter_heartbeat.supplyPriority changed: false (ack = false)
ecoflow-mqtt.0 | 2025-03-20 00:29:31.585 | silly | States user redis pmessage ecoflow-mqtt.0.HW51ZKH4SF5Q0072.inverter_heartbeat.supplyPriority/ecoflow-mqtt.0.HW51ZKH4SF5Q0072.inverter_heartbeat.supplyPriority:{"val":false,"ack":false,"ts":1742430571584,"q":0,"from":"system.adapter.ecoflow-mqtt.0","user":"system.user.admin","lc":1742430571584}

</body>
</html>

foxthefox, 2025-03-20:
Thank for the feedback.
I restructured some code and the "missing msgBuf" indicates that I have forgotten something in this work.
I will correct it on weekend.

Sorry for inconvenience.

walderston, 2025-03-20:
Thank you .. I've rolled back to 1.2.2 for now so no pressure.

foxthefox, 2025-03-23:
You may want to check the version 1.3.0 from git again.
The "missing msgBuf" should be solved.