foxthefox/ioBroker.ecoflow-mqtt #238: Missing option to change AC Charge Power with Delta 3 Plus

D4nb4, 2025-03-30:
I don't think this is a bug at all but just to make sure it isn't (and I didn't mess up things again):
I got a Delta 3 Plus connected to ioBroker and Home Assistant. So far, all works great but I miss  
some functions that I think should also work or at least, would be very cool to have.
Is it just me that I miss these or did I (as usually) successfully messed up things again during setup?

- Enable/Disable AC Output
- Enable/Disable 12V DC
- Enable/Disable USB
- Change AC Charge Power Limit (this would be great since I could use HA to change the AC Charge Slider depending on available solar power.)

**Expected behavior**  
Since these settings are also adjustable via the ecoflow app, I hoped they would be available via iobroker as well.

**Versions:**  
 - Adapter version: v1.3.0
 - JS-Controller version: 7.0.6
 - Node version: <node-version> 20.19.0
 - Operating system: Proxmox -> VM -> Alpine -> Docker -> ioBroker/HA

foxthefox, 2025-03-30:
Indeed these commands are not implemented yet, because I did not have sample telegrams. 
It would be perfect if you could provide these.
For this action the adapter has to be switched in debug mode, on top of adapter config page info-> debug.
In the first tab of adapter config, please tick msgCmd and msgSetGet. On second tab, tick debugEnable left to haEnable. 
After automatic restart you should do all desired commands from the app (on/off, multiple values).
Then I would need the log of that timeframe posted. You can zip it.
Also it would be good if you can note down the ranges in the App of analog values.

D4nb4, 2025-03-30:
Ok, I hope I did it the right way, at least I got alot of dbug messages in the log.

So far I hit the:
AC Output button which is On/Off
12V DC also On/Off
USB On/Off

AC charge limit which goes from 100 to 1500 with a step of 100

I also changed the max discharge & max charge values (slider) of the battery which goes from 0% to 100%. Recommended settings by Ecoflow is 10%-90%

Let me know if you need anything more.

[LOG_Delta_3_Plus.zip](https://github.com/user-attachments/files/19525021/LOG_Delta_3_Plus.zip)

foxthefox, 2025-03-30:
Thanks a lot. I will check it today evening and report.

foxthefox, 2025-03-30:
@D4nb4 
I checked the log.
Everything what I need to know is inside. All done correctly.
I will start tomorrow with the code changes.

D4nb4, 2025-03-30:
Awesome! Thanks!

foxthefox, 2025-03-31:
According to the data in the telegram to ac charge power, there should also a selection in the EF-App for the mode of AC charging. Something like self-defined power level, battery-optimal, silence?! 
Is this mode selector available in the EF-App?

D4nb4, 2025-03-31:
Not really. See the attachments for things availabel. There's a storm protection mode, a plan to save money and load at special tarifs, which we do not have and an ac inverter mode, to switch on a generator if load is below an adjustable level currently set to 35%, but I'm not using this.

[image]
[image]
[image]

D4nb4, 2025-03-31:
Ok, found oit, you can tick user defined, optimal and silent, as you mentioned.

[image]

foxthefox, 2025-03-31:
Thanks a lot.

D4nb4, 2025-03-31:
I also got many errors on this:

State value to set for "ecoflow-mqtt.0.MyDeltaSerial.RuntimePropertyUpload.plugInInfoAcOutVol" has value "233.29647827148438" greater than max "60"

This seems to be the AC Out Voltage, which can of course be as high as 240V

foxthefox, 2025-03-31:
The "min 400" I already discovered yesterday :-)
The other values I will take care in the upcomming 1.3.1.

Thanks for your observation and hints.

D4nb4, 2025-04-01:
I did download the new version and the AC,12V and USB switches are available.
However, switching 12V or USB on off in HA works great, switching AC on off works also, but when switching off it does switch off the AC but in HA it shown status as on again, although in ef app it definately off

foxthefox, 2025-04-01:
The current 1.3.1 in git is the first version, a very early stage.
But thanks for the feedback.

foxthefox, 2025-04-06:
Please have a look to the final 1.3.1 at git or npm.
Hopefully everythink is working properly.

D4nb4, 2025-04-06:
Great, just tested some new features ...

- AC Charge Mode works via HA
- AC Charge Limit (userdefined) does not change anything at all via HA
- Battery Charge/Discharge Limits work via HA, but the value (next to the slider) is not updated. 
- Display brightness works, but the scale seems to be messed up. Max value seems to be 39.0625% (min = 0%)

Thanks for your work on this!

D4nb4, 2025-04-06:
I did delete the complete ecoflow.mqtt tree

Sent via mobile app!
Best regards, Daniel B.
----------------------------------------

*Von: *foxthefox ***@***.***>
*An: *foxthefox/ioBroker.ecoflow-mqtt ***@***.***>
*Kopie: *Danba ***@***.***>; Mention ***@***.***>
*Datum: *06.04.2025 17:16:04
*Betreff: *Re: [foxthefox/ioBroker.ecoflow-mqtt] Missing option to change AC Charge Power with Delta 3 Plus (Issue #238)

Can you repeat the test after stopping the adapter, deleting the objecttree of DisplayPropertyUpload and start the adapter?
Can you also crosscheck with the datapoints in iobroker plugInInfoAcInChgMax, cmsMaxChgSoc and cmsMinDsgSoc?
In my environment the HA changes are transferred to iobroker and in the iobroker log also the set command is done.

Displaybrightness needs an update (will be 1.3.2)

—

   

Can you repeat the test after stopping the adapter, deleting the objecttree of DisplayPropertyUpload and start the adapter?
Can you also crosscheck with the datapoints in iobroker plugInInfoAcInChgMax, cmsMaxChgSoc and cmsMinDsgSoc?
In my environment the HA changes are transferred to iobroker and in the iobroker log also the set command is done.

Displaybrightness needs an update (will be 1.3.2)

foxthefox, 2025-04-06:
in first picture cmsBatPowMax is read only, so it is not to be changed by commands.

second picture shows some custom screen, also the blue node says 300, where it proves that it is 300 and the slider has some strange reaction.

can you go to the mqtt integration and check the behaviour there.
it should look like this:
[image]

maybe a reload of the browser is the problem

D4nb4, 2025-04-06:
[image]
It stays also at 100 W; Reload of the page does not change anything

foxthefox, 2025-04-06:
Looks a little bit strange to me.
What does it mean it stays at 100W? If you slide it for adjustment, then it should move and stay at its new position.
Or does it mean it has returned to 100W after sliding to another value?
Or does it mean that you changed the value within the automation and the update is missing in the slider?

Just now received the Edit:
cmsBattPowInMax is not to be adjusted, it is not a command and therefore altering the value will not have an impact.
It might be updated based on other alogorithm inside the Delta3Plus, but it is not because of cmd send to the device.

You can not tweak sensors, you can only adjust numbers.
So changing the Max AC charging within an automation would be:
number.yourserial_displaypropertyupload_maximum_ac_input_power_for_charging

D4nb4, 2025-04-06:
Yes,the slider works perfectly and stays at the new position. Just the entity value does not fit the new position.

If I move the slider to 300 W, the slider stays there. 
However, on the mushroom number card, as well as on other cards, there is also the raw value available, which stays at the old value, in this case 100. EF App ist also on 300 and all seems good, just not the value that I used within the automation. 
I have never before worked with numbers in automation but I think the raw vale should be the value that the slider points to.

I use the number.yourserial_displaypropertyupload_maximum_ac_input_power_for_charging in an automation, to solve if I have to rise or lower the max charge limit depending on the new available/missing overhead of the current solar power and since this value, for what reason ever, does not update to the new AC max charge limit slider position (although it should be the same value), I'm unable to figure out what to do in the automation.

Please keep an eye on the "Laden"-Gauge, whis comes from the sensor.p351zeh4pgbm0753_displaypropertyupload_cms_batt_pow_in_max and fits the slider position accurately with a small delay

Start: AC Max Charge Limit at 100W - The scale starts at 100 so there is no visible slider bar at all

[image]

200W: Rising the slider to 200 W - Value stays at 100 - Laden-Gauge rises to 200 also

[image]

700W: Rising slider to 700W - Value still at 100 - Laden-Gauge rises to 700

[image]

For the mushroom-number-card - absolutely no magic but why does the value not fit the slider position?

type: custom:mushroom-number-card
entity: >-
  number.p351zeh4pgbm0753_displaypropertyupload_maximum_ac_input_power_for_charging
name: AC Max Charge
icon: ""
icon_color: yellow
icon_type: none
fill_container: true
layout: horizontal

D4nb4, 2025-04-06:
That was magic - It updates now

foxthefox, 2025-04-06:
If you find my work good, give me a star:

[image]

I think we can close the issue now.

closed 2025-04-06 by D4nb4