foxthefox/ioBroker.ecoflow-mqtt #253: Missing Slave check button for Delta 3 Plus

D4nb4, 2025-04-11:
In the latest version v1.3.2 I 'm missing the chance to add a extended battery to my Delta 3 Plus.
There is no chackmark for slave 1 or slave 2.
The Delta 3 Plus can have a maximum of 1 extended battery.

foxthefox, 2025-04-11:
It' correct and it even there will be a checkbox, there will no difference in the moment :-(
I can add it, so it looks nicely.

foxthefox, 2025-07-11:
I have created a preview of the next version.
Could you please check it.

it must be installed via expert mode -> custom, described here:
https://github.com/foxthefox/ioBroker.ecoflow-mqtt/blob/1.4.0/doc/en/IOB_HA/adapter.md#new-version-for-testing-on-github-not-main-branch

after installation you need to:
- stop the adapter
- delete the llcbusvol datapoint
- start the adapter