foxthefox/ioBroker.ecoflow-mqtt #254: Delta 3 Plus working with HA - missing energy settings

phoniclynx, 2025-04-12:
Finally worked out how to install iobroker, HA and MQTT and got it all working together nicely. I see data coming into HA through MQTT and IOBroker and the data seems to be what I see in the app.

The problem becomes when I go to add the values into the Energy Board Tab and the ONLY thing I can place in it is the Maximum Value for the size of the battery.. ie.. "sensor.delta3plus_livingroom_cms_batt_full_energy" which is "1024: I am assuming Wh as there is no value assigned to it.

What I think needs to be able to go in are "sensor.delta3plus_livingroom_total_input_power" and "sensor.delta3plus_livingroom_total_output_power" for battery In and Out, and possibly the DC1 and DC2 values to keep track of the solar. "sensor.delta3plus_livingroom_pow_get_pv" and "sensor.delta3plus_livingroom_pow_get_pv2" I think.. I can't tell as it's midnight as I'm writing this.

[image]

(ignore the crossed out ones as they were from testing with node-red)

foxthefox, 2025-04-12:
I think that only energy values should be connected to the energy dashboard, not power values. I will check if there is something missing, to make them compatible. Something like „total increasing“ is probably missing as additional property. CmsBattFullEnergy is only showing the capacity of the battery and is not the value for the dashboard, because constant.

phoniclynx, 2025-04-13:
I'm thinking of things like the Solar Inputs:

[image]

They just don't show up as options to select from because the entity isn't quite defined properly?

foxthefox, 2025-04-13:
I have checked the situation. In the current set of data is no entitity which can be used for the energy dashboard.
There should be a datastructure incomming with the telegrams, which contain also energy data, but so far I have not analysed it and therfore it is not available as datapoints.
This requires some enhancement work, to make that visible.
But then there will be something to be used in the energy dashboard.
I will keep you updated on the progress of it.

phoniclynx, 2025-07-01:
Have you had any luck with this?