foxthefox/ioBroker.ecoflow-mqtt #255: River 3 series Support

lbatalha, 2025-04-18:
Following up from https://github.com/tolwi/hassio-ecoflow-cloud/issues/465#issuecomment-2813687611 since that is not the right place for this discussion.

I have setup iobroker with ecoflow-mqtt and a River 3 ("UPS Version") to the unsupported devices with debug enabled

Is there anything else I can do on my end to help adding support for this device?

foxthefox, 2025-04-18:
Thank you for continuing here, at the right place.
I think from your previous node-red captures I have enough to start with.

What would interest me are the available commands or settings from the APP.
Which ones are available and what are the possible values (ranges or modes)?

What you can also do is to change/toggle these settings/commands from the APP and then post the portion of the log of that timeframe. There should be several messages then with get/get_reply and set/set_reply.

lbatalha, 2025-05-01:
FYI the BT local hass integration already has support for River 3 and Delta 3, etc..
They have their protobuff definition files here: https://github.com/rabits/ha-ef-ble/blob/main/custom_components/ef_ble/eflib/pb/pr705_pb2.py, hopefully that helps you set it up for iobroker (I am not using iobroker atm)

foxthefox, 2025-05-02:
@lbatalha 
Thank you very much for the hint. In the end not exactly this file, but the repo of the contributor to ha-ef-ble has the information I can digest.

lbatalha, 2025-05-02:
Someone also found this for me https://github.com/maodijim/ecoflow_exporter/blob/e9d3e68121525eedfc6f1da15ed7e6226859271e/proto/proto/pr705.proto
Appears to be a base .proto file for the same device

kvolkers, 2025-05-08:
@foxthefox I also have the RIVER 3 UPS version. If there's any missing information I can help provide let me know!

foxthefox, 2025-05-09:
Thanks. I am working on the implementation, with the above mentioned other repos, I have a good idea.

Where I would need help is at the commands from the App.
- Which commands are available?
- What are the possible settings or ranges (and the steps when adjusting within the range) for each command

Most likely I do not need an capturing of commands right now, since the commands should be the same as at the other gen3 devices. But verification after implementation would be needed.

kvolkers, 2025-05-09:
Here's what I could find quickly. I can look at anything in more detail if needed.

```
RIVER 3 UPS

Home screen:
- Main on/off toggle
- 230V AC on/off toggle
- 12V DC on/off toggle

General settings:
- Device name -> input field

Device settings:
- Charge & discharge limits -> double slider 0-100%, steps of 1%, recommended 20%-80%
- Backup reserve -> on/off toggle
    - When on -> slider controlling both Backup reserve and Solar only simultaneously, reserve:5%(solar:55%)-reserve:60%(solar:0%), steps of 1%
- Scheduled task -> AC charging, Solar/Car charging, AC discharging, 12V DC discharging
- Input settings
    - AC charging speed -> Custom vs Silent charging (80W) toggle
        - When custom -> slider 50W-305W, steps of 1W
    - DC charging mode -> Auto vs Car recharging vs Solar recharging toggle
    - Car input -> slider 4A-8A, steps of 2A
- Output settings
    - X-Boost -> on/off toggle
    - Output port memory -> on/off toggle
- Low battery reminder -> on/off toggle
- Device timeout -> 30min, 1hour, 2hour, 3hour, 4hour, 12hour, 24hour, Never toggle
- LCD screen timeout -> 10s, 30s, 1min, 5min, 30min, Never toggle
- AC timeout -> 30min, 1hour, 2hour, 3hour, 4hour, 12hour, 24hour, Never toggle
- Time zone
    - Set automatically -> on/off toggle
        - When off -> Time zone input e.g. Europe/Amsterdam(?)
- Wi-Fi details
    - Auto join -> on/off toggle
    - Change Wi-Fi button(?)
```

foxthefox, 2025-05-09:
Perfect, that helps a lot.

kvolkers, 2025-08-19:
I've updated the adapter to 1.4.1 and restarted the instance. But I'm new to this so not sure if I'm missing something. Only new options I'm seeing are setDp3 > cfgAcOutOpen and cfgDc12vOutOpen.

kvolkers, 2025-08-20:
Nice. Looks good! I can see all mentioned objects. Changed a few values through Home Assistant and I see the changes reflected in the app. Very cool.

Missing the energy_backup_enable and soc=xxx. Personally this is the one I'm in need of. I've thoroughly went through the app looking for any remaining functionality.

- Backup reserve enable + soc value
- DC charging mode (Auto, Car recharging, Solar recharging)
- Car input (8A, 6A, 4A)
- Low battery reminder (not sure if this is a relevant setting)

Sent you additional logs after toggling all of the above.

I'm pretty sure that's everything. Maybe a device on/off (?) but it'd give me some trouble to turn that off haha.

foxthefox, 2025-08-20:
version 1.4.2 is online (git +npm), hopefully everything is now working.
Please delete the object tree when adapter is stopped. 

The change in backupReserveSoc enables the mode!

kvolkers, 2025-08-21:
Updated. So, backup reserve function switch to turn it off. But to turn it on; set it to a value? I see, works even if a little confusing haha.

In HA this backup reserve slider is 0-30%. In the app it's 5-100%. When setting it to a higher value than 30% HA also still displays it as 30%. Do you need debug logging for this or is it a simple mapping?

Other than that everything seems to work great.

foxthefox, 2025-08-21:
The backupreserve switch should work in both ways. 
The soc value I will check, but I thought to have 5…100 there. And it seems causing that warning message, because in App it was set to 60.
The change in backupSoc turns it on, correct?
I think no logs needed.

foxthefox, 2025-09-02:
@kvolkers 
With version 1.4.3 both issues (range and bidirectional) should be solved.
Can you check it?

foxthefox, 2025-09-03:
ack= false is command, ack=true is feedback of value
So when changing the value (as a cmd) it is normal to have ack=false

So I think I can close the issue
Thanks for the help.