foxthefox/ioBroker.ecoflow-mqtt #256: Delta 3 (not 3+) Support

AlphaZuluNRW, 2025-04-18:
It would be nice if you could use the Delta 3 (without Plus) in HomeAssistant - possibly via the "detour" of ioBroker.

@foxthefox , I looked in the logfile and found:

[image]

The serial number is definitely correct (I've deleted it in the screenshot).

foxthefox, 2025-04-19:
Hmm, strange. It says that the connection is established and subscribing the topics throws an error.
You could try the following:
- delete the row in unknown devices
- put your device into power stations as Delta3Plus, this should come nearest and it might work for most of the datapoints

The object structure stays empty for unknown devices. That is intended.

Only to have it verified, you have only one instance running and the “AndroidID” is not used elsewhere?!

foxthefox, 2025-04-19:
And you may try to update to 1.3.1 or 1.3.2 version.

AlphaZuluNRW, 2025-04-19:
I only use the Android ID here, since the integration with HomeAssistant uses a different login.

I can't see the Delta3 Plus at all. Could this be due to the version? I don't see any update options from version 1.2.2 (Instanzeinstellungen: ecoflow-mqtt.0 v1.2.2):

[image]

foxthefox, 2025-04-19:
You need to activate expert mode in adapter page, then there is the octacat icon one there you can use non or GitHub to install 1.3.2. In the reader of the adapter is a link under HA section with further picturized details.

foxthefox, 2025-04-19:
The details are here

https://github.com/foxthefox/ioBroker.ecoflow-mqtt/blob/main/doc/en/IOB_HA/adapter.md

foxthefox, 2025-04-19:
So the version was too old, better to have it already at 1.3.2 😎
Yes I would need the log from the Delta3 since in most cases there are slightly differences and this would be good to know and take care.

My variant is as fast as the devices are sending data, so in most cases much faster. Since the official API needs polling in the range of seconds.

AlphaZuluNRW, 2025-04-19:
Which log file do you need?

I've also gotten some data into HomeAssistant. However, I'm having trouble with some of it not being sent. For example, the "State of Health" of Delta 3 is at 100%, which is also displayed in ioBroker. Since the value isn't changing (at least not yet), it's apparently not being sent via MQTT, as the entry isn't visible in the MQTT Explorer either. Is there any way to force even values ​​that don't change to be sent?

I also noticed that there doesn't seem to be a value for the charge cycle. At least it's included in the HA integration for my River 2.

foxthefox, 2025-04-20:
I have a load reducing algorithm inside the adapter. Data is only sent when the value is different to the stored one. So only after receiving new data with different value it will be stored in iobroker as well as further transferred to HA. Otherwise the devices create a relative high traffic for nothing.
At the start of the adapter all values are transferred initially to HA.

SoH should not change, otherwise you would have a problem with your battery.

Hopefully that explains some of your observations.

Since you are using the Delta 3 with the Delta 3+ implementation not all values are expected to be available at the device (the difference between plus and non-plus).

I would need the log from iobroker only. 
But there need to be also some settings made (see logging in the previous link). 
The adapter needs to be in debug mode. And then the logging of the device must be tickmarked (second tab beside the HA checkbox) as well as msgSet/Get, msgCmd and msgUpdate and the first tab.

AlphaZuluNRW, 2025-04-20:
Even though I haven't changed anything all day, HA suddenly stopped receiving data. Everything just says "not available."

ioBroker also no longer has any current data. Could this be the same problem as with the HA integration before it was switched to the official Ecoflow access?

If I access the Ecoflow app on my smartphone, any changed data is immediately displayed in ioBroker and then also in HomeAssistant...

foxthefox, 2025-04-20:
It is a similar problem which can occur. But here I think it is even different. Since the “not available” comes only if there is no Mqtt connection. But usually this gets not lost, only no data.

AlphaZuluNRW, 2025-04-20:
And I noticed something else. If I change the maximum wattage for charging from the grid in HA via an automation (because surplus energy is available), the value isn't set in ioBroker. If I change it via the app, the correct value is immediately displayed in ioBroker AND HA.

However, the change appears in the log file

Changing it in HA:
`state ecoflow-mqtt.0.P2##########.DisplayPropertyUpload.plugInInfoAcInChgPowMax changed: 1000 (ack = false)`

Changing in the Ecoflow-app:
`state ecoflow-mqtt.0.P2##########.DisplayPropertyUpload.plugInInfoAcInChgPowMax changed: 900 (ack = true)`

The difference is that "ack = true," which has an effect. "ack = false" is ignored.

foxthefox, 2025-04-21:
Ack=true is a feedback and this value change is confirming a setting. Ack=false is a command. So when in HA the value is changed it is incoming to iobroker as ack=false which triggers the adapter to send a command to the cloud. The device sends a special ack telegram which then will be translated as ack=true which triggers the adapter to only store that value.

Even having the D3 running as D3+ I think we should do the same initial steps as with all other devices. Gather and evaluate the telegrams, make a device integration and then verify that it works. It could be the case the D3 is slightly different and only most things are identical.

AlphaZuluNRW, 2025-04-21:
OK, then I'll try generating the log files.

You wrote:

> The adapter needs to be in debug mode. And then the logging of the device must be tickmarked (second tab beside the HA checkbox) as well as msgSet/Get, msgCmd and msgUpdate and the first tab.

Stupid question: How do I set debug mode for the adapter?

When I call the adapter, I can't configure anything (or I can't see it).

In the instance, I checked the following in the section "Credentials & Debug Settings":
[image]

Under "Device(s) Configuration" I should probably check this:
[image]

AlphaZuluNRW, 2025-04-21:
Is there anything missing (apart from the adapter's debug mode)?

foxthefox, 2025-04-21:
no, only the debug mode.
you need to go to instances and then to the ecoflow-mqtt config (wrench symbol)
and in the header you can change the mode, see here:

[image]

AlphaZuluNRW, 2025-04-21:
So, debug is now active.

How long should I leave it running?

foxthefox, 2025-04-22:
The log must not be very long, some minutes are usually OK. It should contain some get_reply and update telegrams. You may also include already some commands. For the capturing period you should go into the APP and go to the devices and subsequent pages. You can also zip it. You can send it to foxthefox(at)wysiwis.net.

AlphaZuluNRW, 2025-04-22:
The log file is on its way to you via email. 😀

AlphaZuluNRW, 2025-04-22:
Of course, in my haste, I overlooked this: "For the capturing period, you should go into the APP and go to the devices and subsequent pages."

So I'll do it again now...

AlphaZuluNRW, 2025-04-22:
Done! 😁

AlphaZuluNRW, 2025-04-22:
Thank you very much!

I've adjusted the IDs and the ANDROID string, but apparently no connection is established:

[image]

Is there anything missing? I didn't see that I needed to enter the MQTT user name somewhere.

foxthefox, 2025-04-23:
What you need to make the flow working is a different AndroidID than used in the adapter.
If you have before not used the userId, that would explain why the set/gets are not subscribed.
Strangely here it does not have connection to the server.
What you need in the topics is the mqttuserID (the one with 19 digits or so), the mqttUserName (the mone with app-c....) is for the broker settings.

What you could do:
- stop the adapter
- go here https://energychain.github.io/site_ecoflow_mqtt_credentials/ or here as script https://github.com/mmiller7/ecoflow-withoutflow/blob/main/cloud-mqtt/ecoflow_get_mqtt_login.sh and get your credentials
- use the data from that site or script in the node-red flow (mqtt settings as well for the topics, set/gets have mqttuserId and serialnumber inside)
- check what is working

Good luck!

AlphaZuluNRW, 2025-04-26:
Bingo! The solution was the user ID. Before, I hadn't just the numbers there, but the username...

Now it's working perfectly!

Thanks a lot!

foxthefox, 2025-04-26:
After this is working we should now look to the adapter that there are no more errors on the topic subscription.
If that is gone you can produce a log with set/get inside.

AlphaZuluNRW, 2025-04-28:
I don't see any error messages in the log file.

I've now integrated the three plugs in addition to the "Delta 3", because your approach delivers significantly faster data transfer. The two other Power Stations from the 2 Series ("River 2" and "River 2 Pro") will also be moved when the opportunity arises.

The automatic adjustment of the maximum AC input also works perfectly now. I just noticed that only whole increments of 100 are applied. For the other two power stations, 100 is the lowest value, but after that, adjustments are possible in increments of one watt. I don't know if this is due to the "2 Series" or the integration via Home Assistant or (with the "Delta 3") ioBroker.

Do you need any log files to analyze any data that might not yet be included? I'd also be interested in the number of charging cycles so I can have that for all power stations, which will then be displayed in Home Assistant like this:

[image]

foxthefox, 2025-07-11:
I have created a preview of the next version.
Could you please check it.

it must be installed via expert mode -> custom, described here:
https://github.com/foxthefox/ioBroker.ecoflow-mqtt/blob/1.4.0/doc/en/IOB_HA/adapter.md#new-version-for-testing-on-github-not-main-branch

foxthefox, 2025-09-13:
Version 1.4.5 is available in the meantime, any open items?

foxthefox, 2025-11-25:
Most likely we can close this item, if there is something to be done, please reopen it.