foxthefox/ioBroker.ecoflow-mqtt #258: Wave 3 support

beyondgrave, 2025-04-24:
I have the new Wave 3 due to replacement of faulty Wave 2. I have iobroker with the ecoflow adapter set up. I made sure to first add a supported device (Delta 2 Max) to make sure everything was set up properly. 

Then I removed those and added the Wave 3 to unsupported devices. But under "Type" there's no option for Wave 3 so I just left it at none. I'm not showing anything in the logs outside of the initial line showing the serial number of the Wave 3. 

I tried setting device type to River 3 and Delta 3 but same results. 

Please let me know what info you need to be able to add it properly. Thanks!

foxthefox, 2025-04-24:
The unknown section does not have wave3 as a selection option, using one of the types is OK since it is only for indication in the log.
You should put the adapter in debug mode and observe the log, when using the EF app. There should be entries for the received data. What you also should verify is the start of the adapter, that there are no faults in the subscription of topics.
Basically I would need the incomming telegrams for evaluation.

beyondgrave, 2025-04-24:
Ok, I have debug info now. Should I just attach here? Outside of masking the SN, I'm not sure if there's anything else I need to redact before posting.

foxthefox, 2025-04-24:
I think you can post it here, when you have redacted the serial in the log file. You can also zip it, then it is not polluting the size of the comment.

beyondgrave, 2025-04-24:
I didn't run the debug for very long. Let me know if this is at all useful. 

[iobroker.2025-04-24.zip](https://github.com/user-attachments/files/19898954/iobroker.2025-04-24.zip)

foxthefox, 2025-04-25:
It is for sure helpful and it contains data, with some luck some data matches to existing structures.
I saw already a command, did you specifically invoke that or was it something in the backround in the App? Something was switched ON probably?!

beyondgrave, 2025-04-25:
I think it was running the auto mode which is a new feature where you set a low and high temp and it tries to maintain that by switching between cooling and heating. Either that or I put it in standby in the middle of debugging.

Sorry, I've been doing a bunch of testing with it because they only swapped out my faulty Wave 2 and not the battery. And I'm finding that the Wave 3 isn't as compatible with the Wave 2 battery as they claimed. So I can't really test any of the battery functions since it's not being properly detected and all the battery specific menu items are missing.

I'm thinking they're going to be pushing out a firmware update pretty soon.

foxthefox, 2025-04-25:
Never mind, I am right now checking the telegram data and at least there is some usefull inside.
And also unknown things, where an educated guess is needed.
Do you have the device running with °C or °F? since there are values in the 20s or in 60s, where I would guess that this is temperature values.

beyondgrave, 2025-04-25:
Running in °F

There are a more options in the app for the Wave 3 vs Wave 2 so maybe those are the unknowns. The auto mode and pet mode are new. There's also 5 fan speeds vs 3 of the Wave 2.