foxthefox/ioBroker.ecoflow-mqtt #264: Delta 2 + Slave-Battery --> States of Slave are not transfered into object

manuelwolfcgm, 2025-05-03:
Hello, 

When using an EcoFlow Delta 2 in combination with an additional slave-battery, the states of the slave are not transferred into the object.

States of the master (in the bmsMaster-channel) are all set correct.
States of the slave (in the bmsSlave1-channel) are created, but not filled.

Investigating the log-files, it seems that the states are provided in e.g. "bms_slave.actSoc" instead if the expected state in the created channel "bmsSlave1".
The Delta 2 only has one single slave-port. Maybe this is the issue that the API provides the state not in the expected bmsSlave1-channel?

`payload from R331FEB4ZHxxxxxx was: {"addr":0,"cmdFunc":0,"cmdId":0,"id":1989125825489928400,"version":"1.0","timestamp":1746300723,"moduleType":"4","params":{"bms_slave.outputWatts":0,"bms_slave.dsgCap":521,"bms_slave.f32ShowSoc":98.49,"bms_slave.allErrCode":0,"bms_slave.cellVol":[],"bms_slave.mosState":3,"bms_slave.minCellVol":3327,"bms_slave.cellTemp":[],"bms_slave.actSoc":98,"bms_slave.amp":-68,"bms_slave.sysState":2,"bms_slave.soc":98,"bms_slave.maxCellVol":3329,"bms_slave.inputWatts":0}}`

[image]

**Versions:**  
 - Adapter version: 1.3.2
 - JS-Controller version: 7.0.6
 - Node version: 20.19.1
 - Operating system: linux

If the bug is...on my side of the screen...then...sorry ^^ Please advise how to proceed then :-/
If I can assist in providing further data - please tell as well.

Kind regards & thank you for your effort!

foxthefox, 2025-05-03:
Thanks for the feedback.
Interestingly I the sample data I had, was with "bms_slave_bmsSlaveStatus_1." so "bms_slave." is not in the code yet.
I will make the needed addition.

manuelwolfcgm, 2025-05-03:
@foxthefox  No hurry, please ;-)  For a wild guess: when the D2M-data is tagged with e.g. _1 for the display-facing port...maybe the issue is really based on the single-slave-port of the D2. Anyway: appreciate your fast reply! Please reach out to me if there is any further data  I can provide!

foxthefox, 2025-05-03:
The D2M has obviously a complete different tag, it ends with _bmsSlaveStatus_1 or _bmsSlaveStatus_2 after the bms_slave. The the slave battery is clearly distinguished, that is not needed when only one port is there.
I have already made the change, it will be part of the next version.
I will trigger here for testing.

foxthefox, 2025-07-11:
I have created a preview of the next version.
Could you please check it.

it must be installed via expert mode -> custom, described here:
https://github.com/foxthefox/ioBroker.ecoflow-mqtt/blob/1.4.0/doc/en/IOB_HA/adapter.md#new-version-for-testing-on-github-not-main-branch