foxthefox/ioBroker.ecoflow-mqtt #273: Using a local MQTT server

adeebee2, 2025-06-05:
Hello,

I wasn't able to find any reference to it in the documentation but is it possible to use an entirely local MQTT server?

I have my Delta 3 Plus hooked up to my local MQTT server and I can see the messages below (via MQTT Explorer) but from what i can tell even when i connect ioBroker to the same MQTT server its not subscribing/sending the right messages to actually do anything.

[image]

From digging around a bit im assuming the ecoflow-mqtt ioBroker adapter is reliant on some 'special sauce' on the Ecoflow server end?

Kind regards and thank you for this awesome project!

foxthefox, 2025-06-05:
you might want to have a look here:
https://github.com/tomvd/local-powerstream

The APP and the adapter are using other topics than the devices. That means that the adapter in its current state can not communicate directly to the devices, even beeing both on the local server.
There is a sort of bridge between the topics on the EF-server.
However the above link shows that with some ACK telegrams the devices keep communicating and renaming of topics would make it possible to interact on a local server.
Something for the future to have an alternative connection via local MQTT. Most likely the normal user will not be able to reroute and establish a local server.

adeebee2, 2025-06-06:
Thanks for pointing me in that direction I will have a bit of a play around.
I searched around so much but kept hitting solutions reliant on the EF MQTT server. Surprised that one didn't come up.

Yeah I figured it wouldn't be the most user friendly solution unless routers start making it easier for people to put in DNS overrides.
Maybe if stuff like Home Assistant and pi-hole gets a bit more popular it will scale better for users to just override the EF DNS record.

Sounds awesome as a future enhancement, keep up the great work!

foxthefox, 2025-06-06:
Are you using the local MQTT server completely local or as an intermediate one. If intermediate, how did you manage that the rerouting is not for the final route to EF server?

adeebee2, 2025-06-07:
At the moment wile I am playing around with things I'm using an entirely local MQTT server but then some really quick and dirty Node Red flows to proxy ideally the bare minimum messages to/from the EF server.

[image]