foxthefox/ioBroker.ecoflow-mqtt #274: After a Home Assistant Restart, I have to restart the Ecoflow ioBroker addon exactly twice for values to show

ampersandru, 2025-06-05:
**Describe the bug**  
Whenever I restart Home Assistant, the values of my Delta 3 plus no longer update. I have to restart the Ecoflow ioBroker addon exactly two times before it shows up again

**To Reproduce**  
Steps to reproduce the behavior:
1. Restart HA
2. No ecoflow values
3. Restart Ecoflow ioBroker addon
4. still no values
5. Restart Ecoflow ioBroker addon second time
6. Values instantly show

**Expected behavior**  
Not sure why a HA restart screws it up. My MQTT server is not an addon in HA, but in a separate docker container

**Screenshots & Logfiles**  
The following shows up both times after restarting the Ecoflow addon:
<html><body>

ecoflow-mqtt.0 | 2025-06-05 09:58:49.456 | warn | [HA STATE INIT DATA] P351Zxxxxxxxxxxxxxxxx.setDp3.cfgDc12vOutOpen getState returned -> null
-- | -- | -- | --
ecoflow-mqtt.0 | 2025-06-05 09:58:49.414 | warn | [HA STATE INIT DATA] P35xxxxxxxxxxxxxxx.setDp3.cfgAcOutOpen getState returned -> null

</body>
</html>

**Versions:**  
Latest HA, ioBroker, and ecoflow addon

foxthefox, 2025-06-05:
Which version of the adapter do you use?
With >=1.3.1 there were improvements for reconnection to the MQTT server for HA communication. That worked so far.

The message:
```
ecoflow-mqtt.0 | 2025-06-05 09:58:49.456 | warn | [HA STATE INIT DATA] P351Zxxxxxxxxxxxxxxxx.setDp3.cfgDc12vOutOpen getState returned -> null -- | -- | -- | -- ecoflow-mqtt.0 | 2025-06-05 09:58:49.414 | warn | [HA STATE INIT DATA] P35xxxxxxxxxxxxxxx.setDp3.cfgAcOutOpen getState returned -> null 
```
looks not nice, but can be explained.
The datapoints with setDp3 refer to commands to the device. After starting of the adapter all datapoints are null and after having an update they get filled with their values. For commands you have to give one in order to fill the data into the datapoint. It can not be retrieved from the device. Unless you give a command it stays null. Its not a bug.

ampersandru, 2025-06-05:
> Which version of the adapter do you use? With >=1.3.1 there were improvements for reconnection to the MQTT server for HA communication. That worked so far.
>
> The message:
>
> ```
> ecoflow-mqtt.0 | [...]

thanks for the response, the adapter version is v1.3.2. Everything works perfectly for days/weeks until I restart HA

ampersandru, 2025-06-23:
as a workaround, I created a portainer rest command to restart the iobroker container and an automation to run it twice upon HA startup

HAconfiguration.yaml: 
```
rest_command:
  restart_iobroker_container:
    url: "https://192.168.xx.xx:9443/api/endpoints/2/docker/containers/CONTAINER-ID/restart"
    method: post
    verify_ssl: false # Set to false if you use a self-signed certificate and know the risks
    headers:
      X-API-Key: "PORTAINER-API-KEY"    
```

Automation:
```
alias: Home Assistant Restart Actions
description: ""
triggers:
  - event: start
    trigger: homeassistant
actions:
  - action: rest_command.restart_iobroker_container
    metadata: {}
    data: {}
  - delay:
      hours: 0
      minutes: 1
      seconds: 0
      milliseconds: 0
  - action: rest_command.restart_iobroker_container
    metadata: {}
    data: {}
mode: single
```

foxthefox, 2025-09-13:
Please have a check with version 1.4.5, where further improvements were done

the1snm, 2025-09-20:
First off I forgot to mention this in my last message, thank you for all your work on this. This gives me so much more insight into my products and let's me leverage home assistant as well. Regarding the settings you mentioned, I can confirm I have those options configured as you mentioned. Just today I migrated my IOBroker from a Docker container on my Synology NAS over to a Debian LXC in Proxmox so we'll see if that was affecting anything. I also noticed that the version updated from 1.4.5 to 1.4.6 so I'll post back if that potentially plays in as well. 

[image]
[image]

foxthefox, 2026-02-09:
I posted an example log file in #375 when stopping and starting the MQTT Broker in HA.
This should have similar behavior as a complete HA restart