foxthefox/ioBroker.ecoflow-mqtt #277: Very high llcbusvol (450), is it a bug or a problem with the device

nijhawank, 2025-06-24:
I have a River 3 Plus WiFi unit available from Costco. I see a lot of log entries saying this… 
State value to set for "ecoflow-mqtt.0.xxx.RuntimePropertyUpload.llcBusVol" has value "450.709228515625" greater than max "60"
State value to set for "ecoflow-mqtt.0.xxx.RuntimePropertyUpload.llcBusVol" has value "451.37481689453125" greater than max "60"

is the a bug in the code or a real issue with my unit? I don’t see any fault history in the app and the unit seems to be working all fine.

foxthefox, 2025-06-25:
From my point of view the value is 10 times higher than the anticipated value, so the multiplication factor needs to be adapted in the code. I will do this in next version.
Since the API is not declared somewhere, it can happen that some values are not interpreted correcty and rely on feedback, as yours. Thanks.

foxthefox, 2025-07-11:
I have created a preview of the next version.
Could you please check it.

it must be installed via expert mode -> custom, described here:
https://github.com/foxthefox/ioBroker.ecoflow-mqtt/blob/1.4.0/doc/en/IOB_HA/adapter.md#new-version-for-testing-on-github-not-main-branch

after installation you need to:
- stop the adapter
- delete the llcbusvol datapoint
- start the adapter

nijhawank, 2025-07-13:
I followed the instructions and it is now at version 1.4.0 and I also deleted the llcbusvol data point. However, I still see the same warnings.