foxthefox/ioBroker.ecoflow-mqtt #284: Only 3 batteries reported on a 4 battery Delta Pro Ultra setup

DPattee, 2025-07-02:
**Describe the bug**  
Looking in the iobroker object viewer there is BPInfo, BPInfo2, and BPInfo3, and the info in those correctly matches batteries 1,2, and 3 (eg the bpSoc and bpTemp in iobroker match what is shown in the ecoflow app)

But, there should be a BPInfo4 for the 4th battery. 

**To Reproduce**  
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '...'
3. Scroll down to '....'
4. See error

**Expected behavior**  
A clear and concise description of what you expected to happen.

**Screenshots & Logfiles**  
If applicable, add screenshots and logfiles to help explain your problem.

**Versions:**  
 - Adapter version: 1.3.2
 - JS-Controller version: 7.0.6
 - iobroke version: v7.6.17
 - Node version: 20.19.2
 - Operating system: docker, synology

**Additional context**  
Add any other context about the problem here.

foxthefox, 2025-07-02:
Since I am already preparing a major update, I can also include a 4 th battery.
It will be implemented with the next version.
Maybe some testing needs to be done by you.

foxthefox, 2025-07-11:
I have created a preview of the next version.
Could you please check it.

it must be installed via expert mode -> custom, described here:
https://github.com/foxthefox/ioBroker.ecoflow-mqtt/blob/1.4.0/doc/en/IOB_HA/adapter.md#new-version-for-testing-on-github-not-main-branch

foxthefox, 2025-07-16:
Thanks for testing, I will fix this. So that the device battery is BPInfo and the first extra is BPInfo1.

foxthefox, 2025-09-13:
I simulated it and found it correct working.
I will close the issue, if there is something still to do, please reopen it

closed 2025-09-13 by foxthefox

the1snm, 2025-09-13:
> I simulated it and found it correct working. I will close the issue, if there is something still to do, please reopen it

I can confirm these are working now in 1.4.5. The numbering order with BPInfo doesn't line up with the actual pack # but that can be seen in the listed sensors.