foxthefox/ioBroker.ecoflow-mqtt #288: Ecoflow Stream AC Pro / Ultra: Home Assistant communication not fully working

timoage, 2025-07-31:
Hey, thank you very much for your hard work!

I just tested the 1.4.0 for my Stream Ultra and got all the data into iobroker. But for some weird reason Im not able to get the actual values into Home Assistant. All the mqtt devices are created and also all the sensors in them:

[image]

[image]

[image]

Do you have an idea whats happening here?

mobo-git, 2025-08-01:
Hi @timoage, I had the same problem. I've just unchecked the "enable communication with Homeassistant MQTT Broker" checkbox -> save, re-check->  save (Instance restarted automatically), now I receive data in HA.

[image]

timoage, 2025-08-01:
Update: Restarting home assistant resulted in updating values:

[image]

But as soon as I restart the ioBroker instance no further updates are published and the state in HA switches back to "Unavailable"

[image]

foxthefox, 2025-09-02:
@timoage 
did you have a chance to check?
meanwhile 1.4.3 is released, whereby there are no changes in regards to HA communication

foxthefox, 2025-09-09:
Thanks for the update and test.
I will check it again at my setup, but there it was also catching the first case (reboot of HA or restart of MQTT).